marcone/teslausb #292: Add copy-music.sh for rsync and rclone

RichieB2B, 2019-10-03:
Currently `copy-music.sh` is only implemented for CIFS. Please consider adding the same functionality for rsync and rclone as well.

crietzschel, 2019-10-06:
A minimal Version could be this one (I'm using):

/root/bin/copy-music.sh
`#!/bin/bash -eu

log "Copying music from a#rchive..."

DST="/mnt/music"

NUM_FILES_BEFORE=$(find $DST -type f |  wc -l)

source /root/.teslaCamRsyncConfig
rsync -auvh --timeout=60 --no-perms --delete --no-o --no-p --no-g --stats $user@$server:$musicpath $DST

NUM_FILES_NOW=$(find $DST -type f |  wc -l)

log "Copied music file(s), now: $NUM_FILES_NOW, before sync: $NUM_FILES_BEFORE files."

if [ $NUM_FILES_NOW -ne $NUM_FILES_BEFORE ]
then
  /root/bin/send-push-message "TeslaUSB:" "Copied music file(s), now: $NUM_FILES_NOW, before sync: $NUM_FILES_BEFORE files."
fi`

/root/.teslaCamRsyncConfig
`user=<remote>
server=<remote host>
path=/home/pi/cam
musicpath=<remote-dir>`

Known Bugs:
NUM_FILES_NOW / NUM_FILES_BEFORE isn't exact. It's okay for me ;-)

marcone, 2019-11-12:
Happy to consider pull requests, but I won't be working on this. I also feel like people that are using rclone or rsync for archiving won't necessarily store their music on the same rsync/rclone service, so an ideal solution would allow specifying separate archive and music sync methods.

closed 2019-11-12 by marcone

cpbl, 2020-03-06:
@crietzschel Are you still using rsync with music? It would be great if someone could make a pull request with better rsync documentation, and maybe to complete the music copying feature!
I am finding it hard.

RichieB2B, 2020-03-06:
Unfortunately as discovered in #319 rsync doesn't handle network issues too well. I did some experimenting and was not able to get a satisfactory setup where rsync would detect the WiFi disappearing (i.e. when driving off). This would cause rsync to hang indefinitely. So I gave up on it for now.

cpbl, 2020-03-06:
Hm, rsync is ridiculously robust in my experience in other contexts. Did you have any luck with explicitly changing its `--timeout=TIMEOUT` parameter?

cpbl, 2020-03-06:
Sorry to hear it.  
I was actually interested in rsync to an rsync service (or GNU/Linux filesystem), not to CFIS (#319).

rsync is ridiculously robust in my experience in other contexts. Did you have any luck with explicitly changing its --timeout=TIMEOUT parameter?

RichieB2B, 2020-03-06:
Yes, I have tested it with an rsync server. The timeout parameter only works for the initial connection. If you kill the WiFi after the connection is established rsync just hangs.