marcone/teslausb #294: Use rsync over CIFS

RichieB2B, 2019-10-03:
`archive-clips.sh` for [cifs_archive](https://github.com/marcone/teslausb/blob/main-dev/run/cifs_archive/archive-clips.sh) is much more complicated and slower than for [rsync_archive](https://github.com/marcone/teslausb/blob/main-dev/run/rsync_archive/archive-clips.sh)
It is perfectly possible to run rsync with a CIFS mounted target directory. This will simplify and speed up the archiving.

millercentral, 2019-10-12:
Do you have a config example of how to do this? Are the SSH key steps in the rsync setup docs needed for this?

RichieB2B, 2019-10-12:
This requires a change in `archive-clips.sh` for CIFS, making it much simpler. No need for ssh keys when mounting the archive over CIFS.

dopeghoti, 2019-10-13:
Indeed, once CIFS is mounted locally, `rsync` just sees it as another local filesystem to copy things to.

sboesebeck, 2019-10-13:
i was just trying to implement that. I found the Idea of using rsync over CIFS mount quite intreaguing 😉 

I created a branch in my fork called `cifsrsync` - the name of the archiving method is the same...

I literally just finished the first draft, not tested yet...
but if you are interested, I will post here my findings

sboesebeck, 2019-10-14:
I have got a working version in the branch mentioned above, I got it to work. At least in my testing environment.

Use at own risk, did not run it in the wild yet.

RichieB2B, 2019-10-15:
I found your branch: https://github.com/sboesebeck/teslausb/blob/cifsrsync/run/cifsrsync_archive/archive-clips.sh

You created a new new archive system. That works but you could have just changed the cifs system. Once this works there is no need to keep the old script around. 

I would recommend adding `--remove-source-files` to the rsync command so the transferred files are removed from the Pi.. Did you look at https://github.com/marcone/teslausb/blob/main-dev/run/rsync_archive/archive-clips.sh for inspiration?

sboesebeck, 2019-10-15:
Thanks for the feedback. I was creating a new system because I wanted the existing cifs version to stay as it is - works fine. No need to change. 

At some point we could replace the copy approach by rsync - was not sure is that was a to destructive change for my first commit here 😉 

I will have a closer look at the rsync archive-clips and maybe add some ideas. Thanks.

RichieB2B, 2019-10-15:
Replacing custom bash sync code by using rsync is an improvement. No need to create a new archive system for it. And it is not destructive until @marcone accepts your pull request anyway.

sboesebeck, 2019-10-15:
Thanks for the advice. I included it into the cifs archive-sync.sh script and created a pull request.
I also took a look at the rsync archive script. I learned, that rsync has some tricks I did not know for the last 10 years... Thanks 😉 

I added some of that into the script. but I wanted to keep the overall structure the same. This has the advantage, that every directory is synced separately and deleted afterwards. Not sync all and delete all. I like that a bit better...

I also kept the "connection monitoring" part. It will make the script stop as soon as the archive is gone, preventing rsync from doing something "wrong" (for example the mount point is not there, it will find an empty directory and might try syncing there...).

just a thought. 

The script is running on my pi-zero and did work fine today.

RichieB2B, 2019-10-23:
Thanks for merging!

closed 2019-10-23 by RichieB2B

sboesebeck, 2019-10-23:
Thanks a lot!

marcone, 2019-10-31:
I've reverted the change to use rsync over cifs, because when used this way rsync takes way too long to realize the server is no longer reachable (the --timeout option does not work when copying between folders).
Happy to consider a new PR that addresses this.

sboesebeck, 2019-11-02:
Well, the first version if the rsync over cifs was using the detection method, that the copy part uses. It synced each directory separately. This is one of the reasons, why I prefered that solution. And I still prever it... should we try the "old" version once more?

The advantages are it is a bit less complex and would rely on a proven synchronization algorithm of rsync. 
It might be a bit slower compared to copy/move though, as this does not have the somewhat long initialization phase... but this should not be a big problem, when syncing every dir on its own.

I am not sure, but I also think, rsync is capable of resuming sync within a file, which would be a huge benefit...

marcone, 2019-11-03:
Couldn't the connection checker be used just as well when syncing everything at once?
I'm not sure I understand your reasoning for syncing each directory separately, but it kinda sounds like it's a workaround for rsync's inability to deletes files as it goes.

sboesebeck, 2019-11-04:
yes, the connection checker should work in that case also.
But I'd like to split this up in order to be more flexible. For me, one of the reasons that also speak for splitting it up is, that the time for the start to transfer is reduced.

if you have a lot of clips, rsync takes quite some time to start syncing. The more files to sync, the longer it takes to start.

If we split that up, it will sync in smaller portions which might also be an advantage if the cifs directory is not available long enough.  (divide and conquer)

but just using the connection checker will probably also do the trick :-)