marcone/teslausb #296: Update required doing "apt --fix-broken install"

alanhoyle, 2019-10-05:
I was trying to update per the instructions on the [wiki](https://github.com/marcone/teslausb/wiki), but the underlying libraspberrypi libraries needed to be updated, and the I had to run `apt --fix-broken install` before the `/root/bin/setup-teslausb` program would successfully run.

marcone, 2019-10-05:
Not really enough information here do act on.

marcone, 2019-10-05:
(did you check the setup logs)

closed 2019-10-05 by marcone

alanhoyle, 2019-10-05:
```
Downloaded /tmp/configure.sh ...
~ ~
Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
/tmp/create-backingfiles-partition.sh: line 14: hash: mkfs.xfs: not found
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
```
`dpkg --configure -a` didn't fix anything.  It reported:  
```
root@teslausb:~# sudo dpkg --configure -a
dpkg: dependency problems prevent configuration of libraspberrypi-doc:
 libraspberrypi-doc depends on libraspberrypi0 (= 1.20190517-1); however:
  Version of libraspberrypi0 on system is 1.20190401-1.

dpkg: error processing package libraspberrypi-doc (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of libraspberrypi-dev:
 libraspberrypi-dev depends on libraspberrypi0 (= 1.20190517-1); however:
  Version of libraspberrypi0 on system is 1.20190401-1.

dpkg: error processing package libraspberrypi-dev (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 libraspberrypi-doc
 libraspberrypi-dev
```
so I ended up doing an `apt update && apt upgrade` which updated  libraspberrypi0 to 1.20190819~stretch-1 and removed a bunch of /boot diversions.  

Then, the `/root/bin/setup-teslausb` worked.