marcone/teslausb #297: rclone webdav - Stopped working after V10 updates

loftux, 2019-10-05:
I've been using rclone with webdav, and uploads worked prior to V10 teslausb updates.
The error is (sample here, more in diagnostics.txt attached)
`2019/10/04 18:17:13 ERROR : 2019-10-04_08-23-25/2019-10-04_08-22-54-right_repeater.mp4: Failed to copy: Update mkParentDir failed: 409 Conflict`
[diagnostics.txt](https://github.com/marcone/teslausb/files/3693000/diagnostics.txt)

I've run updates for both teslausb and rclone to the lates, this did not help.
Tested to re-create /backingfiles. Teslausb record files fine before and after, and I can retrieve files when connected to a laptop. 
I've deleted all existing folders and files in target server (Alfresco webdav connector) to start clean.

This may very well be an issue with rclone (I've seen references to 409 webdav error there) or an issue with the receiving server (no changes on that server). 
"Update mkParentDir failed" implies that the additonal folders in V10, or a loop when syncing creates conflict with existing folders?

Is there anything I can do to add additional debug output for rclone, or any other step?

marcone, 2019-10-05:
The only reference I found to that message is at https://github.com/rclone/rclone/issues/2716, which seems to indicate it's a known problem that's being worked on.
It's odd that it only started with v10 for you though, since that only adds an extra folder to backup.

DMBlakeley, 2019-10-05:
This is not a solution just a comment of a possible alternative. 

I was using rclone with Microsoft OneDrive as I wanted the files to easily be viewable from my iPhone or iPad if necessary. Had a number of problems with this as time on the RPi needed to be synchronized otherwise the license key was considered invalid. Did quite a bit of investigation into the rclone_archive routines and also submitted some corrections. Found that for rclone and OneDrive the 2019-04-08-raspbian-stretch-lite.img worked much more consistently than the 2019-07-10-raspbian-buster-lite.img particularly due to buster not handling the time synchronization.

Everything was working fine but then I realized a bit simpler way as the one-step method with prebuilt image and teslausb_setup_variables.conf was just easier to update and maintain. After a bit of frustration setting up CIFS on my iMac I found that I could mount a folder in the OneDrive folder on my iMac as a shared folder. Now when I get home the files move smoothly to my iMac and in turn the iMac synchronizes to OneDrive for viewing on other devices.

I have of course not been updated to v10 yet so we'll see if success continues. I will also only use the stretch pre-built image for now.

loftux, 2019-10-06:
I did try to create a dummy folder structure + files, and then ran the same command found in archive-clips.sh.
Getting the same 409 error, so this looks like an rclone issue. The only thing I can think of is that there is something in raspbian package that has changed. Will try to start over without upgrading packages.

loftux, 2019-10-07:
Started over with image teslausb-20190615. No updates done.
Installed ntp separately as this wasn't installed.

So now the files upload with rclone webdav, there is still some errors. Only 8 files actually uploaded.

```
Mon  7 Oct 18:12:44 CEST 2019: Sending IFTTT message for moved files.
Mon  7 Oct 18:12:45 CEST 2019: Moving clips to rclone archive...
2019/10/07 18:16:00 ERROR : : error reading source directory: directory not found
2019/10/07 18:16:00 ERROR : Attempt 1/3 failed with 2 errors and: directory not found
2019/10/07 18:16:00 ERROR : : error reading source directory: directory not found
2019/10/07 18:16:00 ERROR : Attempt 2/3 failed with 2 errors and: directory not found
2019/10/07 18:16:00 ERROR : : error reading source directory: directory not found
2019/10/07 18:16:00 ERROR : Attempt 3/3 failed with 2 errors and: directory not found
2019/10/07 18:16:00 Failed to move with 2 errors: last error was: directory not found
Mon  7 Oct 18:16:00 CEST 2019: Moved 9 file(s).
Mon  7 Oct 18:16:00 CEST 2019: Sending IFTTT message for moved files.
Mon  7 Oct 18:16:02 CEST 2019: Unmounting /mnt/cam...
```

marcone, 2019-10-12:
The "directory not found" errors could be because the rclone archive script blindly tries to archive the TeslaCam/SavedClips and TeslaCam/RecentClips folders, without first checking if they actually exist.
The number files should be number of actual files, not directories, but I think there's an error in the script in that it counts directories under SavedClips as well.

I'll fix both of these.

closed 2019-10-13 by marcone