marcone/teslausb #301: Setup correctly but not seeing any recorded videos

wadewegner, 2019-10-06:
I believe I was able to setup the pi (at least reviewing the `teslausb-healess-setup.log` file everything appears to have worked properly after the 5th try`) but I'm not seeing any videos synching to my network share. When I take out the micro SD and plug into my PC, I don't see any shares with videos either. Any debug pointers?

curiousgrge, 2019-10-06:
Unless you are mounting the partitions through some *nix flavor, you won't see the files.  Suggest you ssh into the pi to see if the files are located somewhere in /backingfiles before attempting to tackling file share issues.

japhule, 2019-10-07:
Plug the Pi into your PC (do not remove the sd card). After a minute or two, it should detect the CAM drive with a TeslaCam folder with your footage in it. If it doesn't show then set up is not correct. If it does show, then SSH into your pi and check the archiveloop.log file found in the /mutable folder, it's possible that its not able to connect to your file share to upload data.