marcone/teslausb #303: not archiving files to samba share

stepir, 2019-10-08:
Hi,

my archive server is reachable - actually I can see in its logs my PI is logging in OK.

/mutable/archiveloop.log  show the following:

`Tue  8 Oct 23:45:12 CEST 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/SentryClips/2019-10-08_21-21-42/2019-10-08_21-21-13-back.mp4
Tue  8 Oct 23:45:12 CEST 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/SentryClips/2019-10-08_21-21-42/2019-10-08_21-21-13-front.mp4
Tue  8 Oct 23:45:12 CEST 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/SentryClips/2019-10-08_21-21-42/2019-10-08_21-21-13-left_repeater.mp4
Tue  8 Oct 23:45:12 CEST 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/SentryClips/2019-10-08_21-21-42/2019-10-08_21-21-13-right_repeater.mp4
Tue  8 Oct 23:45:12 CEST 2019: made all links for /backingfiles/snapshots/newsnap/mnt
Tue  8 Oct 23:45:12 CEST 2019: Waiting for time to be set by ntpd...
Tue  8 Oct 23:45:39 CEST 2019: Time still not set, attempting to force it
Tue  8 Oct 23:45:48 CEST 2019: Archiving...
Tue  8 Oct 23:45:48 CEST 2019: Disconnecting usb from host...
Tue  8 Oct 23:45:48 CEST 2019: Disconnected usb from host.
Tue  8 Oct 23:45:48 CEST 2019: Ensuring cam archive is mounted...
Tue  8 Oct 23:45:49 CEST 2019: Mounting /mnt/archive...
Tue  8 Oct 23:45:49 CEST 2019: Mounted /mnt/archive.
Tue  8 Oct 23:45:49 CEST 2019: Ensured cam archive is mounted.
Tue  8 Oct 23:45:49 CEST 2019: Checking saved folder count...
Tue  8 Oct 23:45:49 CEST 2019: Ensuring cam file is mounted...
Tue  8 Oct 23:45:49 CEST 2019: Mounting /mnt/cam...
Tue  8 Oct 23:45:49 CEST 2019: Mounted /mnt/cam.
Tue  8 Oct 23:45:50 CEST 2019: Ensured cam file is mounted.
Tue  8 Oct 23:45:50 CEST 2019: Running fsck on /mnt/cam...
Tue  8 Oct 23:45:50 CEST 2019: | fsck from util-linux 2.29.2
Tue  8 Oct 23:45:53 CEST 2019: | fsck.fat 4.1 (2017-01-24)
Tue  8 Oct 23:45:53 CEST 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Tue  8 Oct 23:45:53 CEST 2019: |  Automatically removing dirty bit.
Tue  8 Oct 23:45:53 CEST 2019: | Performing changes.
Tue  8 Oct 23:45:53 CEST 2019: | /dev/loop1: 223 files, 132582/3520805 clusters
Tue  8 Oct 23:45:53 CEST 2019: Finished fsck on /mnt/cam.
Tue  8 Oct 23:45:53 CEST 2019: There are 1 event folder(s) with 44 file(s) to move.
Tue  8 Oct 23:45:53 CEST 2019: Starting recording archiving: 1 event folder(s) with 44 file(s)
Tue  8 Oct 23:46:00 CEST 2019: Temporarily enabling Sentry Mode to power the RPi while archive job completes...
Success: Sentry Mode Enabled
Tue  8 Oct 23:46:06 CEST 2019: Moving clips to archive...
Tue  8 Oct 23:46:06 CEST 2019: /mnt/cam/TeslaCam/SavedClips does not exist, skipping
Tue  8 Oct 23:46:06 CEST 2019: Creating output directory 'SentryClips/2019-10-08_21-21-42'
Tue  8 Oct 23:46:06 CEST 2019: Failed to create 'SentryClips/2019-10-08_21-21-42', check that archive server is writable and has free space
Tue  8 Oct 23:46:07 CEST 2019: Moved 0 file(s), failed to copy 0, deleted 0.
Tue  8 Oct 23:46:07 CEST 2019: Finished moving clips to archive.
Tue  8 Oct 23:46:07 CEST 2019: Restoring Sentry Mode to its previous state (disabled)...
Success: Sentry Mode Disabled
Tue  8 Oct 23:46:14 CEST 2019: Unmounting /mnt/cam...
Tue  8 Oct 23:46:15 CEST 2019: Unmounted /mnt/cam.
Tue  8 Oct 23:46:15 CEST 2019: Finished archiving.
Tue  8 Oct 23:46:15 CEST 2019: Music archive not configured or unreachable
Tue  8 Oct 23:46:15 CEST 2019: unmounting /mnt/archive
Tue  8 Oct 23:46:15 CEST 2019: Connecting usb to host...
Tue  8 Oct 23:46:15 CEST 2019: Connected usb to host.
Tue  8 Oct 23:46:15 CEST 2019: Waiting for archive to be unreachable...`

No files are being transferred in the configured cifs share. What else can I check to understand the issue?

Thanks

marcone, 2019-10-08:
Run `sudo mount /mnt/archive` on the pi and see what it says.
If that succeeds, try `sudo mkdir -p /mnt/archive/SentryClips/2019-10-08_21-21-42` and see what that says

stepir, 2019-10-08:
pi@eldiablousb:~ $ sudo mount /mnt/archive
pi@eldiablousb:~ $ sudo mkdir -p /mnt/archive/SentryClips/2019-10-08_21-21-42
mkdir: cannot create directory '/mnt/archive/SentryClips': Permission denied

closed 2019-10-09 by marcone

j450h1, 2019-10-16:
What was the resolution to this?

j450h1, 2019-10-16:
I think I can answer my own question here. I went to the sharing settings of the folder I was sharing in Windows and discovered that my user didn't have the appropriate permissions. So once I updated the permissions, I now no longer have the **Permission denied** message.