marcone/teslausb #305: Car says, SD card is corrupted, but the music parition is working

bahree, 2019-10-09:
So this started recently and rebooting the car or the pi doens't help. I get the grey x and a warning logo - touching which opens up a pop up saying the SD card is corrupted and cannot save clips.

I do see the music partition is loading and working as expected. I did pull the latest update a couple of days ago and this seems to have been since then. I am on the latest V10 build.

Any one else has seen this or have any suggestions? This is on the PiZero and using the image here.

I haven't ran a diagnostics yet - but happy to do that.

Thanks.

randallzapata, 2019-10-09:
Is the cam mounted?
mount /mnt/cam/

bahree, 2019-10-09:
Yes, it is - I can see that when it is connected to the car (over ssh). Yes I see the TeslaCam folder, and within that the other three folders: RecentClips  SavedClips  SentryClips

I can also see the folder show up on Windows when I connect it via USB and the pi boots up. 

Attached is what the car shows. And I also [uploaded the diagnostics here](https://pastebin.com/s0zVqGfF). I did have to truncate some of the archiveloop logs in there as I was exceeding the pastbin's limit. 

As far as I can tell nothing seems out of the ordinary, except when I see some I/O errors towards the end - looks like bad sectors? Pasted a subset here. Could that be the issue? 

```
[  116.886490] Buffer I/O error on dev loop1, logical block 51125, lost async page write
[  116.886695] loop: Write error at byte offset 27254272, length 512.
[  116.886708] print_req_error: I/O error, dev loop1, sector 51183
[  116.886721] Buffer I/O error on dev loop1, logical block 51183, lost async page write
[  116.886924] loop: Write error at byte offset 30310400, length 512.
[  116.886937] print_req_error: I/O error, dev loop1, sector 57152
[  116.886949] Buffer I/O error on dev loop1, logical block 57152, lost async page write
[  116.887153] loop: Write error at byte offset 30517760, length 512.
[  116.887166] print_req_error: I/O error, dev loop1, sector 57557
[  116.887178] Buffer I/O error on dev loop1, logical block 57557, lost async page write
[  117.240835] Mass Storage Function, version: 2009/09/11

```
And when I try and run fsck, I get the following error. Any suggestions?

Many thanks.

```
root@teslausb:~# fsck
fsck from util-linux 2.29.2
e2fsck 1.43.4 (31-Jan-2017)
/dev/mmcblk0p2 is mounted.
e2fsck: Cannot continue, aborting.
```

[image]
[image]

marcone, 2019-10-11:
Since dmesg shows I/O errors, it looks like your sd card has gone bad. You said you see the drive and folders on Windows. Can you write to it too?

mane-wt, 2019-10-11:
I experienced The exakt same behavior. Music is working but a Grey cross on The dashcam symbol. At first it said sd-card too slow. After reboot it worked again for a while then it said sd-card corrupted but The Music still worked. Reboot did not fix The dashcam.

marcone, 2019-10-12:
They are not partitions, neither to the car (which sees music and recordings as two separate USB drives) nor on the Pi (which uses two different files in the same partition to emulate the USB drives)

The messages in dmesg seem to indicate at least some sectors on the sd card are bad.

closed 2019-10-12 by marcone