marcone/teslausb #307: My Tesla kills USB slot while Pi is archiving. Requires Tesla reboot to get it back up.

dsgerbc, 2019-10-09:
Hello. Just discovered a backlog of not synced files on my Pi. Upgraded to v10-compatible build. Rebooted. It started archiving, but has so far cut out 2-3 times, completely taking down USB.  I was connected via SSH and lost communications. No lights on the Pi when I got to the car, and the same USB slot won't even supply power until after interface reboot later on. The other slot powered the Pi up, but then it quickly killed the second slot as well. Not a clue why this is happening.  There is a large-ish back log, however. Maybe 2-3 hours worth of Sentry videos.

It's been working okayish prior to V10 update, with just some nagging issues of occasional USB drops when leaving home.

Diagnostics attached.
[diagnostics.txt](https://github.com/marcone/teslausb/files/3705512/diagnostics.txt)

marcone, 2019-10-09:
Looks like you need to configure teslausb to use the tesla api to keep the car awake while archiving (see doc)

closed 2019-10-09 by marcone

dsgerbc, 2019-10-09:
Tesla API has been configured long time ago and worked fine. Teslausb_setup_variables.conf has all the right commands setup/uncommented.  When this has not been configured a while ago USB power came back after getting back into the car. With this issue, USB does not come back, does not charge the phone, nothing. Requires Tesla reboot to come back.