marcone/teslausb #308: Not copying off in version 10

spalt, 2019-10-09:
Any idea about this?  I ran a "self update" but it used to work but doesn't any longer.

`==============================================
Thu  3 Nov 13:17:11 EDT 2016: Starting archiveloop...
Thu  3 Nov 13:17:13 EDT 2016: mounted all snapshots
Thu  3 Nov 13:17:14 EDT 2016: low space, deleting /backingfiles/snapshots/snap-000000
Wed  9 Oct 07:03:00 EDT 2019: warning: low space for snapshots
Wed  9 Oct 07:03:00 EDT 2019: taking snapshot of cam disk: /backingfiles/snapshots/snap-000000/snap.bin
Wed  9 Oct 07:03:13 EDT 2019: took snapshot
Wed  9 Oct 07:03:14 EDT 2019: comparing /backingfiles/snapshots/snap--00001/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Wed  9 Oct 07:03:14 EDT 2019: making links for /backingfiles/snapshots/newsnap/mnt, retargeted to /backingfiles/snapshots/snap-000000/mnt
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-01-50-back.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-01-50-front.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-01-50-left_repeater.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-01-50-right_repeater.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-02-51-back.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-02-51-front.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-02-51-left_repeater.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-02-51-right_repeater.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-03-51-back.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-03-51-front.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-03-51-left_repeater.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-03-51-right_repeater.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-04-52-back.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-04-52-front.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-04-52-left_repeater.mp4
Wed  9 Oct 07:03:14 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-04-52-right_repeater.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-05-52-back.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-05-52-front.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-05-52-left_repeater.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-05-52-right_repeater.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-06-53-back.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-06-53-front.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-06-53-left_repeater.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-06-53-right_repeater.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-07-53-back.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-07-53-front.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-07-53-left_repeater.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-07-53-right_repeater.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-08-53-back.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-08-53-front.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-08-53-left_repeater.mp4
Wed  9 Oct 07:03:15 EDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-08_21-08-53-right_repeater.mp4
Wed  9 Oct 07:03:15 EDT 2019: made all links for /backingfiles/snapshots/newsnap/mnt
Wed  9 Oct 07:03:15 EDT 2019: Waiting for time to be set by ntpd...
Wed  9 Oct 07:03:16 EDT 2019: Time now set
Wed  9 Oct 07:03:16 EDT 2019: Ensuring music backing file is mounted...
Wed  9 Oct 07:03:16 EDT 2019: Mounting /mnt/music...
Wed  9 Oct 07:03:16 EDT 2019: Mounted /mnt/music.
Wed  9 Oct 07:03:16 EDT 2019: Ensured music drive is mounted.
Wed  9 Oct 07:03:17 EDT 2019: Running fsck on /mnt/music...
Wed  9 Oct 07:03:17 EDT 2019: | fsck from util-linux 2.29.2
Wed  9 Oct 07:03:18 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Wed  9 Oct 07:03:18 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Wed  9 Oct 07:03:18 EDT 2019: |  Automatically removing dirty bit.
Wed  9 Oct 07:03:18 EDT 2019: | Performing changes.
Wed  9 Oct 07:03:18 EDT 2019: | /dev/loop1: 34 files, 9458/1537826 clustersWed  9 Oct 07:03:18 EDT 2019: Finished fsck on /mnt/music.
Wed  9 Oct 07:03:18 EDT 2019: Unmounting /mnt/music...
Wed  9 Oct 07:03:18 EDT 2019: Unmounted /mnt/music.
Wed  9 Oct 07:03:18 EDT 2019: Archiving...
Wed  9 Oct 07:03:18 EDT 2019: Disconnecting usb from host...
Wed  9 Oct 07:03:18 EDT 2019: Disconnected usb from host.
Wed  9 Oct 07:03:19 EDT 2019: Ensuring cam archive is mounted...
Wed  9 Oct 07:03:19 EDT 2019: /mnt/archive is already mounted.
Wed  9 Oct 07:03:19 EDT 2019: Ensured cam archive is mounted.
Wed  9 Oct 07:03:19 EDT 2019: Ensuring music archive is mounted...
Wed  9 Oct 07:03:19 EDT 2019: /mnt/musicarchive is already mounted.
Wed  9 Oct 07:03:19 EDT 2019: Ensured music archive is mounted.
Wed  9 Oct 07:03:19 EDT 2019: Checking saved folder count...
Wed  9 Oct 07:03:19 EDT 2019: Ensuring cam file is mounted...
[... 20 lines cut ...]
Wed  9 Oct 07:03:23 EDT 2019: Ensuring music backing file is mounted...
Wed  9 Oct 07:03:23 EDT 2019: Mounting /mnt/music...
Wed  9 Oct 07:03:23 EDT 2019: Mounted /mnt/music.
Wed  9 Oct 07:03:23 EDT 2019: Ensured music drive is mounted.
Wed  9 Oct 07:03:23 EDT 2019: Running fsck on /mnt/music...
Wed  9 Oct 07:03:23 EDT 2019: | fsck from util-linux 2.29.2
Wed  9 Oct 07:03:24 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Wed  9 Oct 07:03:24 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Wed  9 Oct 07:03:24 EDT 2019: |  Automatically removing dirty bit.
Wed  9 Oct 07:03:24 EDT 2019: | Performing changes.
Wed  9 Oct 07:03:24 EDT 2019: | /dev/loop1: 34 files, 9458/1537826 clustersWed  9 Oct 07:03:24 EDT 2019: Finished fsck on /mnt/music.
Wed  9 Oct 07:03:24 EDT 2019: Copying music from archive...
Wed  9 Oct 07:03:24 EDT 2019: Copied 0 music file(s), skipped 28 previously-copied files, encountered 0 errors.
Wed  9 Oct 07:03:24 EDT 2019: Unmounting /mnt/music...
Wed  9 Oct 07:03:25 EDT 2019: Unmounted /mnt/music.
Wed  9 Oct 07:03:25 EDT 2019: Finished copying music.
Wed  9 Oct 07:03:25 EDT 2019: unmounting /mnt/archive
Wed  9 Oct 07:03:25 EDT 2019: unmounting /mnt/musicarchive
Wed  9 Oct 07:03:25 EDT 2019: Connecting usb to host...
Wed  9 Oct 07:03:25 EDT 2019: Connected usb to host.
Wed  9 Oct 07:03:25 EDT 2019: Waiting for archive to be unreachable...
pi@teslausb:/mutable $
pi@teslausb:/mutable $
pi@teslausb:/mutable $
pi@teslausb:/mutable $`

closed 2019-10-09 by marcone

spalt, 2019-10-09:
There definitely are, however.  If I plug the pi into my PC I can see 196 items in RecentClips and 12 folders in SentryClips....

marcone, 2019-10-09:
RecentClips is not archived (never has been).
Are there actual files in those SentryClips folders? (sometimes the files are truncated, often to 595 bytes), and those are ignored when archiving.

spalt, 2019-10-09:
Yes, the folders within SentryClips each had 4-40ish files each.  I cleared them out manually.