marcone/teslausb #309: Not delete or check Sentry Files have transferred

simonrb2000, 2019-10-09:
Hi,

The idea behind this project is brilliant. I have used it twice and both times the car must have turned off power to USB port before it finished archiving and had lost Sentry Mode clips (as in they didn't upload).

Is there a way for TeslaUSB ensures that the all files have successfully transferred before removing them from TeslaUSB? I appreciate there are options to keep car awake however in the long term is this an option so we don't loose data?

Many Thanks

Simon

marcone, 2019-10-09:
> Is there a way for TeslaUSB ensures that the all files have successfully transferred before removing them from TeslaUSB?

It already does this. The files aren't deleted from the recording drive until they've been archived.
However note that in v10 the car itself may delete Sentry clips if it needs space.

closed 2019-10-09 by marcone

simonrb2000, 2019-10-09:
This ain’t how mine behaves. Files are removed even if they are not sync’d to the shared drive. The card is a 128gb card running only TeslaUSB so space isn’t an issue. 

Very strange..

marcone, 2019-10-09:
What archive method do you use? (cifs/rsync/rclone)

simonrb2000, 2019-10-09:
cifs

marcone, 2019-10-09:
The [cifs archiving script](https://github.com/marcone/teslausb/blob/main-dev/run/cifs_archive/archive-clips.sh) works by using the `mv` (move) command to move the files from the recording drive to the destination folder on the archive server. The `mv` command only deletes the source file after successfully moving the file to the destination folder.

simonrb2000, 2019-10-09:
Thanks marcone. If the power on the pi is cut before moving files is complete will this cause the issue I am having?

simonrb2000, 2019-10-09:
I have just done a test at home. 6 sentry mode recordings and left Tesla login credentials and the car didn't keep it awake and all 6 sentry mode recordings lost again. Is there any logs that I can post here that would help? As it stands I can't use this as each time I have used it I have gotten home and lost all the sentry mode recordings. Cheers

marcone, 2019-10-09:
It normally shouldn't, but loser power while writing to the card might leave it in a weird state, so possibly.

simonrb2000, 2019-10-09:
Is it simple to change the mv commands in the script to cp commands just for initial testing.

marcone, 2019-10-09:
You can certainly try that. It'll mean that the next time it archives, it'll re-archive those same files.