marcone/teslausb #311: Changing SSH Password

jbienzms, 2019-10-11:
I keep getting warned that my Pi is running the default user name and password. I have read the instructions in #249. I have performed the following:

```
sudo -i
/root/bin/remountfs_rw
passwd
(changed password)
reboot
```

Every time I do those steps, the Pi reboots and has returned to the default user name and password.

What am I doing wrong? Do I need to somehow flush changes to disk before I reboot with the reboot command?

marcone, 2019-10-11:
Run `passwd pi` instead.

closed 2019-10-11 by marcone

miles267, 2019-10-11:
I’ve left my pi login info as default since it’s only on my LAN. Does it matter?

jbienz, 2019-10-11:
Yeah, I realized I'd left the pi part off. Sorry about that. Thanks @marcone.

@miles267 I've wondered this myself. If it's only ever supposed to join my SSID I'm not sure why this matters. If they can get physical access to the device (breaking into the car) then they can read the files right off the microSD card.

marcone, 2019-10-11:
The message being printed is a standard Raspbian thing, it's not something teslausb added.
In my opinion in doesn't matter if you trust everyone and everything on the network(s) that the Pi will connect to.