marcone/teslausb #316: Car fails to detect drive - v2019.32.12.2

stevenvergenz, 2019-10-15:
I've set up the pi with the buster image, setup succeeded AFAIK, and the drive shows up on my PC. But the M3 fails to detect it. No dashcam, no sentry. I've attached my diagnostics log.

[diagnostics.txt](https://github.com/marcone/teslausb/files/3730010/diagnostics.txt)

marcone, 2019-10-15:
The only thing I notice in the diagnostics is that the Pi doesn't appear to be seeing the car either. The usual reason for that is using a charge-only USB cable, or plugging the cable in the Pi's power port instead of the USB/data port.

closed 2019-10-15 by marcone

seattleandrew, 2019-10-20:
related to issue #321

stevenvergenz, 2019-10-20:
Changing the cable fixed this, thanks. I didn't know power-only cables existed.