marcone/teslausb #323: setup-teslausb get_script failed, retrying and curl errors on pi-gen build

magicalyak, 2019-10-22:
I am constantly getting an error if I run my own image using the build instructions and pi-gen.  I create the image, copy the conf to boot and boot up.  However, once I try to run setup-teslausb, I see in the headless log (even if I say false to headless) that there is a curl error repeated.  

at first I got this error
```
Mon 21 Oct 2019 04:45:22 PM EDT : Grabbing main setup file.
Mon 21 Oct 2019 04:45:28 PM EDT : get_script failed, retrying
Mon 21 Oct 2019 04:45:31 PM EDT : get_script failed, retrying
Mon 21 Oct 2019 04:45:34 PM EDT : get_script failed, retrying
Mon 21 Oct 2019 04:45:38 PM EDT : get_script failed, retrying
Mon 21 Oct 2019 04:45:41 PM EDT : get_script failed, retrying
Mon 21 Oct 2019 04:45:55 PM EDT : get_script failed, retrying
Mon 21 Oct 2019 04:45:58 PM EDT : get_script failed, retrying
Mon 21 Oct 2019 04:46:01 PM EDT : get_script failed, retrying
Mon 21 Oct 2019 04:46:05 PM EDT : get_script failed, retrying
Mon 21 Oct 2019 04:46:08 PM EDT : get_script failed, retrying
```

I first saw the error and found by linking the ca-certificates.crt to the local ca it would move forward
`ln -s /etc/ssl/certs/ca-certificates.crt /usr/local/share/ca-certificates/ca-certificates.crt
`
However, I still got this error in the teslausb-headless-setup.log

```
Mon 21 Oct 2019 05:38:37 PM EDT : reading config from /root/teslausb_setup_variables.conf
Mon 21 Oct 2019 05:38:37 PM EDT : curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev
Mon 21 Oct 2019 05:38:38 PM EDT : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Mon 21 Oct 2019 05:38:46 PM EDT : reading config from /root/teslausb_setup_variables.conf
Mon 21 Oct 2019 05:38:46 PM EDT : curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/pi/setup-teslausb
Mon 21 Oct 2019 05:38:47 PM EDT : 'curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/pi/setup-teslausb' failed, retrying
Mon 21 Oct 2019 05:38:50 PM EDT : 'curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/pi/setup-teslausb' failed, retrying
Mon 21 Oct 2019 05:38:54 PM EDT : 'curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/pi/setup-teslausb' failed, retrying
Mon 21 Oct 2019 05:38:57 PM EDT : 'curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/pi/setup-teslausb' failed, retrying
Mon 21 Oct 2019 05:39:01 PM EDT : 'curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/pi/setup-teslausb' failed, retrying
```

The solution that worked for me was to add the environmental variable in /etc/profile.d/ under a new curlssl.sh program that set the CA location.  After a reboot (you could also just source /etc/profile ) it worked as expected.

```
root@teslausb-Model3:/boot# cat /etc/profile.d/curlssl.sh 
export CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt
```

This is better tested running the following command which will say no local CA if it fails
`curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev -v`

It appears similar to this issue https://github.com/Hexxeh/rpi-update/issues/206

I submitted a PR for this https://github.com/marcone/teslausb/pull/322
but I may be the only having the issue.  If anyone else hits this (using Buster and main-dev) on pi-gen please let me know.

Jarery, 2019-10-27:
I have same issue:

Sun 22 Sep 22:55:11 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:55:11 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sun 22 Sep 22:55:11 BST 2019 : Rebooting...
Sun 22 Sep 22:55:12 BST 2019 : Grabbing main setup file.
Sun 22 Sep 22:55:50 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:55:50 BST 2019 : Grabbing main setup file.
Sun 22 Sep 22:56:01 BST 2019 : get_script failed, retrying
Sun 22 Sep 22:56:14 BST 2019 : get_script failed, retrying
Sun 22 Sep 22:56:27 BST 2019 : get_script failed, retrying

magicalyak, 2019-10-27:
@jarery can you confirm it’s not the wrong time and the same as my issue?

Jarery, 2019-10-28:
- I started fresh, used etcher with image.
- On the variables.conf file i also set the date this time to export timezone="America/Los_Angeles"
- On my laptop, using terminal and "date" brings up correct date
- The imaged sd card put back into raspberry and power up, the log is the same error, but the time is still BST and not PDT.
- Correct local time is Sun 27 Oct 2019 19:22:35 PDT not Sun 22 as in the log

Sun 22 Sep 22:55:17 BST 2019 : Grabbing main setup file.
Sun 22 Sep 22:56:09 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:56:09 BST 2019 : Grabbing main setup file.
Sun 22 Sep 22:56:20 BST 2019 : get_script failed, retrying

magicalyak, 2019-10-28:
Does adding this get it to work?
export CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt

Jarery, 2019-10-28:
went through all the above again, with the new export curl command added, same result. time is still on BST instead of PDT, and script fails.

Sun 22 Sep 22:55:48 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:55:48 BST 2019 : Grabbing main setup file.
Sun 22 Sep 22:55:58 BST 2019 : get_script failed, retrying
Sun 22 Sep 22:56:12 BST 2019 : get_script failed, retrying
Sun 22 Sep 22:56:25 BST 2019 : get_script failed, retrying

magicalyak, 2019-10-28:
I think you’re having this issue https://github.com/marcone/teslausb/issues/211

closed 2019-11-12 by marcone

marcone, 2020-12-09:
I tried building a new image (since the current image doesn't boot from USB on the 8GB Pi4), and ran into the same problem with curl. Adding `export CURL_CA_BUNDLE=/etc/ssl/certs/ca-certificates.crt` to the profile as you described did fix curl for the interactive shell, but it didn't fix it for teslausb setup which is started from rc.local.
What did fix the issue was to `apt remove ca-certificates` and then `apt install ca-certificates`. Strangely, all the files that `dpkg-query -L ca-certificates` says are part of the packages were identical before and after the reinstall, so there must be some directory, symlink or config file somewhere that is created/updated as part of the reinstall that makes curl work.