marcone/teslausb #324: Wifi does not reconnect

clean-ev, 2019-10-23:
@marcone - one of the issue I keep getting into is where ZeroW will connect to WiFi first time when I power it up - it also creates AP successfully that I can connect. After being out and upon return, ZeroW only shows AP and does not connect to Wifi. I try to login thru AP and reboot and still no go. Any specifics I can check to see why this is happening. Today Iv have been bringing ZeroW inside and connect to power and then it starts to go thru archiving to Box.
Thanks in advance

marcone, 2019-10-23:
So you're saying that when you bring it inside it *does* connect to wifi?
Sounds like the Pi isn't able to reach wifi then when it's in the car.

clean-ev, 2019-10-23:
My car and Pi both connect perfectly fine to same 2.4Ghz network. 
When I first connect PiZeroW to USB on my car it connects as well as create AP. 
After a drive when I return PiZeroW will only show AP and not connect to wifi network
Car connects to WiFi right away.

marcone, 2019-10-24:
If wifi is configured correctly, and it sounds like it is since it connects sometimes, then this isn't really within the scope of teslausb. Teslausb doesn't do anything with wifi other than the initial configuration to set the ssid and password of your wifi. After that, it's up to the OS to connect and disconnect as needed.
There may be some logs somewhere you can look at (`logread` will show the system log, look for anything mentioning "wpa"), but I can't help with this.

closed 2019-10-24 by marcone

chilman408, 2019-10-31:
> My car and Pi both connect perfectly fine to same 2.4Ghz network.
> When I first connect PiZeroW to USB on my car it connects as well as create AP.
> After a drive when I return PiZeroW will only show AP and not connect to wifi network
> Car connects to WiFi right away.

See issue 334