marcone/teslausb #326: Unique ID (MAC Address) changes and renders DHCP Reservation Useless

SemoTech, 2019-10-26:
For my teslausb's WiFi connection I setup a DHCP reservation on the router, but each time the Pi 0W gets re-flashed, that "MAC Address" changes, so the reservation is useless.
  
I tried to set the IP to static by following the steps below, but it is NOT working and still uses DHCP.

```
0 - become root:  sudo -i 
1 - edit dhcp config file:     nano /etc/dhcpcd.conf
2 - add the lines:    

interface wlan0
static ip_address=192.168.0.220/24
static routers=192.168.0.1
static domain_name_servers=192.168.0.7 1.1.1.1

3 - save and reboot!
```

Can anyone advise how I can ideally ensure the Pi's Device Unique ID (MAC Address) is NOT changing, so the DHCP reservation holds even after re-flashes, or if there is a way to specify the desired IP, route and DNS in the config file?  I'm even OK with somehow properly manually editing the Pi 0W networking files to force it to use a fixed IP!  Thanks.

marcone, 2019-10-26:
> For my teslausb's WiFi connection I setup a DHCP reservation on the router, but each time the Pi 0W gets re-flashed, that "MAC Address" changes, so the reservation is useless.

That shouldn't be the case, since the Pi's MAC address is derived from the CPU serial number, which is unique and fixed for each Pi.
(specifically, the first 24 bits of the MAC address are 0xb827eb, and the last 24 bits of the MAC address are the last 24 bits of the CPU serial number XOR'd with 0x555555)

closed 2019-10-26 by marcone

SemoTech, 2019-10-26:
@marcone I would normally agree, however I am seeing this behavior in my DHCP server each time I re-flash the microSD card with the image.  The Pi's MAC address changes and a new IP is issued instead of the reserved one.  I have to manually copy the new Mac Address, and edit the registration with it, to get the Pi to latch to the registration.

Also, the Mac Address is always strangely very long, unlike any other device. Something like:
eb5c4a8400010001154711bbb827eb5c4a84

marcone, 2019-10-26:
I don't know what that is, but it's not a mac address.
It might be a DUID or similar, in which case it seems like this is something you can configure either on your dhcp server (to make it use the mac address instead) or on the Pi (by setting the ID to a specific value). See https://wiki.archlinux.org/index.php/dhcpcd#DHCP_Client_Identifier.

This seems to specifically address your situation: https://www.raspberrypi.org/forums/viewtopic.php?t=110331 and possibly this also: https://www.raspberrypi.org/forums/viewtopic.php?t=112134

SemoTech, 2019-10-26:
That looks promising indeed @marcone, told ya I was not crazy and the "MAC" was indeed changing after each re-image :-)  

Let me try these fixes and see if maybe we can include them into the next image so no more need to do it manually and everyone just gets a static regular MAC address.

SemoTech, 2019-10-26:
OK, so in /etc/dhcpcd.conf there is a section which turns on one or other of the options:  
clientid or duid

By default in @marcone 's Buster image for teslausb the "clientid" is active, and "duid" is commented out, so this SHOULD be using a regular MAC address.  Instead it is working as if "duid" is on since the "MAC Address" is a huge string that keeps changing.

Still unsure how to fix this....

SemoTech, 2019-10-26:
Also in /etc/dhcpcp.conf I tried to change:

# Generate SLAAC address using the Hardware Address of the interface
#slaac hwaddr
# OR generate Stable Private IPv6 Addresses based from the DUID
slaac private

TO:

# Generate SLAAC address using the Hardware Address of the interface
slaac hwaddr
# OR generate Stable Private IPv6 Addresses based from the DUID
#slaac private

And no effect after reboot, same huge MAC Address identifier... UGH!