marcone/teslausb #329: TeslaUSB appeared to die

bojennett, 2019-10-28:
I setup my Tesla to try and keep power to the device all the time, so I enabled Sentry Mode with the car in my garage.  I was just doing this as a test before deciding whether I wanted to go with the API mechanism.

I work from home, so didn't drive for a couple of days.  I had been ssh'd into the device from my desktop Mac just to see that the device was still alive.

Sometime on Saturday, the device appeared to die.  Connection was lost, and I couldn't SSH into the device anymore.  Thinking that maybe the car still decided to power down the device despite having Sentry mode on, I retrieved the device, brought it in the house, and powered it up locally.

And... nothing.  It didn't show up on the network.  Plugging it into my Mac, it didn't show up as a disk, either.

It was like the thing was just dead.

Anybody else have this issue - where the device just seems to die?

bojennett, 2019-10-28:
The green LED on the RPI0 was solid green after booting, for the record.

miles267, 2019-10-28:
I’ve had this happen recently. So I took the RP0 to my workbench and hooked up a monitor and keyboard, booted it, to discover it was booting to emergency mode. Why, I still don’t know. But I ended up reimaging it because I couldn’t get to a shell prompt. Didn’t know what else to do.

bojennett, 2019-10-28:
Got it.  This is... not ideal.  If the device can go "poof", can't really use it.  Worse than having a regular stick that got full.

marcone, 2019-10-29:
Could be the card crapped out. The car is always recording when it's on, so it would have been writing recordings to the car the entire time.

David-Kirchner, 2019-10-29:
I have yet to get one of a pair of new RP0's with new MicroSD cards with teslausb to work.  I tried 2 days ago just to format a SanDisk endurance 64GB with 2 drive letters (TeslaCam and MP3) and placed in USB adapter to Telsa Model3. This worked for one day. 
Day 2 my external memory was not found by my Tesla.  I tried moving between the different Tesla USB ports and even did a Steering wheel reboot and still did not get my external memory back online.  I read the post above about the card crapped out, and decided to check on my MicroSD.  My windows PC showed a message like 'Do you want to scan and FIX?' I let Window fix it, it was the TeslaCam drive letter that needed to be fixed, not the MP3.  Now I have my MicroSD back to working with my Tesla.
note:  when the TeslaCam drive was corrupt, the MP3 drive did not get recognized by my Tesla.
note: I still have not got the teslausb on RP0 working, I just decided to wait for 2019-09-22-teslausb-lite to be listed as stable with Buster and Tesla V10.

This may be old info, but a telsausb script error I ran into (not sure if it was image 2019-09-22 or 2019-06-15) on my 128GB was: 
./setup-teslausb
Updating package index files...
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.2 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Fetched 40.2 kB in 3s (12.7 kB/s)                                
Reading package lists... Done
Verifying that the requested configuration is valid...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   693  100   693    0     0   1646      0 --:--:-- --:--:-- --:--:--  1650
Downloaded /tmp/verify-configuration.sh ...
Verifying that there is sufficient space available on the MicroSD card...
STOP: The MicroSD card is too small.

Model: SD SH128 (sd/mmc)
Disk /dev/mmcblk0: 128GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      277MB   2215MB  1938MB  primary  ext4
 3      2215MB  128GB   126GB   primary  xfs
 4      128GB   128GB   105MB   primary  ext4

JanVanHyfte, 2019-11-01:
David,
I'm not an expert, but your partition 1 looks very small, and seems the only one FAT32 filesystem, what TeslaCam needs ...

marcone, 2019-11-01:
> STOP: The MicroSD card is too small.

@David-Kirchner I don't know which version of setup-teslausb you're running, but it doesn't appear to be mine :)
That line normally includes the available size, e.g. something like:
```
STOP: The MicroSD card is too small: 11811160064 bytes available.
```

David-Kirchner, 2019-11-02:
I SSH in and tried instructions from https://github.com/cimryan/teslausb.
wget https://raw.githubusercontent.com/cimryan/teslausb/master/setup/pi/setup-teslausb
chmod +x setup-teslausb
./setup-teslausb
Sorry, I used setup-teslausb code version found in cimryan tree and posted here.

**I reflashed tonight keeping notes.**
I flash 128GB with 2019-09-22-teslausb-lite

I copied my version of teslausb_setup_variables.conf to /boot/

[teslausb_setup_variables.128b.txt](https://github.com/marcone/teslausb/files/3799985/teslausb_setup_variables.128b.txt)

I place 128GB in RP0 and power on.
I watch on the HDMI monitor as the RP0 runs.
I see at least 3 start up screens - then the last goes by quick and I have no screen.
I saw two messages that said it was not available for another 20? some days

No HDMI video, but the WiFi is working, I can SSH.
There is a /boot/TESLAUSB_SETUP_FINISHED
and the /boot/teslausb-headless-setup.log looks clean.

On PC:
PWR Micro USB, does power up for WiFi.
USB Micro USB, only powers up for WiFi.  
PC Does not recognize.

another USB Cable. and JOY!!!  
now working! I got PI in my Tesla.

marcone, 2019-11-02:
@David-Kirchner I still don't understand what you actually did, since cimryan's version of teslausb won't even parse the config file you attached above, but if it works for you, great.

closed 2019-11-08 by marcone