marcone/teslausb #333: The /tmp/archive-rsync-cmd.log File Grows

DonC52, 2019-10-29:
The file above grows each time the archive is run. Should it be deleted, truncated or sent into the regular log file and then deleted?

I would hate to see the file just deleted as the data may be useful if something goes wrong.

DonC52, 2019-10-29:
In my fork, I sent the /tmp file to the regular log file and then deleted the /tmp file. That way it gets all of the file in case something goes wrong during the rsync like the power going off or driving away before it is finished.

DonC52, 2019-10-29:
The changes to archive-clips.sh (in cifs_archive and rsync_archive directories) are in my rsync branch. Seems to work fine and keeps the size of the file smaller.

marcone, 2019-10-30:
It should be deleted. I wouldn't want to put it in the regular log file because it's too verbose and doesn't really contribute much in most cases. I'd just delete it before starting the archive, so you can still see the most recent log if something went wrong.

DonC52, 2019-10-30:
I am not sure you don't want to delete it before the archive/ That would destroy why it failed in the first place.  The reason I put it into the log file (which is truncated all the time anyway) is so you can see why/where it failed. Booting restarts the archive process again so if the archive is outa space or the power goes out on the Pi it would be harder to figure out whats happening. It is less stuff in the log than the old method for sure.

When the rsync is done, I just cat /tmp/... >> "$LOG_FILE" so its not using much resources.

I can modify my mod if you still think just delete it before the rsync starts.

marcone, 2019-10-30:
how about adding "--ignore-missing-args" to the command line options, and then only adding the rsync log to the main log when there actually was an error?

DonC52, 2019-10-30:
Why add the "--ignore-missing-args" to the command line options.  This option suppresses errors if a source file is missing. The rsync command creates a file list to process when is starts but when it tries to transfer that file it is missing. That is a should not occur error for sure.  Why were you suggesting it?

Logging if there were an error, it would catch the out of space on the archive machine but you may not actually get an error if the power goes away while copying the files. It would show in the log file the rsync command creates. 

Most all of the log file on my machine is the linking for the snapshots. Adding the rsync log file will not add much more and the stats at the bottom are very informative.

DonC52, 2019-10-30:
I think my testing answered the first question on the "--ignore-missing-args" on the rsync command line. If either the SavedClips or SentryClips directory is missing, it will get an error. I added it to my rsync branch of my fork to test.

marcone, 2019-10-31:
I've just reverted the use of rsync for cifs archiving, because rsync takes too long to realize the server is no longer reachable.

marcone, 2019-10-31:
(closing since this is now no longer a relevant issue)

closed 2019-10-31 by marcone