marcone/teslausb #334: Add country to wpa_supplicant.conf

chilman408, 2019-10-31:
Hello,

Just reporting that I have a similar issue as Issue 324 which is closed.

I did a new setup with a RPi4 (used a Pi0W for a year).

I also saw the RPi4 would not connect from my M3 but would connect if I take it out and power it in my office.  

I added 'country=US' in /etc/wpa_supplicant/wpa_supplicant.conf

And now it connects from my car --- super weird solution but it works.

marcone, 2019-10-31:
Interesting. I never had any problems without it on the Pi4, but it does seem like setting country=US is the right/safe thing to do, if only to avoid using the disallowed channels. As an added bonus, it seems that specifying the country enables the 5GHz channels on the Pi4, since without it it always picks the 2.4GHz channels for me.

chilman408, 2019-10-31:
Yes, when I connect to the Pi's AP while it is in the car and use "sudo iwlist wlan0 scan", I can see that the network it is suppose to connect to is visible to the Pi but it wouldn't connect.  

When I then take it to my office to try to figure out the problem, everything works fine; it started archiving...super weird behavior with connection

marcone, 2019-11-03:
setup-teslaus now adds "country=US" to wpa_supplicant.conf if it's not already there.

closed 2019-11-03 by marcone