marcone/teslausb #335: Feature request: Allow to upload video only at specific location

ortodox, 2019-10-31:
I dont have wifi in my garage so i have bought my self a 4g wifi router that is allways connected via WIFI to RaspPi. Therefor i cannot use this solution because rasp is setup to upload videos when its reaching "home" wifi, but my home wifi is all the time since im driving with the router.
 - Request to develop a "check" on current location at the same place its checking the home wifi availibility. 
Using this maybe: https://developers.google.com/maps/documentation/geolocation/intro

marcone, 2019-11-01:
I doubt a geolocation API would work when you're always connected to the same wifi hotspot.
It might be possible to get the location from the car using the Tesla API, but I consider this to be a very niche feature and I have no interest in working on it.
PRs are welcome, but it might be more expedient to get a wifi extender instead.

closed 2019-11-01 by marcone

ortodox, 2019-11-04:
@marcone i understand, i really think this is not only issue for me but those who live in appartments with garage down below where there are no wifi. What i realise correct me if im wrong but if i have turned on "sentry mode" everywhere else but home, then it will only upload files when im home and i dont have to manipulate the code right?

marcone, 2019-11-04:
> if i have turned on "sentry mode" everywhere else but home, then it will only upload files when im home and i dont have to manipulate the code right?

I'm not sure what you mean by that. Whether Sentry mode is on or off plays no part in when teslausb will archive files.
Teslausb will archive files when it goes from "unable to connect to the archive server" to "able to connect to the archive server" (so you could turn your hotspot/router off and back on to trigger an archive cycle)