marcone/teslausb #336: Teslausb seems to work fine, but no CAM icon appears in mX

JanVanHyfte, 2019-11-01:
**I have been able to setup the TeslaUsb and it was working fine.** 
Recently, I tried the RSYNC update, but that didn't work out, and I wasn't able anymore to get my device properly working. Yesterday I noticed the code change was reverted by Marcone because of issues. So I reïnstalled.
But, I am not able anymore to have a full working RPO-TeslaUSB:
I can SSH, everything seems ok. The led flashes once on the road, and twice when in the perimeter of my network.
In the /mutable/archiveloop.log I see even music files being transferred.
But in the car, no CAM symbol. When I plug in another 'plain' USB disk, the CAM symbol appears.
When I plug in the RPO in a WIN10, I do not see a USB-disk appearing.
Is my RPO's USB port dead? No, when I plug in the RPO without MicroSD card in my WIN10, I can see the RPO in the Device Manager as 'BCM2708' ...
[diagnostics.txt](https://github.com/marcone/teslausb/files/3797426/diagnostics.txt)
[archiveloop.log](https://github.com/marcone/teslausb/files/3797427/archiveloop.log)
HELP!!!

marcone, 2019-11-01:
Are you sure your cable is alright? The diagnostics show the music and cam drives being exported, but nothing appears to be connected to the Pi. That's usually an indication that a charge-only cable is being used, or that the cable is plugged into the Pi's power port instead of the USB port.

JanVanHyfte, 2019-11-01:
Marcone, it is a new data cable, connected to the USB Data slot.. If I connect the RPO withhout MicroSD to my Windows10, it is recognized in the device Manager ... So the cable is working as a data cable...

JanVanHyfte, 2019-11-02:
Marcone,
thank you for you confirming the logs look ok, I decided to purchase a new RPO, and after inserting MicroSD and Cable, It worked ;-)
Now I only need to add wondershaper (HINT!) to limt bandwidth usage, and ready to rumble !
Thank you !!!

closed 2019-11-02 by JanVanHyfte