marcone/teslausb #337: Disabling snapshots does not restore performance

jbienz, 2019-11-01:
## Background
Snapshots can be disabled using either of the following methods:

1. Adding `export SNAPSHOTS_ENABLED=false` to **teslausb_setup_variables.conf** during setup.
2. Setting `SNAPSHOTS_ENABLED=false` in **/root/teslausb.conf**.

## Snapshot Remnants
If snapshots are enabled at any point and then later disabled (through **/root/teslausb.conf**), some snapshot related performance penalties will still be in effect.

For example:

- Any snapshots which had already been created will still exist in `/backingfiles/snapshots`. (These can be quite large.)
- Snapshots are still mounted during the archive loop. (Log reports `mounted all snapshots`. This is because in `archiveloop`, the call to `mount_all_snapshots` comes before the check to see if snapshots are enabled.)
- _Many_ virtual files will still remain in `/backingfiles/TeslaCam/*` which are symbolic links to files in `/backingfiles/snapshots/snapshot-####/*`.

## Cleaning Up
I would like to see the code detect when snapshots were disabled after previously being enabled. When this is detected, I'd like to see it perform cleanup. Here are some potential things it should do:

1. In `/backingfiles/TeslaCam` and subfolders, delete all files which are symbolic links.
2. In the `/backingfiles/snapshots` folder, delete all files and subfolders.

I also recommend considering if the call to `mount_all_snapshots` should be moved inside the `if [ "${SNAPSHOTS_ENABLED:-true}"` check.

joetomasone, 2019-11-01:
Because I am somewhat unclear: What exactly is the purpose of the snapshots?

What are the ramifications of disabling them?

marcone, 2019-11-01:
Nobody ever claimed (or has evidence) that disabling snapshots "restores performance", so the title of this issue is a bit presumptuous.
You are welcome to submit a PR that cleans up leftover snapshots after disabling them, but since I have no interest in teslausb without snaphots, this is not something I'm interested in working on myself.

closed 2019-11-01 by marcone

jbienz, 2019-11-01:
I personally am claiming that disabling snapshots restores performance. My evidence is in the fact that I have never received any 'Too Slow' errors while snapshots are disabled and I have always received 'Too Slow' errors when snapshots are enabled.

You may dislike my evidence, or you may claim it to be circumstantial, but your assertion that no one is making this claim or no one has any evidence is a false assertion.

I see that you have closed this issue because you don't believe it has value and don't want to work on it. It's perfectly fine if you don't want to work on it. But if other users report that disabling snapshots allows them to at least use their drive, I'd like to request that you re-open this issue.

jbienz, 2019-11-01:
I also changed the title, since that appears to be a point of contention for this issue.

marcone, 2019-11-02:
People are having the "usb too slow" issue with regular USB drives where teslausb isn't even involved, while others have snapshots enabled without issue, myself included. So in my opinion you have anecdotal evidence at best.
And then based on what you said [here](https://github.com/marcone/teslausb/issues/239#issuecomment-548897604) and above, disabling snapshots alone isn't even enough to "restore performance", but you also to have to remove all existing snapshot files, which seems even stranger.
In any case, I don't think reopening this issue would serve a purpose. All of the various things that people have tried are already listed over on https://github.com/marcone/teslausb/issues/239, including disabling snapshots.

SaccoNicola, 2019-11-23:
I've tried many times with and without snapshots and can confirm that disabling snapshots does work without errors. When it's enabled, I get the drive slow message. There must be something in the making snapshots that interferes with the drive writing that the Tesla is doing.

joetomasone, 2019-11-23:
Taking too much CPU and impacting the overall write speed, maybe?

jbienz, 2019-11-27:
If others also verify that snapshots on = too slow error and snapshots off = functional, I'd like this issue to be reopened.

githubsean, 2019-11-27:
For what it's worth, I've been running with snapshots enabled for some weeks now without issue.
All I have done is run Buster (very latest image), I expanded the root partition so there was room to perform updates, which I did. And that's pretty much it I think.
I have a model 3 with V10 (2019.36.2.2)

SaccoNicola, 2019-11-27:
I haven't tried with expanded root. Without restarting from scratch and enabling in setup conf, can someone shed light on how to do it?

DMBlakeley, 2019-11-27:
Similar experience to @githubsean. A month now without issues. SanDisk 64GB Ultra Plus, Buster prebuilt image, 16GB capsize, 20GB musicsize, no snapshots, CIFS to Mac on home network, and Model 3 with V10 (2019.36.2.2). I did not expand the root partition but was able to keep up with the updates.

SaccoNicola, 2019-11-27:
@DMBlakeley , you mean similar experience to @jbienz or myself - based on the no snapshots comment and no expansion to root partition. Right?

DMBlakeley, 2019-11-27:
Similar experience to @githubsean (post just prior to yours). My understanding is that you only really need root expansion if you are adding additional functionality beyond what is provided by the @marcone build. The minor updates that have been posted by @marcone over the last month have not required additional space.

SaccoNicola, 2019-11-27:
@DMBlakeley, we were saying that with no snapshots, it's working fine. I think what @githubsean is doing is using the expanded root with snapshots enabled to get it to work.

DMBlakeley, 2019-11-27:
@githubsean stated "I expanded the root partition so there was room to perform updates". Don't believe he was stating the expansion was connected to snapshots. My understanding is that space available for snapshots is = card size - image size (including expansion) - cam size - music size.

githubsean, 2019-11-27:
I expanded the root partition because when performing all updates - debian and pi firmware - it ran out of space.

The snapshots thing is separate, I just enabled it after @marcone said than he has snapshots enabled with no problems. And indeed, I E had no problems.

Now I have had problems in the past, but I can't say whether it was the older builds, older SD cards, older Tesla firmware or something else.

There have been some recent changes where old snapshots are better managed (some conversation about XFS extents) which I think then also allows the underlying file systems to properly us the trim facility of the flash hardware.)

I had actually parked the project for about a month, but then about three weeks ago I put it back in the car, and I've had no problems.