marcone/teslausb #340: Beta usage report

natrash, 2019-11-05:
Hi,

Here is my experience with Beta version

- Drive often dies. Meaning it just stops. can happen anytime like 2 minutes after boot and creating virtual drives, or 2 hours when driving the icon disappear... Unplug usb and restart needed.
I was watching a tail -f /mutable/archiveloop.log at the moving clip part of the script when it suddenly stopped (no more LED blink).

After many times, I aslo noticed the drive stops when moving clips to the NAS. IF this happens, after rebooting (usb cycle) the drive will fail copying clips regularly. so drive gets full...

I plugged it into the MAC (because on the PC, when the drives CAM and MUSIC come up it ask for format the drive.
I ban it from Wifi (so the sync script do not start), and I manually move the non-fucked-up clips to the NAS, Gb by Gb, becaus if I move everything at once, the drive dies.

I've launched a setup-teslausb selfupdate etc...after making changes in the config, but it happened the drive died in the middle of the update. After reboot the setup-teslausb.sh was just 0 Bytes...
I had to reflash.

Now I still have issues. Find attached the diagnostic.txt, I hope this will help.
[diagnostics.txt](https://github.com/marcone/teslausb/files/3808422/diagnostics.txt)

Thanks.

chilman408, 2019-11-05:
Not to provide input to the diagnostics, but I honestly think doing Snapshot and Music partitions (and monitoring it via SSH) might be choking the Pi Zero out.

Have you tried doing the simplest form - just archiving (disable snapshot and music partitiions, and monitor the archiving from the destination)?

natrash, 2019-11-05:
I removed the snapshots

```
Tue  5 Nov 21:43:54 CET 2019: Finished fsck on /mnt/cam.
Tue  5 Nov 21:43:54 CET 2019: There are 1 event folder(s) with 4 file(s) to move.
Tue  5 Nov 21:43:54 CET 2019: Starting recording archiving: 1 event folder(s) with 4 file(s)
Tue  5 Nov 21:44:00 CET 2019: Moving clips to archive...
Tue  5 Nov 21:44:00 CET 2019: /mnt/cam/TeslaCam/SavedClips does not exist, skipping
Tue  5 Nov 21:44:00 CET 2019: Creating output directory 'SentryClips/2019-11-05_17-36-17'
Tue  5 Nov 21:44:00 CET 2019: Moving 'SentryClips/2019-11-05_17-36-17/2019-11-05_17-36-17-back.mp4'
Tue  5 Nov 21:44:07 CET 2019: Moved 'SentryClips/2019-11-05_17-36-17/2019-11-05_17-36-17-back.mp4'
Tue  5 Nov 21:44:07 CET 2019: Moving 'SentryClips/2019-11-05_17-36-17/2019-11-05_17-36-17-front.mp4'
Connection reset by 192.168.8.98 port 22
```

I don't know where to look to see why the PI crashes and goes off...
Thanks for your help.

marcone, 2019-11-06:
The only thing that stands out to me in the diagnostics is that it apparently frequently restarts while in the middle of doing something.
It looks to me like you have a bad Pi, or it's not getting reliable power.

closed 2019-11-06 by marcone