marcone/teslausb #343: Snapshot issues

emanspeaks, 2019-11-07:
I have been following the discussion in #239 where many people have recommended not using snapshots.  However, for those of us who want to use snapshots, I would like to help get it working in a stable fashion.  While I have been able to mitigate the slow USB messages with some of the fixes mentioned in that issue, I do see spurious failures related to snapshots where the device disconnects from the car.  I never had the Pi working in a stable state before now, so it is unclear if these new issues related to snapshots are the result of having the two lines below in my `/etc/sysctl.conf`:
```
vm.dirty_background_bytes = 2048
vm.dirty_ratio = 80
```
or if I would have seen this snapshot failure before the config change.

My device is currently in such a stuck state seemingly following an attempted snapshot.  I see the following in my `/mutable/archiveloop.log`:
```
Wed  6 Nov 08:34:15 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:16 CST 2019: Retrying...
Wed  6 Nov 08:34:16 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:17 CST 2019: Retrying...
Wed  6 Nov 08:34:17 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:18 CST 2019: Retrying...
Wed  6 Nov 08:34:18 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:20 CST 2019: Retrying...
Wed  6 Nov 08:34:20 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:21 CST 2019: Retrying...
Wed  6 Nov 08:34:21 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:22 CST 2019: Retrying...
Wed  6 Nov 08:34:22 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:23 CST 2019: Retrying...
Wed  6 Nov 08:34:23 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:24 CST 2019: Retrying...
Wed  6 Nov 08:34:24 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:25 CST 2019: Retrying...
Wed  6 Nov 08:34:25 CST 2019: Sleeping before retry...
Wed  6 Nov 08:34:26 CST 2019: Retrying...
Wed  6 Nov 08:34:26 CST 2019: Attempts exhausted.
Wed  6 Nov 08:34:26 CST 2019: Archive is unreachable.
Wed  6 Nov 08:34:26 CST 2019: Waiting for archive to be reachable...
Wed  6 Nov 09:06:54 CST 2019: low space, deleting /backingfiles/snapshots/snap-000015
Wed  6 Nov 09:06:56 CST 2019: taking snapshot of cam disk: /backingfiles/snapshots/snap-000031/snap.bin
Wed  6 Nov 09:07:00 CST 2019: took snapshot
Wed  6 Nov 09:07:03 CST 2019: comparing /backingfiles/snapshots/snap-000030/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Wed  6 Nov 09:07:03 CST 2019: making links for /backingfiles/snapshots/newsnap/mnt, retargeted to /backingfiles/snapshots/snap-000031/mnt
Wed  6 Nov 09:07:05 CST 2019: made all links for /backingfiles/snapshots/newsnap/mnt
Wed  6 Nov 10:05:25 CST 2019: low space, deleting /backingfiles/snapshots/snap-000016
Wed  6 Nov 10:05:27 CST 2019: taking snapshot of cam disk: /backingfiles/snapshots/snap-000032/snap.bin
Wed  6 Nov 10:05:32 CST 2019: took snapshot
Wed  6 Nov 10:05:34 CST 2019: comparing /backingfiles/snapshots/snap-000031/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Wed  6 Nov 10:05:34 CST 2019: making links for /backingfiles/snapshots/newsnap/mnt, retargeted to /backingfiles/snapshots/snap-000032/mnt
Wed  6 Nov 10:05:37 CST 2019: made all links for /backingfiles/snapshots/newsnap/mnt
Wed  6 Nov 11:03:58 CST 2019: taking snapshot of cam disk: /backingfiles/snapshots/snap-000033/snap.bin
Wed  6 Nov 11:04:02 CST 2019: took snapshot
Wed  6 Nov 11:04:05 CST 2019: comparing /backingfiles/snapshots/snap-000032/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Wed  6 Nov 11:04:05 CST 2019: new snapshot is identical to previous one, discarding
Wed  6 Nov 12:02:27 CST 2019: taking snapshot of cam disk: /backingfiles/snapshots/snap-000033/snap.bin
Wed  6 Nov 20:11:57 CST 2019: Archive is reachable.
Wed  6 Nov 20:11:57 CST 2019: Waiting for time to be set by ntpd...
Wed  6 Nov 20:11:58 CST 2019: Time now set
```

Note the jump in the timestamps towards the end.  The drive had been working flawlessly for about four days (no USB slow messages, no dropouts in connectivity such as this) until this afternoon around the time of the final "taking snapshot" message before the long gap.  I arrived home about two hours ago, and I just now pulled the log message where the last time stamp is when I arrived home.

If I try to go into `/mnt/cam`, the directory is empty.  I believe this is because the disk is not currently mounted, but I have not confirmed or tried to remount it.  I suspect this is also why the Pi has not uploaded data to the archive server, despite obviously being able to connect to the server.

I also tried running a `/root/bin/setup-teslausb diagnose` and the script seems hung somewhere.  I would guess this is also because of the odd state of `/mnt/cam`.

I'm hesitant to actually reboot the Pi at this point so that I can continue to get troubleshooting data while it is in this state.  As such, any advice on where to start looking to help debug this would be appreciated.  

(I might have been able to figure out more details/chase some leads on my own, but I am out of time for tonight and just wanted to get something on here to start the discussion at least.)

Thanks!

marcone, 2019-11-07:
The jump in timestamps is consistent with a device that booted up without a network connection (and thus couldn't set its clock) and then connects to wifi and sets the time via ntp.

`/mnt/cam` not being mounted is quite normal: it is only mounted during an archive operation.

You said `/root/bin/setup-teslausb diagnose` hangs, but didn't say at what point. It might help if you at least post the part that it did print.

emanspeaks, 2019-11-07:
Sorry, to clarify, the log is just an excerpt of a much longer log that runs for quite a few days when it was working successfully. I started from the point when I left the house this morning. NTP had already set the time. The jump in time stamps is legitimately a lack of activity during that time until I got home. 

For the diagnose, it didn’t print anything after loading variables from my config file. Just a linking cursor for 5 minutes before I ctrl+c’d it

emanspeaks, 2019-11-07:
To further clarify, I keep the Pi powered via a Zendure X6 so it has not rebooted for about five days.

marcone, 2019-11-07:
"setup-teslausb diagnose" should only take a few seconds to complete. If it doesn't the problem is likely not related to teslausb. You may have a bad Pi or sd card.

marcone, 2019-11-07:
(you could try if `dmesg`shows anything I/O or other errors)

emanspeaks, 2019-11-07:
from `dmesg`:

```
[92266.461006] dwc2 20980000.usb: new device is full-speed
[92266.902432] dwc2 20980000.usb: new device is full-speed
[92266.987429] dwc2 20980000.usb: new device is full-speed
[92267.204293] dwc2 20980000.usb: new device is high-speed
[92267.267331] dwc2 20980000.usb: new device is high-speed
[92267.319266] dwc2 20980000.usb: new address 31
[92267.333952] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
[93846.171244] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[93846.175384] pgd = 60348da1
[93846.177561] [00000000] *pgd=00000000
[93846.179665] Internal error: Oops: 17 [#1] ARM
[93846.181761] Modules linked in: g_mass_storage usb_f_mass_storage libcomposite arc4 ecb md4 md5 sha512_generic hmac nls_utf8 cifs ccm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic xfs brcmfmac brcmutil sha256_generic cfg80211 raspberrypi_hwmon hwmon rfkill snd_bcm2835(C) snd_pcm snd_timer bcm2835_codec(C) snd bcm2835_v4l2(C) v4l2_mem2mem v4l2_common bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq uio fixed dwc2 udc_core ip_tables x_tables ipv6 [last unloaded: libcomposite]
[93846.199038] CPU: 0 PID: 25493 Comm: file-storage Tainted: G         C        4.19.66+ #1253
[93846.203510] Hardware name: BCM2835
[93846.207283] PC is at xfs_trans_free_items+0x38/0x15c [xfs]
[93846.211246] LR is at xfs_log_commit_cil+0x4a4/0x654 [xfs]
[93846.213649] pc : [<bf55525c>]    lr : [<bf55b980>]    psr: 00000013
[93846.215930] sp : c5557b58  ip : c5557b90  fp : c5557b8c
[93846.218150] r10: 00000000  r9 : 00000000  r8 : c67bb790
[93846.220304] r7 : 00000000  r6 : 0000045b  r5 : bf55b980  r4 : 00000000
[93846.222420] r3 : 00000000  r2 : 0000045b  r1 : 00000000  r0 : c67bb640
[93846.224536] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[93846.226721] Control: 00c5387d  Table: 17274008  DAC: 00000055
[93846.228930] Process file-storage (pid: 25493, stack limit = 0x2a6822fe)
[93846.231145] Stack: (0xc5557b58 to 0xc5558000)
[93846.233294] 7b40:                                                       00000000 00000000
[93846.237366] 7b60: c5557b8c 00000000 c67bb640 d766a9c0 da62fa00 0000045b 00000000 00000000
[93846.241363] 7b80: c5557bfc c5557b90 bf55b980 bf555230 00000000 bf55a558 00000000 00000000
[93846.245358] 7ba0: 00000000 c67bb640 d766a9c0 00000000 c5557c10 d75a1800 d766a9cc d766a9c0
[93846.249419] 7bc0: 00000000 00000000 00000000 36d4373f c67bb640 00000000 00000000 bf555924
[93846.253676] 7be0: d75a1800 00000000 c09e9028 00000001 c5557c4c c5557c00 bf5555cc bf55b4e8
[93846.258128] 7c00: c67bb640 d75a1800 bf542c68 c67bb640 0000045b 00000000 c5557c3c 36d4373f
[93846.262740] 7c20: 00000006 00004000 c09e9028 00000006 00000000 d5cf2fc0 d5cf3090 c09e9028
[93846.267624] 7c40: c5557c5c c5557c50 bf555924 bf55538c c5557cac c5557c60 bf542d48 bf555914
[93846.272639] 7c60: 00000000 c5557c78 5dc306a6 c5557cc0 d75a1800 00000006 c67bb640 36d4373f
[93846.277748] 7c80: 1b2dded2 d5cf3090 c09e9028 00000006 00000000 d9317640 c5557cc0 c5557df8
[93846.283063] 7ca0: c5557cfc c5557cb0 c01902bc bf542ba4 5dc306a6 00000000 1b2dded2 c5557df8
[93846.288457] 7cc0: 5dc306a6 00000000 1b2dded2 c5557df8 c5557d34 36d4373f d5cf3090 c5557e10
[93846.293847] 7ce0: c5557d7c 00000000 ba65f000 00000006 c5557d4c c5557d00 bf535d64 c01901c8
[93846.299239] 7d00: 00004000 c06d2db8 c5557d2c c5557d18 c06d2db8 d9317640 00004000 d5cf2fc0
[93846.304632] 7d20: c5557d4c c5557e10 c5557df8 d5cf2fc0 c5557df8 d5cf3090 00000000 00000000
[93846.310022] 7d40: c5557dcc c5557d50 bf536144 bf535c60 bf0b18b0 c005222c c09f21c0 c6464630
[93846.315416] 7d60: 00000001 00000000 d72681a0 c09ff900 c004af68 c6464600 c06d064c 00000001
[93846.320802] 7d80: c5557dac c5557d90 c00d415c c00d525c d72681a0 da59c600 00000001 36d4373f
[93846.326187] 7da0: c5557de4 c5557e10 c5557df8 aee9c000 00000000 c5557ee0 00000000 d77fbc00
[93846.331575] 7dc0: c5557de4 c5557dd0 bf53649c bf5360e0 d9317640 c09e9028 c5557e5c c5557de8
[93846.336963] 7de0: c0174d30 bf536408 00004000 c09ff900 c64ac000 00004000 00000003 00000000
[93846.342347] 7e00: 00004000 c5557df0 00000001 00000000 d9317640 00000000 aee9c000 00000000
[93846.347733] 7e20: 00000000 00000000 00000000 00000000 c5557e5c 36d4373f 00004000 d9317640
[93846.353120] 7e40: c64ac000 c5557ee0 aee9c000 00000000 c5557e8c c5557e60 c0174f7c c0174c34
[93846.358505] 7e60: 00000000 c6464600 00000000 00000000 00004000 00000006 aee9c000 00000000
[93846.363892] 7e80: c5557ea4 c5557e90 c01750c0 c0174ed8 d7e4f0d8 cb239a00 c5557f1c c5557ea8
[93846.369281] 7ea0: bf0fe028 c0175094 c5557ec0 bf0fd7cc c06ce1f4 bf101964 0000006f c5556000
[93846.374671] 7ec0: 00000000 00006000 ba65f000 00000006 aeea2000 00000000 00000000 00000000
[93846.380065] 7ee0: aee9c000 00000000 c64ac000 36d4373f c5557f1c d77fbc00 00000000 d7e4f0d8
[93846.385457] 7f00: c64ac000 d77fbc00 43425355 d77fbc20 c5557f7c c5557f20 bf0ff504 bf0fdd60
[93846.390847] 7f20: bf101bd0 bf101bac c5557f54 c5557f38 00000000 00062900 60000013 c5556000
[93846.396241] 7f40: d7e4fd40 00000040 c5557f7c 36d4373f c0042864 d7e4fd40 d73b6820 00000000
[93846.401636] 7f60: d77fbc00 bf0fef3c d7e4fd58 c54f5910 c5557fac c5557f80 c0042ff0 bf0fef48
[93846.407026] 7f80: ffffffff d73b6820 c0042ed4 00000000 00000000 00000000 00000000 00000000
[93846.412420] 7fa0: 00000000 c5557fb0 c00090ac c0042ee0 00000000 00000000 00000000 00000000
[... cut ...]
[93846.523492] [<c0042ff0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[93846.526209] Exception stack(0xc5557fb0 to 0xc5557ff8)
[93846.528865] 7fa0:                                     00000000 00000000 00000000 00000000
[93846.534024] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[93846.539125] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[93846.541729] Code: e5db9004 e320f000 e5b840a8 e1540008 (e5945000)
[93846.544444] ---[ end trace 2a3a37f199fce5ca ]---
```

marcone, 2019-11-07:
well there you go: the kernel is crashing. Looks like it's crashing in filesystem code, probably due to mishandled filesystem corruption. You could try a newer kernel or fixing the filesystem, but it's probably easiest to reinstall from scratch.

closed 2019-11-07 by marcone