marcone/teslausb #344: Can't get this thing working!

bojennett, 2019-11-07:
I get this message in **teslausb-headless-setup.log**

```
Thu  7 Nov 18:10:46 GMT 2019 : rebooting for changes to take effect
Thu  7 Nov 18:11:12 GMT 2019 : Detecting whether to update wpa_supplicant.conf
Thu  7 Nov 18:11:13 GMT 2019 : Starting setup.
Thu  7 Nov 18:11:13 GMT 2019 : reading config from /root/teslausb_setup_variables.conf

```
And then nothing else.  It clearly is getting somewhat far, because it manages to connect to my Wifi network so that I can get in, but...

Nothing shows up as a USB drive (I'm powering it from the USB port).  There is also no /backingfiles directory like there was on previous attempts to build it.

Here is my config file (HIDDEN means I edited the values posted here as they are obviously private):

```
# Default variables for CIFS (Windows/Mac file sharing) setup
export ARCHIVE_SYSTEM=cifs
export archiveserver=<HIDDEN>
export sharename=<HIDDEN>
export shareuser=<HIDDEN>
export sharepassword=<HIDDEN>
# export cifs_version="3.0"
export camsize=100%

# Wifi setup information. Note that Raspberry Pi Zero W only supports 2.4 GHz wifi
export SSID='<HIDDEN>'
export WIFIPASS='<HIDDEN>'

# If doing a headless (i.e. automatic) setup
export HEADLESS_SETUP=true

export SAMBA_ENABLED=true
export SAMBA_GUEST=true

# uncomment to enable teslausb to act as a wifi access point with the given
# SSID, so you can access it while on the road.
export AP_SSID='<HIDDEN>'
# Change this! The setup script will not accept the default 'password'
# Also note that the wifi password must be at least 8 characters.
export AP_PASS='<HIDDEN>'

export TESLAUSB_HOSTNAME=teslausb-btbs

export tesla_email=<HIDDEN>
export tesla_password=<HIDDEN>
export tesla_vin=<HIDDEN>

```

bojennett, 2019-11-07:
The green LED is solid green.  I can SSH into it just fine and navigate around.

marcone, 2019-11-07:
run `sudo /etc/rc.local` and it will probably show where it goes wrong. I suspect a typo in the config file.

bojennett, 2019-11-07:
My IP address is 169.254.227.5 10.0.1.20 
Detecting whether to update wpa_supplicant.conf
Setup doesn't seem to have completed, there is no /root/bin/archiveloop.

bojennett, 2019-11-07:
I have now made sure that `export HEADLESS_SETUP=true` and am re-running the Teslausb-setup script.  it is now downloading stuff

bojennett, 2019-11-07:
ok, that seemed to do it.  thanks!

closed 2019-11-07 by bojennett