marcone/teslausb #346: cam and music not mounted from host (either my pc or my tesla)

stepir, 2019-11-08:
Hi,

I have reflashed my PI with the latest stable version of teslausb.  The setup went smooth:

pi@eldiablousb:~ $ tail -20 /boot/teslausb-headless-setup.log
Thu  7 Nov 22:52:46 CET 2019 : verify-and-configure-archive: The archive share is mountable using: mount -t cifs '//qnapstepir.fritz.box/Recordings/cc-eldiablo/Download' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=3.0,'
Thu  7 Nov 22:52:46 CET 2019 : verify-and-configure-archive: Configuring the archive...
Thu  7 Nov 22:52:46 CET 2019 : verify-and-configure-archive: Configured the archive.
Thu  7 Nov 22:52:46 CET 2019 : configure: Skipping rc.local installation
Thu  7 Nov 22:52:46 CET 2019 : curl -o /tmp/configure-samba.sh https://raw.githubusercontent.com/marcone/teslausb/887868f26561ef5cbd4d801e609282683dac7421/setup/pi/configure-samba.sh
Thu  7 Nov 22:52:47 CET 2019 : Downloaded /tmp/configure-samba.sh ...
Thu  7 Nov 22:52:47 CET 2019 : configure-samba: Installing samba and dependencies...
Thu  7 Nov 22:54:50 CET 2019 : configure-samba: Done.
Thu  7 Nov 22:54:50 CET 2019 : curl -o /tmp/configure-ap.sh https://raw.githubusercontent.com/marcone/teslausb/887868f26561ef5cbd4d801e609282683dac7421/setup/pi/configure-ap.sh
Thu  7 Nov 22:54:53 CET 2019 : Downloaded /tmp/configure-ap.sh ...
Thu  7 Nov 22:54:53 CET 2019 : configure-ap: installing dnsmasq and hostapd
Thu  7 Nov 22:55:33 CET 2019 : configure-ap: configuring AP 'El_Diablo' with IP 192.168.66.1
Thu  7 Nov 22:55:33 CET 2019 : make-root-fs-readonly: start
Thu  7 Nov 22:55:34 CET 2019 : make-root-fs-readonly: Removing unwanted packages...
Thu  7 Nov 22:56:29 CET 2019 : make-root-fs-readonly: Installing ntp and busybox-syslogd...
Thu  7 Nov 22:57:24 CET 2019 : make-root-fs-readonly: Configuring system...
Thu  7 Nov 22:57:24 CET 2019 : make-root-fs-readonly: Moving fake-hwclock data
Thu  7 Nov 22:57:24 CET 2019 : make-root-fs-readonly: done
Thu  7 Nov 22:57:24 CET 2019 : Skipping package upgrade.
Thu  7 Nov 22:57:24 CET 2019 : All done.

But the host is not mounting the folder/music drives:

pi@eldiablousb:~ $ tail -50 /mutable/archiveloop.log
Fri  8 Nov 13:34:52 CET 2019: Time still not set, attempting to force it
Fri  8 Nov 13:35:03 CET 2019: Ensuring music backing file is mounted...
Fri  8 Nov 13:35:03 CET 2019: Mounting /mnt/music...
Fri  8 Nov 13:35:03 CET 2019: Mounted /mnt/music.
Fri  8 Nov 13:35:03 CET 2019: Ensured music drive is mounted.
Fri  8 Nov 13:35:03 CET 2019: Running fsck on /mnt/music...
Fri  8 Nov 13:35:03 CET 2019: | fsck from util-linux 2.29.2
Fri  8 Nov 13:35:04 CET 2019: | fsck.fat 4.1 (2017-01-24)
Fri  8 Nov 13:35:04 CET 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Fri  8 Nov 13:35:04 CET 2019: |  Automatically removing dirty bit.
Fri  8 Nov 13:35:04 CET 2019: | Performing changes.
Fri  8 Nov 13:35:04 CET 2019: | /dev/loop1: 1 files, 1/523006 clusters
Fri  8 Nov 13:35:04 CET 2019: Finished fsck on /mnt/music.
Fri  8 Nov 13:35:04 CET 2019: Unmounting /mnt/music...
Fri  8 Nov 13:35:04 CET 2019: Unmounted /mnt/music.
Fri  8 Nov 13:35:04 CET 2019: Archiving...
Fri  8 Nov 13:35:04 CET 2019: Disconnecting usb from host...
Fri  8 Nov 13:35:04 CET 2019: Disconnected usb from host.
Fri  8 Nov 13:35:04 CET 2019: Ensuring cam archive is mounted...
Fri  8 Nov 13:35:04 CET 2019: Mounting /mnt/archive...
Fri  8 Nov 13:35:05 CET 2019: Mounted /mnt/archive.
Fri  8 Nov 13:35:05 CET 2019: Ensured cam archive is mounted.
Fri  8 Nov 13:35:05 CET 2019: Checking saved folder count...
Fri  8 Nov 13:35:05 CET 2019: Ensuring cam file is mounted...
Fri  8 Nov 13:35:05 CET 2019: Mounting /mnt/cam...
Fri  8 Nov 13:35:05 CET 2019: Mounted /mnt/cam.
Fri  8 Nov 13:35:05 CET 2019: Ensured cam file is mounted.
Fri  8 Nov 13:35:05 CET 2019: Running fsck on /mnt/cam...
Fri  8 Nov 13:35:05 CET 2019: | fsck from util-linux 2.29.2
Fri  8 Nov 13:35:07 CET 2019: | fsck.fat 4.1 (2017-01-24)
Fri  8 Nov 13:35:07 CET 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Fri  8 Nov 13:35:07 CET 2019: |  Automatically removing dirty bit.
Fri  8 Nov 13:35:07 CET 2019: | Performing changes.
Fri  8 Nov 13:35:07 CET 2019: | /dev/loop1: 2 files, 2/2464576 clusters
Fri  8 Nov 13:35:07 CET 2019: Finished fsck on /mnt/cam.
Fri  8 Nov 13:35:08 CET 2019: There are 0 event folder(s) with 0 file(s) to move.
Fri  8 Nov 13:35:08 CET 2019: Unmounting /mnt/cam...
Fri  8 Nov 13:35:08 CET 2019: Unmounted /mnt/cam.
Fri  8 Nov 13:35:08 CET 2019: Finished archiving.
Fri  8 Nov 13:35:08 CET 2019: Music archive not configured or unreachable
Fri  8 Nov 13:35:08 CET 2019: unmounting /mnt/archive
Fri  8 Nov 13:35:08 CET 2019: Connecting usb to host...
Fri  8 Nov 13:35:08 CET 2019: Connected usb to host.
Fri  8 Nov 13:35:08 CET 2019: Waiting for archive to be unreachable...
Fri  8 Nov 13:36:20 CET 2019: Sleeping before retry...
Fri  8 Nov 13:36:21 CET 2019: Retrying...
Fri  8 Nov 13:36:23 CET 2019: Sleeping before retry...
Fri  8 Nov 13:36:24 CET 2019: Retrying...
Fri  8 Nov 13:39:18 CET 2019: Sleeping before retry...
Fri  8 Nov 13:39:19 CET 2019: Retrying...

What can I look further to diagnose?

Thanks

stepir, 2019-11-08:
the same is happening with the buster image!!! why!!!!???

marcone, 2019-11-08:
See https://github.com/marcone/teslausb/issues/35

closed 2019-11-08 by marcone

j450h1, 2020-01-09:
I'm having the same issue with the buster image as well. Here is my diagnostics info as per instructions on #35: 

https://pastebin.com/u3J7PN7s

j450h1, 2020-01-09:
I think it might have something to do with **music archive not configured or unreachable as per below**

```root@teslausb:~# tail -50 /mutable/archiveloop.log```

Wed  8 Jan 20:31:37 PST 2020: Music archive not configured or unreachable
Wed  8 Jan 20:31:37 PST 2020: unmounting /mnt/archive
Wed  8 Jan 20:31:37 PST 2020: Connecting usb to host...
Wed  8 Jan 20:31:37 PST 2020: Connected usb to host.
Wed  8 Jan 20:31:37 PST 2020: Waiting for archive to be unreachable...
Wed  8 Jan 20:36:04 PST 2020: Sleeping before retry...
Wed  8 Jan 20:36:05 PST 2020: Retrying...
==============================================
Wed  8 Jan 20:47:07 PST 2020: Starting archiveloop...
Wed  8 Jan 20:47:07 PST 2020: mounting snapshot /backingfiles/snapshots/snap-000000/snap.bin
Wed  8 Jan 20:47:09 PST 2020: mounted all snapshots
Wed  8 Jan 20:47:10 PST 2020: taking snapshot of cam disk: /backingfiles/snapshots/snap-000001/snap.bin
Wed  8 Jan 20:47:14 PST 2020: took snapshot
Wed  8 Jan 20:47:15 PST 2020: comparing /backingfiles/snapshots/snap-000000/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Wed  8 Jan 20:47:15 PST 2020: new snapshot is identical to previous one, discarding
Wed  8 Jan 20:47:15 PST 2020: Running fsck on /backingfiles/cam_disk.bin...
Wed  8 Jan 20:47:16 PST 2020: | fsck from util-linux 2.33.1
Wed  8 Jan 20:47:17 PST 2020: | fsck.fat 4.1 (2017-01-24)
Wed  8 Jan 20:47:17 PST 2020: | /dev/loop1p1: 2 files, 2/1339921 clusters
Wed  8 Jan 20:47:17 PST 2020: Finished fsck on /backingfiles/cam_disk.bin.
Wed  8 Jan 20:47:17 PST 2020: Running fsck on /backingfiles/music_disk.bin...
Wed  8 Jan 20:47:17 PST 2020: | fsck from util-linux 2.33.1
Wed  8 Jan 20:47:18 PST 2020: | fsck.fat 4.1 (2017-01-24)
Wed  8 Jan 20:47:18 PST 2020: | /dev/loop1p1: 1 files, 1/261372 clusters
Wed  8 Jan 20:47:18 PST 2020: Finished fsck on /backingfiles/music_disk.bin.
Wed  8 Jan 20:47:18 PST 2020: Waiting for time to be set by ntpd...
Wed  8 Jan 20:47:45 PST 2020: Time still not set, attempting to force it
Wed  8 Jan 20:47:53 PST 2020: Archiving...
Wed  8 Jan 20:47:53 PST 2020: Ensuring cam archive is mounted...
Wed  8 Jan 20:47:53 PST 2020: Mounting /mnt/archive...
Wed  8 Jan 20:47:54 PST 2020: Mounted /mnt/archive.
Wed  8 Jan 20:47:54 PST 2020: Ensured cam archive is mounted.
Wed  8 Jan 20:47:54 PST 2020: Checking saved folder count...
Wed  8 Jan 20:47:54 PST 2020: Ensuring cam file is mounted...
Wed  8 Jan 20:47:54 PST 2020: Mounting /mnt/cam...
Wed  8 Jan 20:47:54 PST 2020: Mounted /mnt/cam.
Wed  8 Jan 20:47:54 PST 2020: Ensured cam file is mounted.
Wed  8 Jan 20:47:56 PST 2020: There are 0 event folder(s) with 0 file(s) to move.
Wed  8 Jan 20:47:56 PST 2020: Trimming free space in /mnt/cam, which has 8 extents
Wed  8 Jan 20:47:56 PST 2020: Trim complete, image now has 8 extents
Wed  8 Jan 20:47:56 PST 2020: Unmounting /mnt/cam...
Wed  8 Jan 20:47:56 PST 2020: Unmounted /mnt/cam.
Wed  8 Jan 20:47:56 PST 2020: Finished archiving.
Wed  8 Jan 20:47:56 PST 2020: Music archive not configured or unreachable
Wed  8 Jan 20:47:56 PST 2020: unmounting /mnt/archive
Wed  8 Jan 20:47:56 PST 2020: Connecting usb to host...
Wed  8 Jan 20:47:56 PST 2020: Connected usb to host.
Wed  8 Jan 20:47:56 PST 2020: Waiting for archive to be unreachable...
Wed  8 Jan 20:49:32 PST 2020: Sleeping before retry...
Wed  8 Jan 20:49:33 PST 2020: Retrying...

marcone, 2020-01-09:
@j450h1 your diagnostics are incomplete (the beginning is missing), but the parts that are there are consistent with using a charge-only USB cable, or using the power port instead of the USB port on the Pi.

j450h1, 2020-01-09:
@marcone thank you very much! Yes, looks I switched to a charge-only USB cable without realizing it. Now I see the CAM and MUSIC folders on my mac and should work in the car as well.