marcone/teslausb #352: Music Disk Files Show Errored in FSCK

DonC52, 2019-11-12:
It seems that with the new code on the server, the music "disk" is corrupting files and keeps getting copied over and over again. I noticed that there is now a Trimming Free Space message in the log as well. The archiveloop.log looks like:
''
Mon 11 Nov 18:55:54 PST 2019: Finished archiving.
Mon 11 Nov 18:55:54 PST 2019: Copying music...
Mon 11 Nov 18:55:54 PST 2019: Starting music sync...
Mon 11 Nov 18:55:54 PST 2019: Ensuring music backing file is mounted...
Mon 11 Nov 18:55:54 PST 2019: Mounting /mnt/music...
Mon 11 Nov 18:55:54 PST 2019: Mounted /mnt/music.
Mon 11 Nov 18:55:54 PST 2019: Ensured music drive is mounted.
Mon 11 Nov 18:55:54 PST 2019: Running fsck on /mnt/music...
Mon 11 Nov 18:55:55 PST 2019: | fsck from util-linux 2.29.2
Mon 11 Nov 18:55:55 PST 2019: | fsck.fat 4.1 (2017-01-24)
Mon 11 Nov 18:55:55 PST 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Mon 11 Nov 18:55:55 PST 2019: |  Automatically removing dirty bit.
Mon 11 Nov 18:55:55 PST 2019: | Performing changes.
Mon 11 Nov 18:55:55 PST 2019: | /dev/loop1: 228 files, 204214/261372 clusters
Mon 11 Nov 18:55:55 PST 2019: Finished fsck on /mnt/music.
Mon 11 Nov 18:55:55 PST 2019: Copying music from archive...
Mon 11 Nov 18:56:06 PST 2019: Copied 1 music file(s), deleted 2, skipped 199 previously-copied files, encountered 0 copy errors and 0 delete errors.
Mon 11 Nov 18:56:06 PST 2019: Trimming free space in /mnt/music...
Mon 11 Nov 18:56:06 PST 2019: Unmounting /mnt/music...
Mon 11 Nov 18:56:06 PST 2019: Unmounted /mnt/music.
Mon 11 Nov 18:56:07 PST 2019: Finished copying music.
''

marcone, 2019-11-12:
There's no corruption. The "Dirty bit is set" message has always been there and is harmless.
There does appear to be a problem where the test command reports that some source files are newer than the existing destination files because the time stamps are off by a second or so, which causes it to copy files again unnecessarily.

marcone, 2019-11-12:
I've disabled the copying of updated files for now.

closed 2019-11-12 by marcone

DonC52, 2019-11-12:
I know that the dirty bit is not the issue but it is strange that it deletes 2 files and copies 1 file. There are 200 music Files and 24 Folders as more information.

Further it says the same thing 
**"Copied 1 music file(s), deleted 2, skipped 199 previously-copied files, encountered 0 copy errors and 0 delete errors."**
each time the Pi is booted AND each time the copy of the music files occur.

I do not have any Recent or Sentry clips to see if they are having issues as well. I am going on a weeks trip so I will have more things to look at if the log holds enough. ;-) 

I am not familiar that much with the fstrim command to say it an issue with the trimming being done. Not sure if it can be used on a software type function or not. I just installed this fresh again (was playing) and even on the first archive I got the above error. Should have had nothing to trim.

marcone, 2019-11-12:
It should only delete files that are no longer present on the server. I wonder if it deletes them, and then 'fsck' restores/undeletes them somehow, so they get deleted again.

DonC52, 2019-11-12:
It is the fstrim that is deleting the space. I put a -v in the command and it get:
/mnt/music: 223.3 MiB (234127360 bytes) trimmed

I think the delete is a music file and a directory. I am wondering rather it knows that the file was just written or something like that. The fsck shows 226 files which, I think, is the 200 file + 24 folders + . and ..?  Then the copy seems to be deleting the folder and music file and copying it back. Then the fstrim deletes 223.3 MiB and then hangs on again.

marcone, 2019-11-12:
> It is the fstrim that is deleting the space. I put a -v in the command and it get:
> /mnt/music: 223.3 MiB (234127360 bytes) trimmed
> 
> I think the delete is a music file and a directory.

`fstrim` doesn't delete files/directories. It instructs the filesystem to let the underlying block device know which parts of the filesystem are not currently in use. The "deleted 2" comes from the `rm` in copy-music.sh. You could put a log there and see which files it deleted.

> I am wondering rather it knows that the file was just written or something like that.

It should know, since both the delete and the trim go through the filesystem layer.

DonC52, 2019-11-12:
Okay - I put in some log entries in for the delete and copy of music files. I now no longer get the deletes but it seems to copy the same music file over and over again. (No deletes now.)
The music file plays fine on the PC and nothing is updating it. If I change the name on the file, I get a delete of the old file and a copy of the new name. I have not tried to play it in the car (getting ready for a long trip) but I am guessing that it will play.

If I delete the file, it picks on another file to copy over and over again.

For when I return, is there a way to see what the Pi thinks it has in the music area without plugging it into the PC?

Can you un-close this issue?

marcone, 2019-11-12:
> Okay - I put in some log entries in for the delete and copy of music files. I now no longer get the deletes but it seems to copy the same music file over and over again. (No deletes now.)

The repeated copy should be fixed now. I made it so it only copies files if they don't exist yet, not when they were updated on the server.

> The music file plays fine on the PC and nothing is updating it. If I change the name on the file, I get a delete of the old file and a copy of the new name. I have not tried to play it in the car (getting ready for a long trip) but I am guessing that it will play.

> If I delete the file, it picks on another file to copy over and over again.

Interesting behavior, but again: that should be fixed now.
 
> For when I return, is there a way to see what the Pi thinks it has in the music area without plugging it into the PC?

You could ssh in and `mount -o ro /mnt/music`

> Can you un-close this issue?

Why, are you still having problems with the current copy-music.sh ?

jun3280net, 2019-11-13:
I found this thread as my music files aren't copying.  I'm not sure if it's related to the FSCK - so I'm hoping somebody may be able to point me in the right direction.  My archiveloop.log looks like the following.  I have no issues with syncing my Teslacam videos though.  Thanks

[image]

marcone, 2019-11-13:
including the entire output of `/root/bin/setup-teslausb diagnose` is probably more useful than including screenshots of parts of log files.

jun3280net, 2019-11-13:
@marcone  Thank you!  I am pasting just today's output from your aforementioned command (unless there is a way to output all of it to txt file) : 

Wed 13 Nov 00:07:27 MST 2019: Archive is reachable.
Wed 13 Nov 00:07:27 MST 2019: Waiting for time to be set by ntpd...
Wed 13 Nov 00:07:28 MST 2019: Time now set
Wed 13 Nov 00:07:48 MST 2019: taking snapshot of cam disk: /backingfiles/snapshots/snap-000132/snap.bin
Wed 13 Nov 00:07:59 MST 2019: took snapshot
Wed 13 Nov 00:08:08 MST 2019: comparing /backingfiles/snapshots/snap-000131/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Wed 13 Nov 00:08:08 MST 2019: making links for /backingfiles/snapshots/newsnap/mnt, retargeted to /backingfiles/snapshots/snap-000132/mnt
Wed 13 Nov 00:08:20 MST 2019: made all links for /backingfiles/snapshots/newsnap/mnt
Wed 13 Nov 00:08:20 MST 2019: Archiving...
Wed 13 Nov 00:08:20 MST 2019: Disconnecting usb from host...
Wed 13 Nov 00:08:20 MST 2019: Disconnected usb from host.
Wed 13 Nov 00:08:20 MST 2019: Ensuring cam archive is mounted...
Wed 13 Nov 00:08:20 MST 2019: Mounting /mnt/archive...
Wed 13 Nov 00:08:21 MST 2019: Mounted /mnt/archive.
Wed 13 Nov 00:08:21 MST 2019: Ensured cam archive is mounted.
Wed 13 Nov 00:08:21 MST 2019: Ensuring music archive is mounted...
Wed 13 Nov 00:08:21 MST 2019: Mounting /mnt/musicarchive...
Wed 13 Nov 00:08:21 MST 2019: Mounted /mnt/musicarchive.
Wed 13 Nov 00:08:21 MST 2019: Ensured music archive is mounted.
Wed 13 Nov 00:08:21 MST 2019: Checking saved folder count...
Wed 13 Nov 00:08:21 MST 2019: Ensuring cam file is mounted...
Wed 13 Nov 00:08:21 MST 2019: Mounting /mnt/cam...
Wed 13 Nov 00:08:21 MST 2019: Mounted /mnt/cam.
Wed 13 Nov 00:08:21 MST 2019: Ensured cam file is mounted.
Wed 13 Nov 00:08:22 MST 2019: Running fsck on /mnt/cam...
Wed 13 Nov 00:08:23 MST 2019: | fsck from util-linux 2.33.1
Wed 13 Nov 00:08:30 MST 2019: | fsck.fat 4.1 (2017-01-24)
Wed 13 Nov 00:08:30 MST 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Wed 13 Nov 00:08:30 MST 2019: |  Automatically removing dirty bit.
Wed 13 Nov 00:08:30 MST 2019: | Performing changes.
Wed 13 Nov 00:08:30 MST 2019: | /dev/loop5: 489 files, 404402/7130605 clusters
Wed 13 Nov 00:08:30 MST 2019: Finished fsck on /mnt/cam.
Wed 13 Nov 00:08:38 MST 2019: There are 13 event folder(s) with 294 file(s) to move.
Wed 13 Nov 00:08:38 MST 2019: Starting recording archiving: 13 event folder(s) with 294 file(s)
Wed 13 Nov 00:08:38 MST 2019: Sending Pushover message for moved files.
Wed 13 Nov 00:08:39 MST 2019: Sending IFTTT message for moved files.
Wed 13 Nov 00:08:40 MST 2019: Moving clips to archive...
Wed 13 Nov 00:08:40 MST 2019: Creating output directory 'SavedClips/2019-11-12_18-06-50'
Wed 13 Nov 00:08:40 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-56-07-front.mp4'
Wed 13 Nov 00:08:51 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-56-07-front.mp4'
Wed 13 Nov 00:08:51 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-56-07-left_repeater.mp4'
Wed 13 Nov 00:09:01 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-56-07-left_repeater.mp4'
Wed 13 Nov 00:09:01 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-56-07-right_repeater.mp4'
Wed 13 Nov 00:09:11 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-56-07-right_repeater.mp4'
Wed 13 Nov 00:09:11 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-57-08-front.mp4'
Wed 13 Nov 00:09:20 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-57-08-front.mp4'
Wed 13 Nov 00:09:20 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-57-08-left_repeater.mp4'
Wed 13 Nov 00:09:30 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-57-08-left_repeater.mp4'
Wed 13 Nov 00:09:30 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-57-08-right_repeater.mp4'
Wed 13 Nov 00:09:40 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-57-08-right_repeater.mp4'
Wed 13 Nov 00:09:40 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-58-11-front.mp4'
Wed 13 Nov 00:09:50 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-58-11-front.mp4'
Wed 13 Nov 00:09:50 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-58-11-left_repeater.mp4'
Wed 13 Nov 00:10:00 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-58-11-left_repeater.mp4'
Wed 13 Nov 00:10:00 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-58-11-right_repeater.mp4'
Wed 13 Nov 00:10:09 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-58-11-right_repeater.mp4'
Wed 13 Nov 00:10:09 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-59-13-front.mp4'
Wed 13 Nov 00:10:18 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-59-13-front.mp4'
Wed 13 Nov 00:10:19 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-59-13-left_repeater.mp4'
Wed 13 Nov 00:10:27 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-59-13-left_repeater.mp4'
Wed 13 Nov 00:10:27 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-59-13-right_repeater.mp4'
Wed 13 Nov 00:10:35 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_17-59-13-right_repeater.mp4'
Wed 13 Nov 00:10:35 MST 2019: Moving 'SavedClips/2019-11-12_18-06-50/2019-11-12_18-00-15-front.mp4'
Wed 13 Nov 00:10:43 MST 2019: Moved 'SavedClips/2019-11-12_18-06-50/2019-11-12_18-00-15-front.mp4'
[... cut ...]
root     10227 10208  6 10:00 pts/0    00:00:00 sed -E s/\o033\[0m//
root     10286 10218  0 10:00 pts/0    00:00:00 ps -eaf
root     12166     2  0 03:52 ?        00:00:02 [loop111]
root     19691     2  0 07:47 ?        00:00:02 [loop115]
root     22106     2  0 04:50 ?        00:00:02 [loop112]
root     29644     2  0 08:45 ?        00:00:02 [loop116]
root     32057     2  0 05:49 ?        00:00:02 [loop113]
teslausb-mdls has been up 7 hours, 9 minutes. System time is Wed 13 Nov 10:00:14 MST 2019
====== end of diagnostics ======

marcone, 2019-11-13:
Just redirect the output to a file, e.g.
```
/root/bin/setup-teslausb diagnose > /tmp/diagnostics.txt
```
then attach that file to a new issue.

jun3280net, 2019-11-14:
> /root/bin/setup-teslausb diagnose > /tmp/diagnostics.txt

Thanks @marcone.  Will attach to new issue.  On a separate note to retrieve the file from PI to Local.  On COMMAND PROMPT Window (similar to SSH) : 

SFTP pi@teslausb
cd /tmp
get diagnostics.txt

if you are unsure of commands - type HELP