marcone/teslausb #353: Rasbperry Pi setup causing Model X touch screen unresponsive

jdw999, 2019-11-12:
I am a newbie here, and I followed the instruction, was able to get the Raspberry Pi setup working on my 2018 Model X, SW (2019.36.2.1), for a few days. Then it started to have issues. 

1. no clips were being uploaded (only worked on 1st day when I got the setup working)
2. Then I got the error message "Touch screen unresponsive", press scroll wheel to reset.  I tried double scroll wheel reset method multiple times (4+), it's not solving the problem.  Screen was still unresponsive.   Finally after googling, Tesla forum mentioned that USB drive could cause the screen to go unresponsive. https://forums.tesla.com/forum/forums/touchscreen-unresponsive 

So I removed the Raspberry Pi from the USB plug, and reboot the Tesla screen through scroll wheel method, and guess what, screen is back alive, responding to touch actions.  

So I was wondering what could cause this? how can I resolve it on Raspberry Pi and still having the dashcam.   
Thanks.

marcone, 2019-11-12:
I really don't think the two are related at all.

jdw999, 2019-11-12:
I certainly hope so.  But my screen wasn't working at all for a good 1+ hour, even after multiple reboots.  The only thing I did was unplugging the Pi from USB port, and rebooted the Model X.  So that's why it seemed to be the only culprit to me. 

I re-plugged the Pi back to USB port, screen is working still, I'll keep an eye on it, and see when it will re-appear again. 

On my Pi, I run the selfupdate and setup again, tried another sentry test, now sentry clips are uploading again.    thanks for the great setup.

jdw999, 2019-11-15:
Hi, 
Just want to give an update, there were another 2 instances of screen freeze with Raspberry Pi setup in a two-day span, was able to recover by unplug the Pi and reboot the X.  Since my wife is driving this car, I don't think she could handle this.  So for the time being, had to put this setup aside and go back to regular USB thumb drive.   Anyway, just want to put this out there in case other people might report similar behavior.   thanks.

marcone, 2019-12-04:
I had been seeing the occasional "touch screen unresponsive" messages too, but they always went away on their own after a little while. It does seem that this is related to the Pi though, since as you noted, unplugging the Pi immediately fixes the issue.
Also, I've noticed some odd behavior when plugging the Pi into my Mac, where the mouse becomes very jittery/sluggish while the Pi is booting up, so clearly the Pi is having some effect on USB (and I assume the car's touch screen is connected to the MCU via USB).

Unfortunately I don't know what I could do about this, since this appears to be something the Pi is doing, unrelated to teslausb itself.

gonzonia, 2019-12-05:
I can confirm I was having the same issue. I actually stopped using the teslausb months ago and was just checking back in to see what updates had been made and maybe trying again.

endurance1968, 2019-12-07:
Hmmm is this the reason why after the BETA there is no further release? It would be great to still get an update release which I would like to use with a bigger SD card on my M3. Now using the old release since month without issues (on car side) except SD to small.

marcone, 2019-12-07:
So get a bigger one? You don't need a new release to use a bigger sd card.

endurance1968, 2019-12-07:
sure not - but If I spent efforts I would like to combine both activities.

alectrocute, 2019-12-08:
I’m shooting in the dark as I have yet to try this on a Tesla, but what if you had the Raspberry Pi boot with the USB devices disabled and eventually enable them after all system services load successfully? Is there any reason why USB needs to be enabled immediately on boot, unless you’re powering the device via USB itself?

alectrocute, 2019-12-08:
@endurance1968 Just upgrade your SD card. The software is not limiting your ability to do so.

gonzonia, 2019-12-08:
On pi0 that's exactly what you're doing.

alectrocute, 2019-12-08:
> On pi0 that's exactly what you're doing.

That’s suspect. I have a suspicion it’s drawing too much power, causing the Tesla entertainment MCU to panic a bit. This is just speculation.

alectrocute, 2019-12-08:
A ridiculous workaround would be using a splitter with a tiny LiPo battery plugged inline, charging while the Tesla’s USB port is active but also supplementing amperage when needed. But surely there’s a better way.

gonzonia, 2019-12-08:
I think I tried it with an external power (i.e. a car charger vs. the Tesla USB outlets) and still had the problem. If I have time this week maybe I'll re-visit and try again.

jdw999, 2019-12-08:
> A ridiculous workaround would be using a splitter with a tiny LiPo battery plugged inline, charging while the Tesla’s USB port is active but also supplementing amperage when needed. But surely there’s a better way.
> 
> Are there any easily spliced 5V leads nearby the host USB female port?

I think we can rule out this work around, which I have already tried. 
- I had a battery pack powered by 12V from cigarette lighter, which powers the Pi HW at the same time.  The battery pack supports pass-through charging.  This is the original setup which I was seeing the screen freeze.

marcone, 2020-04-19:
I'm going to assume that the mouse freezing issue I [mentioned above](https://github.com/marcone/teslausb/issues/353#issuecomment-561760051) and the touchscreen unresponsive issue are related.
The mouse issue occurs even with no sd card plugged in to the Pi, i.e. no Raspbian let alone Teslausb involved, so it seems this is a fundamental issue with the Raspberry Pi hardware or firmware.
Since there is nothing I can do about that, I'm going to close this issue.

closed 2020-04-19 by marcone