marcone/teslausb #354: Dashcam not saving

idealerror, 2019-11-13:
Oddly I get the dashcam icon on the screen on my Model S, but when it goes to archive the data, it only has the SentryClips folder.

Does the dash cam data save as well or am I missing something?

marcone, 2019-11-13:
The car is always recordings when it is on, and writes those recordings to RecentClips. If you press the Dashcam icon on the screen, it will move the last 10 minutes worth of recordings to a folder under SavedClips. If you turn on Sentry mode and there's a Sentry event, it will move the last 10 minutes worth of recordings to a folder under SentryClips.
Only SavedClips and SentryClips are archived.

closed 2019-11-13 by marcone

idealerror, 2019-11-13:
Ah I kept thinking of RecentClips.  Feeling dumb.

I wonder if it's possible to grab RecentClips and move them out and store them so they could potentially be used as dash cam footage.

marcone, 2019-11-13:
Not sure what you mean by "used as dash cam footage", but you can access a copy of RecentClips via Samba.