marcone/teslausb #356: Pi 0 not showing up as usb drive

jyen, 2019-11-14:
https://pastebin.com/7QvBFS2M

Setup Pi 0 and it is not being recognized as usb drive. I've tried several cables and none were working. Pi 0 used to work as a usb drive from one of these cables but constantly getting drive too slow message so I moved over to pi4. Now I'm redoing it on pi0 again but have been unsuccessful due to this issue.

marcone, 2019-11-14:
I still suspect the cable, or using the wrong port on the Pi, as the diagnostics show that the drive is being exported, but nothing is connecting to it.

jyen, 2019-11-14:
Seems to be the case, i tried bunch of cables and finally found one that works. Closing the issue now. Thanks!

closed 2019-11-14 by jyen