marcone/teslausb #357: Teslausb 99% Working : Music Not Syncing & Unable To Turn Off IFTTT Notifications

jun3280net, 2019-11-14:
Hi @marcone my teslausb is working great.  The core function (recording video and syncing) is working as needed.  It took me a number of days and multiple re-installs to debug and get everything working properly.  

That said - I'd like to get my music to sync properly - which seemed to only work on the original installation.  Since then, I've done a few updates to get other features up and running (eg) Notifications (IFTTT, Pushover), Wireless AP (SMB), etc.  I'm not sure what exactly what I changed in the configuration to break the music sync.  To simplify the SHARE (because I know read/write is working), I have my music folder nested in the same parent SHARE Directory (children SavedClips, SentryClips, TeslaMusic).

On a separate note - I can't seem to get IFTTT notifications to turn off (I have it set to FALSE now).  It is much slower than Pushover.

Lastly - there are a number of snapshots being created (when I run df -h).  How exactly, do I make use of these snapshots - can they be saved to my Share as well?

Below is my teslausb_setup_variables.conf file.  I've also attached my diagnostics.txt file from the following command : 
[diagnostics.txt](https://github.com/marcone/teslausb/files/3846236/diagnostics.txt)

/root/bin/setup-teslausb diagnose > /tmp/diagnostics.txt

[image]
[image]
[image]
[image]

marcone, 2019-11-14:
Setup doesn't support turning notifications off, so you'll have to manually remove the IFTTT config file, "/root/.teslaCamIftttSettings"

As for syncing music, there appears to be a problem with the music image, since it initially mounts:
```
Thu 14 Nov 03:15:22 MST 2019: Mounting /mnt/music...
Thu 14 Nov 03:15:22 MST 2019: Mounted /mnt/music.
Thu 14 Nov 03:15:22 MST 2019: Ensured music drive is mounted.
```
but then a few seconds later it's apparently not mounted anymore:
```
Thu 14 Nov 03:15:26 MST 2019: Copying music from archive...
Thu 14 Nov 03:15:26 MST 2019: /mnt/music not mounted, skipping
Thu 14 Nov 03:15:26 MST 2019: Could not trim free space in /mnt/music. Not Mounted.
```
You could try manually mounting and unmounting it to see if it prints any additional information then.

jun3280net, 2019-11-14:
Hi @marcone thanks again for your eyes on this.

1.  ran **"mount -o ro /mnt/music"** no error messages from that
2.  ran **"df -h"** takes a while to return (upwards of 10 minutes is this normal?)
3.  directory structure looks good - dominated by "snapshots" (do these need to be deleted - as I really don't know what they are for)
[image]
[image]

4.  ran **"/root/bin/copy-music.sh"** it returns the following : **/root/bin/copy-music.sh: line 3: log: command not found**
5.  Would deleting the two folders I currently have on the PI Music Drive help?
6.  Are there any other logs I could/should be checking?
7.  Referring back to Nov 12 I see the drive is mounted successfully before trying to copy files, but it is unable to create directories : 
[image]
[image]

8.  Is there something here that might help - as it mentions fixing TRIM issues? 
[https://github.com/marcone/teslausb/commit/44fc740042918a3d735ad5934272079366a81502](https://github.com/marcone/teslausb/commit/44fc740042918a3d735ad5934272079366a81502)

marcone, 2019-11-14:
Re: point 4: "log" is a functional available to the scripts to log things. You could define it manually to just call "echo", but then copy-music.sh would still have failed because you mounted /mnt/music read-only.

Re: point 7: it says "/mnt/music is already mounted", so perhaps it was already mounted read-only, which would explain why it couldn't create any directories on it.

I would try the following:
```
modprobe -r g_mass_storage
mount /mnt/music
fsck /mnt/music
touch /mnt/music/testfile
umount /mnt/music
modprobe g_mass_storage
```
and see what that says

jun3280net, 2019-11-14:
Thanks @marcone 

Here's what I see when running the aforementioned commands : 
[image]
[image]

If I unmount the drive .... then run "/root/bin/copy-music.sh" would it work?

marcone, 2019-11-14:
> If I unmount the drive .... then run "/root/bin/copy-music.sh" would it work?

No, /mnt/music needs to be mounted when copy-music is called

jun3280net, 2019-11-14:
@marcone the "touch" looks to have worked.  But that was after I "unmounted" then "remounted" the /mtn/music folder using your commands.

When I log into SSH, the drive is already mounted.  I probably should have tried "touch" with the drive mounted from the beginning to see if it was "read only".

Any idea what might be causing the drive to be mounted as read only?

Thanks,

J

marcone, 2019-11-14:
> Any idea what might be causing the drive to be mounted as read only?

Filesystem errors encountered while accessing it would cause it to be remounted read-only.
Are you sure it's actually mounted, or do you mean that the /mnt/music directory exists?
(running the `mount` command will show what is actually mounted)

jun3280net, 2019-11-15:
@marcone - not sure what the issue was, but since running the commands you provided (maybe it was the "touch") when I got home from work, the music synced.  I added a few more albums to the Share.  I left the house - and when I returned home - upon syncing the Tesla Cam clips, the music was also update.

Thanks for your help - but it's not clear to me what exactly what I did to cause the problem to begin with.  However, I'm happy everything is working.

That said - let's say I have an album I want to load up on TeslaUSB before going out on a road trip, what is the best way/command to achieve this.

Cheers

marcone, 2019-11-15:
> That said - let's say I have an album I want to load up on TeslaUSB before going out on a road trip, what is the best way/command to achieve this.

Copy them to your share, then ssh into the pi and `touch /tmp/archive_is_unreachable` to trigger a sync.

closed 2019-11-15 by marcone

jun3280net, 2019-11-18:
> touch /tmp/archive_is_unreachable

Thanks @marcone.  I gave it a try it didn't trigger.  Do I need to issue a "reboot" after?  If it triggers - would the IFTTT or Pushover send a notification indicating the sync is starting?  Aside from that - everything is working great - Thank you for all your help.

marcone, 2019-11-18:
@jun3280net if you typed it exactly as I said, then it should trigger an archive cycle. It should log
`Simulating archive being unreachable` immediately followed by `Waiting for archive to be reachable...` and `Archive is reachable` (assuming you're actually in wifi range).

jun3280net, 2019-11-19:
> @jun3280net if you typed it exactly as I said, then it should trigger an archive cycle. It should log
> `Simulating archive being unreachable` immediately followed by `Waiting for archive to be reachable...` and `Archive is reachable` (assuming you're actually in wifi range).

@marcone - I feel terrible for bugging you regarding something so trivial, but I am typing exactly what you've posted.  I even checked the directory to confirm details.  I can confirm "touch /tmp/archive_test" will create the file "archive_test" in the tmp folder.  However, when typing your command  "touch /tmp/archive_is_unreachable" - does not appear to be working (unless "archive_is_unreachable" is being created and deleted instantly).  No messages are outputted.

[image]

marcone, 2019-11-19:
Look at /mutable/archiveloop.log and you'll see that it is actually doing something in response to that file being created (namely, deleting it, and then going through an archive cycle)