marcone/teslausb #358: Archive dies when connection is lost

joetomasone, 2019-11-15:
Second time I've seen this issue - for whatever reason, the WiFi connection is lost, once when I drove away while it was still archiving, and tonight it just disconnected while parked (yet it is connected since I can SSH into it).    This causes archiveloop to sit and wait for the archive to be unreachable, and thus far it doesn't recover unless rebooted.  

Here is a snippet of the log showing the issue and a ping of the archive server. 

`Thu 14 Nov 21:57:44 EST 2019: Moving 'SavedClips/2019-11-11_13-01-16/2019-11-11_12-53-03-left_repeater.mp4'
Thu 14 Nov 21:58:18 EST 2019: Moved 'SavedClips/2019-11-11_13-01-16/2019-11-11_12-53-03-left_repeater.mp4'
Thu 14 Nov 21:58:18 EST 2019: Moving 'SavedClips/2019-11-11_13-01-16/2019-11-11_12-53-03-right_repeater.mp4'
Thu 14 Nov 21:58:51 EST 2019: Moved 'SavedClips/2019-11-11_13-01-16/2019-11-11_12-53-03-right_repeater.mp4'
Thu 14 Nov 21:58:51 EST 2019: Moving 'SavedClips/2019-11-11_13-01-16/2019-11-11_12-53-03-back.mp4'
Thu 14 Nov 21:59:25 EST 2019: Moved 'SavedClips/2019-11-11_13-01-16/2019-11-11_12-53-03-back.mp4'
Thu 14 Nov 21:59:25 EST 2019: Moving 'SavedClips/2019-11-11_13-01-16/2019-11-11_12-54-03-front.mp4'
Thu 14 Nov 21:59:58 EST 2019: connection dead, killing archive-clips
Thu 14 Nov 21:59:59 EST 2019: Unmounting /mnt/cam...
umount: /mnt/cam: target is busy.
Thu 14 Nov 21:59:59 EST 2019: Failed to unmount /mnt/cam, trying lazy unmount.
Thu 14 Nov 21:59:59 EST 2019: lazily unmounted /mnt/cam
Thu 14 Nov 21:59:59 EST 2019: Finished archiving.
Thu 14 Nov 22:00:04 EST 2019: Music archive not configured or unreachable
Thu 14 Nov 22:00:04 EST 2019: unmounting /mnt/archive
Thu 14 Nov 22:00:37 EST 2019: unmounting /mnt/musicarchive
Thu 14 Nov 22:00:37 EST 2019: Connecting usb to host...
Thu 14 Nov 22:00:37 EST 2019: Connected usb to host.
Thu 14 Nov 22:00:37 EST 2019: Truncating log...
Thu 14 Nov 22:00:38 EST 2019: Waiting for archive to be unreachable...
Thu 14 Nov 22:10:09 EST 2019: Sleeping before retry...
Thu 14 Nov 22:10:10 EST 2019: Retrying...
Thu 14 Nov 22:32:53 EST 2019: Sleeping before retry...
Thu 14 Nov 22:32:54 EST 2019: Retrying...
Thu 14 Nov 22:34:21 EST 2019: Sleeping before retry...
Thu 14 Nov 22:34:22 EST 2019: Retrying...
Thu 14 Nov 22:37:01 EST 2019: Sleeping before retry...
^C
pi@teslausb:~ $ ping 192.168.1.5
PING 192.168.1.5 (192.168.1.5) 56(84) bytes of data.
64 bytes from 192.168.1.5: icmp_seq=1 ttl=64 time=7.01 ms
64 bytes from 192.168.1.5: icmp_seq=2 ttl=64 time=23.6 ms
^C
--- 192.168.1.5 ping statistics ---
3 packets transmitted, 2 received, 33.3333% packet loss, time 6ms
rtt min/avg/max/mdev = 7.014/15.300/23.586/8.286 ms
pi@teslausb:~ $
`

joetomasone, 2019-11-15:
The issue happened again:

Fri 15 Nov 02:15:05 EST 2019: Moving 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-38-43-front.mp4'
Fri 15 Nov 02:16:12 EST 2019: Moved 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-38-43-front.mp4'
Fri 15 Nov 02:16:12 EST 2019: Moving 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-38-43-left_repeater.mp4'
Fri 15 Nov 02:16:48 EST 2019: Moved 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-38-43-left_repeater.mp4'
Fri 15 Nov 02:16:48 EST 2019: Moving 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-38-43-right_repeater.mp4'
Fri 15 Nov 02:17:19 EST 2019: Moved 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-38-43-right_repeater.mp4'
Fri 15 Nov 02:17:19 EST 2019: Moving 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-38-43-back.mp4'
Fri 15 Nov 02:18:26 EST 2019: Moved 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-38-43-back.mp4'
Fri 15 Nov 02:18:26 EST 2019: Moving 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-39-44-front.mp4'
Fri 15 Nov 02:18:45 EST 2019: Moved 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-39-44-front.mp4'
Fri 15 Nov 02:18:45 EST 2019: Moving 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-39-44-left_repeater.mp4'
Fri 15 Nov 02:19:03 EST 2019: Moved 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-39-44-left_repeater.mp4'
Fri 15 Nov 02:19:03 EST 2019: Moving 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-39-44-right_repeater.mp4'
Fri 15 Nov 02:19:20 EST 2019: Moved 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-39-44-right_repeater.mp4'
Fri 15 Nov 02:19:20 EST 2019: Moving 'SavedClips/2019-11-14_10-40-26/2019-11-14_10-39-44-back.mp4'
Fri 15 Nov 02:19:32 EST 2019: connection dead, killing archive-clips
Fri 15 Nov 02:19:33 EST 2019: Unmounting /mnt/cam...
umount: /mnt/cam: target is busy.
Fri 15 Nov 02:19:33 EST 2019: Failed to unmount /mnt/cam, trying lazy unmount.
Fri 15 Nov 02:19:33 EST 2019: lazily unmounted /mnt/cam
Fri 15 Nov 02:19:33 EST 2019: Finished archiving.
Fri 15 Nov 02:19:38 EST 2019: Music archive not configured or unreachable
Fri 15 Nov 02:19:38 EST 2019: unmounting /mnt/archive
Fri 15 Nov 02:20:30 EST 2019: unmounting /mnt/musicarchive
Fri 15 Nov 02:20:30 EST 2019: Connecting usb to host...
Fri 15 Nov 02:20:30 EST 2019: Connected usb to host.
Fri 15 Nov 02:20:30 EST 2019: Truncating log...
Fri 15 Nov 02:20:30 EST 2019: Waiting for archive to be unreachable...
Fri 15 Nov 02:21:29 EST 2019: Sleeping before retry...
Fri 15 Nov 02:21:30 EST 2019: Retrying...
Fri 15 Nov 02:29:47 EST 2019: Sleeping before retry...
Fri 15 Nov 02:29:48 EST 2019: Retrying...
Fri 15 Nov 02:35:58 EST 2019: Sleeping before retry...

joetomasone, 2019-11-15:
Repeated again. 

Fri 15 Nov 03:54:58 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-53-11-left_repeater.mp4'
Fri 15 Nov 03:55:31 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-53-11-left_repeater.mp4'
Fri 15 Nov 03:55:31 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-53-11-right_repeater.mp4'
Fri 15 Nov 03:56:01 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-53-11-right_repeater.mp4'
Fri 15 Nov 03:56:02 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-53-11-back.mp4'
Fri 15 Nov 03:56:45 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-53-11-back.mp4'
Fri 15 Nov 03:56:45 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-54-12-front.mp4'
Fri 15 Nov 03:57:15 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-54-12-front.mp4'
Fri 15 Nov 03:57:15 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-54-12-left_repeater.mp4'
Fri 15 Nov 03:57:43 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-54-12-left_repeater.mp4'
Fri 15 Nov 03:57:43 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-54-12-right_repeater.mp4'
Fri 15 Nov 03:58:15 EST 2019: connection dead, killing archive-clips
Fri 15 Nov 03:58:15 EST 2019: Unmounting /mnt/cam...
umount: /mnt/cam: target is busy.
Fri 15 Nov 03:58:15 EST 2019: Failed to unmount /mnt/cam, trying lazy unmount.
Fri 15 Nov 03:58:15 EST 2019: lazily unmounted /mnt/cam
Fri 15 Nov 03:58:15 EST 2019: Finished archiving.
Fri 15 Nov 03:58:20 EST 2019: Music archive not configured or unreachable
Fri 15 Nov 03:58:20 EST 2019: unmounting /mnt/archive
Fri 15 Nov 03:59:12 EST 2019: unmounting /mnt/musicarchive
Fri 15 Nov 03:59:12 EST 2019: Connecting usb to host...
Fri 15 Nov 03:59:12 EST 2019: Connected usb to host.
Fri 15 Nov 03:59:12 EST 2019: Truncating log...
Fri 15 Nov 03:59:13 EST 2019: Waiting for archive to be unreachable...

joetomasone, 2019-11-15:
Just rebooted the pi and ran a tail -f on the log.   It lost connection again shortly after starting:

pi@teslausb:~ $ tail -f /mutable/archiveloop.log
Fri 15 Nov 12:49:54 EST 2019: | /dev/loop0: 3202 files, 1309310/1309310 clusters
Fri 15 Nov 12:49:54 EST 2019: Finished fsck on /mnt/music.
Fri 15 Nov 12:49:54 EST 2019: Unmounting /mnt/music...
Fri 15 Nov 12:49:54 EST 2019: Unmounted /mnt/music.
Fri 15 Nov 12:49:54 EST 2019: Connecting usb to host...
Fri 15 Nov 12:49:54 EST 2019: Connected usb to host.
Fri 15 Nov 12:49:54 EST 2019: Waiting for archive to be reachable...
Fri 15 Nov 12:49:54 EST 2019: Archive is reachable.
Fri 15 Nov 12:49:55 EST 2019: Waiting for time to be set by ntpd...
Fri 15 Nov 12:50:21 EST 2019: Time still not set, attempting to force it
Fri 15 Nov 12:50:51 EST 2019: Archiving...
Fri 15 Nov 12:50:51 EST 2019: Disconnecting usb from host...
Fri 15 Nov 12:50:51 EST 2019: Disconnected usb from host.
Fri 15 Nov 12:50:51 EST 2019: Ensuring cam archive is mounted...
Fri 15 Nov 12:50:51 EST 2019: Mounting /mnt/archive...
Fri 15 Nov 12:50:52 EST 2019: Mounted /mnt/archive.
Fri 15 Nov 12:50:52 EST 2019: Ensured cam archive is mounted.
Fri 15 Nov 12:50:52 EST 2019: Ensuring music archive is mounted...
Fri 15 Nov 12:50:52 EST 2019: Mounting /mnt/musicarchive...
Fri 15 Nov 12:50:52 EST 2019: Mounted /mnt/musicarchive.
Fri 15 Nov 12:50:52 EST 2019: Ensured music archive is mounted.
Fri 15 Nov 12:50:52 EST 2019: Checking saved folder count...
Fri 15 Nov 12:50:52 EST 2019: Ensuring cam file is mounted...
Fri 15 Nov 12:50:52 EST 2019: Mounting /mnt/cam...
Fri 15 Nov 12:50:52 EST 2019: Mounted /mnt/cam.
Fri 15 Nov 12:50:52 EST 2019: Ensured cam file is mounted.
Fri 15 Nov 12:50:52 EST 2019: Running fsck on /mnt/cam...
Fri 15 Nov 12:50:53 EST 2019: | fsck from util-linux 2.33.1
Fri 15 Nov 12:50:57 EST 2019: | fsck.fat 4.1 (2017-01-24)
Fri 15 Nov 12:50:57 EST 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Fri 15 Nov 12:50:57 EST 2019: |  Automatically removing dirty bit.
Fri 15 Nov 12:50:57 EST 2019: | Performing changes.
Fri 15 Nov 12:50:57 EST 2019: | /dev/loop0: 2240 files, 1553954/2096607 clusters
Fri 15 Nov 12:50:57 EST 2019: Finished fsck on /mnt/cam.
Fri 15 Nov 12:50:57 EST 2019: There are 228 event folder(s) with 1784 file(s) to move.
Fri 15 Nov 12:50:58 EST 2019: Starting recording archiving: 228 event folder(s) with 1784 file(s)
Fri 15 Nov 12:50:58 EST 2019: Sending Pushover message for moved files.
Fri 15 Nov 12:50:58 EST 2019: Moving clips to archive...
Fri 15 Nov 12:50:58 EST 2019: Creating output directory 'SavedClips/2019-11-07_15-24-04'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-07_15-26-03'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-07_15-34-57'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-09_21-22-15'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-11_12-50-59'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-11_13-01-16'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-11_15-38-56'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-12_11-24-30'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-14_10-40-26'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-14_14-06-24'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-14_16-47-04'
Fri 15 Nov 12:50:59 EST 2019: Creating output directory 'SavedClips/2019-11-14_16-57-45'
Fri 15 Nov 12:50:59 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-54-12-right_repeater.mp4'
Fri 15 Nov 12:51:44 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-54-12-right_repeater.mp4'
Fri 15 Nov 12:51:44 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-54-12-back.mp4'
Fri 15 Nov 12:52:18 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-54-12-back.mp4'
Fri 15 Nov 12:52:18 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-55-13-front.mp4'
Fri 15 Nov 12:52:54 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-55-13-front.mp4'
Fri 15 Nov 12:52:54 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-55-13-left_repeater.mp4'
Fri 15 Nov 12:53:32 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-55-13-left_repeater.mp4'
Fri 15 Nov 12:53:32 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-55-13-right_repeater.mp4'
Fri 15 Nov 12:54:30 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-55-13-right_repeater.mp4'
Fri 15 Nov 12:54:30 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-55-13-back.mp4'
Fri 15 Nov 12:55:12 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-55-13-back.mp4'
Fri 15 Nov 12:55:13 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-56-14-front.mp4'
Fri 15 Nov 12:56:12 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-56-14-front.mp4'
Fri 15 Nov 12:56:12 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-56-14-left_repeater.mp4'
Fri 15 Nov 12:56:47 EST 2019: Moved 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-56-14-left_repeater.mp4'
Fri 15 Nov 12:56:47 EST 2019: Moving 'SavedClips/2019-11-14_16-57-45/2019-11-14_16-56-14-right_repeater.mp4'
[... 16 lines cut ...]
Fri 15 Nov 12:59:52 EST 2019: Music archive not configured or unreachable
[... cut ...]
64 bytes from 192.168.1.5: icmp_seq=1 ttl=64 time=7.75 ms
64 bytes from 192.168.1.5: icmp_seq=2 ttl=64 time=5.25 ms
64 bytes from 192.168.1.5: icmp_seq=3 ttl=64 time=9.38 ms
64 bytes from 192.168.1.5: icmp_seq=4 ttl=64 time=9.85 ms
64 bytes from 192.168.1.5: icmp_seq=5 ttl=64 time=10.6 ms
^C
--- 192.168.1.5 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 11ms
rtt min/avg/max/mdev = 5.251/8.561/10.573/1.900 ms
pi@teslausb:~ $

/root/teslausb.conf:

ARCHIVE_HOST_NAME=192.168.1.5
ARCHIVE_DELAY=20
SNAPSHOTS_ENABLED=false

marcone, 2019-11-16:
https://github.com/marcone/teslausb/pull/359 should improve things. Please give it a try.

closed 2019-11-16 by marcone

joetomasone, 2019-11-17:
This seems to have fixed the issue for me.   Thanks!