marcone/teslausb #360: As designed?

BrianDrury, 2019-11-16:
First, a big thank you to Marcone.

I have the pi installed in a model 3 and all looks good so far but I have two observations which may be as designed I am not sure.
1.	When I return home the USB option under the car music tab disappears and the track I was listening to stops plus the camera icon disappears.
2.	I notice that a lot of wifi data is sent to teslausb immediately after a trip even though the music folders are in sync with the linked computer.
Is this correct?

lborloz, 2019-11-16:
1) is definitely intentional. See here: https://github.com/marcone/teslausb/blob/main-dev/run/archiveloop#L355

2) can you define "a lot"?

BrianDrury, 2019-11-16:
Thank you.

Today my PC trafic monitor says 1.18 GB was uploaded to somewhere. During normal PC operation I see very little outward bound traffic but when the car returns to base the PC uploads at 3Mbps for several minutes (not sure how long as I don't watch). Coincidentally my TeslaMusic folder is 1.18GB.

Could the PC be uploading everything everytime I return home?

marcone, 2019-11-16:
> Could the PC be uploading everything everytime I return home?

It shouldn't, but you could check the log to see what it was doing.

marcone, 2019-11-16:
Also, there was a bug a few days ago that could cause some (approximately half) the files to be reuploaded unncessarily. You could try running `/root/bin/setup-teslausb` again to get the fix and see if that fixes your problem too.

BrianDrury, 2019-11-16:
OK, I just tried /root/bin/setup-teslausb and when the pi re-booted it did not do the usual big upload. I shall be using the car tomorrow so will check out some music and capture some video. Will report back the result.

Incidentally, where would I find the log file?

marcone, 2019-11-16:
in /mutable

BrianDrury, 2019-11-17:
#360 update:

I used the car today and on return home there was no unexpected upload from the PC to the pi so that is good.

During my trip I had the camera icon visible on the car screen and music was playing from the USB. However, on the return trip the music suddenly stopped and a few seconds later the camera icon disappeared. I waited a couple of minutes and it did not re-appear so removed the USB connection and replaced. This restored both the music and the camera icon but within about 10 minutes both were lost again. This was repeated but without playing any music and this time the camera icon was present all the way home.

When I arrived home I tried to SSH logon using normal user name and password but the response to a correct password was no prompt. This was repeated several times with the same result. I then tried using a wrong password and received the correct 'Access denied' response.

I have extracted the log file and attached it here.

[archiveloop.txt](https://github.com/marcone/teslausb/files/3855985/archiveloop.txt)

marcone, 2019-11-18:
What you describe about music stopping and the camera icon disappearing sounds like a bad cable or loose connection or something. It could also be that the Pi is too slow to do dashcam and music at the same time (see https://github.com/marcone/teslausb/issues/239 for more tales of woe regarding it being too slow). The part of the log you attached shows the Pi booting up around 17:32 GMT, then apparently losing power and then booting up again later at 18:47 GMT, but otherwise there's nothing unusual about it..
Ssh isn't handled by teslausb, so I can't help with that.

closed 2019-11-18 by marcone