marcone/teslausb #361: No files copied...

loop2, 2019-11-16:
Today I have create a new installation but on the destination directory on my server I can see only 4 files that aren't playable....

So I think that missing server files and the 4 files are corrupted....
My pastebin: [https://pastebin.com/ChNscKgi](https://pastebin.com/ChNscKgi)

thanks a lot!

marcone, 2019-11-16:
I don't see anything out of the ordinary in your logs. There were only 4 files to archive, and they were archived. Corrupt files are usually a problem with the car, and restarting the MCU will likely make that problem go away.

closed 2019-11-16 by marcone

loop2, 2019-11-17:
Thanks a lot marcone! Today I have reset the car, I've drive for about 30 minutes and when at home the system copied some files on the server... of about 30 files 2 are corrupted and the files copied are't complete... of the following 3 drives no files are copied.... i can exclude problems on the server side because is fully operational 24/24h... moreover I can see some glitches on the video that the system transfert... can be a slower writing speed on the raspi? the micro sdcard is a very good lexar 600x ...
Thanks a lot!

loop2, 2019-11-18:
This my archiveloop.log
Thu 14 Feb 10:12:43 GMT 2019: Starting archiveloop...
Thu 14 Feb 10:12:43 GMT 2019: mounting snapshot /backingfiles/snapshots/snap-000000/snap.bin
Thu 14 Feb 10:12:44 GMT 2019: mounted all snapshots
Thu 14 Feb 10:12:44 GMT 2019: low space, deleting /backingfiles/snapshots/snap-000000
Thu 14 Feb 10:12:50 GMT 2019: warning: low space for snapshots
Thu 14 Feb 10:12:50 GMT 2019: taking snapshot of cam disk: /backingfiles/snapshots/snap-000000/snap.bin
Thu 14 Feb 10:12:54 GMT 2019: took snapshot
Thu 14 Feb 10:12:55 GMT 2019: comparing /backingfiles/snapshots/snap--00001/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Thu 14 Feb 10:12:56 GMT 2019: making links for /backingfiles/snapshots/newsnap/mnt, retargeted to /backingfiles/snapshots/snap-000000/mnt
Thu 14 Feb 10:12:56 GMT 2019: made all links for /backingfiles/snapshots/newsnap/mnt
Thu 14 Feb 10:12:56 GMT 2019: Waiting for time to be set by ntpd...
Thu 14 Feb 10:13:23 GMT 2019: Time still not set, attempting to force it
Mon 18 Nov 18:34:46 GMT 2019: Ensuring music backing file is mounted...
Mon 18 Nov 18:34:46 GMT 2019: Mounting /mnt/music...
Mon 18 Nov 18:34:46 GMT 2019: Mounted /mnt/music.
Mon 18 Nov 18:34:46 GMT 2019: Ensured music drive is mounted.
Mon 18 Nov 18:34:46 GMT 2019: Running fsck on /mnt/music...
Mon 18 Nov 18:34:47 GMT 2019: | fsck from util-linux 2.33.1
Mon 18 Nov 18:34:48 GMT 2019: | fsck.fat 4.1 (2017-01-24)
Mon 18 Nov 18:34:48 GMT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Mon 18 Nov 18:34:48 GMT 2019: |  Automatically removing dirty bit.
Mon 18 Nov 18:34:48 GMT 2019: | Performing changes.
Mon 18 Nov 18:34:48 GMT 2019: | /dev/loop1: 180 files, 138254/560394 clusters
Mon 18 Nov 18:34:48 GMT 2019: Finished fsck on /mnt/music.
Mon 18 Nov 18:34:49 GMT 2019: Unmounting /mnt/music...
Mon 18 Nov 18:34:49 GMT 2019: Unmounted /mnt/music.
Mon 18 Nov 18:34:49 GMT 2019: Archiving...
Mon 18 Nov 18:34:49 GMT 2019: Disconnecting usb from host...
Mon 18 Nov 18:34:49 GMT 2019: Disconnected usb from host.
Mon 18 Nov 18:34:49 GMT 2019: Ensuring cam archive is mounted...
Mon 18 Nov 18:34:49 GMT 2019: /mnt/archive is already mounted.
Mon 18 Nov 18:34:49 GMT 2019: Ensured cam archive is mounted.
Mon 18 Nov 18:34:49 GMT 2019: Checking saved folder count...
Mon 18 Nov 18:34:49 GMT 2019: Ensuring cam file is mounted...
Mon 18 Nov 18:34:49 GMT 2019: Mounting /mnt/cam...
Mon 18 Nov 18:34:50 GMT 2019: Mounted /mnt/cam.
Mon 18 Nov 18:34:50 GMT 2019: Ensured cam file is mounted.
Mon 18 Nov 18:34:50 GMT 2019: Running fsck on /mnt/cam...
Mon 18 Nov 18:34:50 GMT 2019: | fsck from util-linux 2.33.1
Mon 18 Nov 18:34:52 GMT 2019: | fsck.fat 4.1 (2017-01-24)
Mon 18 Nov 18:34:52 GMT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Mon 18 Nov 18:34:52 GMT 2019: |  Automatically removing dirty bit.
Mon 18 Nov 18:34:52 GMT 2019: | Performing changes.
Mon 18 Nov 18:34:52 GMT 2019: | /dev/loop1: 124 files, 90035/1263255 clusters
Mon 18 Nov 18:34:52 GMT 2019: Finished fsck on /mnt/cam.
Mon 18 Nov 18:34:53 GMT 2019: There are 0 event folder(s) with 0 file(s) to move.
Mon 18 Nov 18:34:53 GMT 2019: Unmounting /mnt/cam...
Mon 18 Nov 18:34:53 GMT 2019: Unmounted /mnt/cam.
Mon 18 Nov 18:34:53 GMT 2019: Finished archiving.
Mon 18 Nov 18:34:53 GMT 2019: Music archive not configured or unreachable
Mon 18 Nov 18:34:53 GMT 2019: unmounting /mnt/archive
Mon 18 Nov 18:34:53 GMT 2019: Connecting usb to host...
Mon 18 Nov 18:34:53 GMT 2019: Connected usb to host.
Mon 18 Nov 18:34:53 GMT 2019: Waiting for archive to be unreachable...

marcone, 2019-11-18:
That log looks perfectly normal

loop2, 2019-11-18:
> That log looks perfectly normal

ok but why not copy to the server? the shared directory is visible from the pi (I tried it via ssh) where I can search the issue? Any idea?
thanks a lot!

marcone, 2019-11-18:
> ok but why not copy to the server?

Because there was nothing to copy:
```
Mon 18 Nov 18:34:53 GMT 2019: There are 0 event folder(s) with 0 file(s) to move.
```
You did read the documentation and see that only Sentry and manually saved events are backed up, right?