marcone/teslausb #367: Unable to mount CIFS share on Synology

Twiederh, 2019-11-23:
Hi,
I am unable to mount the CIFS share on my Synology that is attached to a Windows domain. I troubleshooted this and found that I can mount it, when the domain is specified with the username ans password. Right now it is only possible to specify the username and password in the config file, I failed to find the right spot to add the domain, also when I put it into the teslaCamArchiveCredentials file in /tmp this does not work. Could you please add this option to the configuration file? Apart from that - thanks for providing this on GIThub.

Many regards,
Thomas.

Twiederh, 2019-11-23:
Actually, it is more complicated than this: the mount in the test script works like this:
mount -t cifs '//nas1.wiederhol.de/Tesla' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials'
When I have the options set, I get the following message back: mount error(22): Invalid argument

marcone, 2019-11-23:
That doesn't make sense. You definitely don't want to expose usernames and passwords in /etc/fstab, and you certainly shouldn't *have* to, regardless of the server.

Twiederh, 2019-11-23:
I made it work in the meantime with the following changes: I changed the entry in /etc/fstab to //<server>/<share> /mnt/archive cifs credentials=/root/.teslaCamArchiveCredentials,auto 0 0 and added domain=<Windowsdomain> to the credentials file manually. Now I can at least mount the volume manually, though strangely it does not automount after a reboot.

-----------------------------------------------------------------------
Thomas Wiederhold
thomas@wiederhol.de<mailto:thomas@wiederhol.de>
Tel. +491796907106
-----------------------------------------------------------------------

marcone, 2019-11-23:
So the `domain=...` is the critical part here? I suspect that the lack of that might have also causes the "invalid argument" message, since the user would be invalid without the correct name (for cases where the server cares about that)
As for automounting: it's not supposed to automatically mount on boot. The archiveloop script will mount and unmount as needed.

Twiederh, 2019-11-23:
Without the domain= I get the following:
mount -t cifs '//<server>/<share>' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=3.0,'
mount error(13): Permission denied

with the domain= I get the Invalid Argument above. I suppose that the parameters changed with the latest mount.cifs command.

marcone, 2019-11-23:
> Without the domain= I get the following:
> mount -t cifs '///' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=3.0,'

To clarify: that's not the actual command line you used, right? (because it's missing the servername and path)

> mount error(13): Permission denied

And just to be sure, you did `sudo -i` before that, i.e. the "permission denied" is not just because you're trying to run this as a regular user?

> with the domain= I get the Invalid Argument above. I suppose that the parameters changed with the latest mount.cifs command.

Twiederh, 2019-11-23:
I used //<servername>/<share>, somehow It vanished in the reply. I entered the command with the root account active (sudo -i). The permission denied went away as soon as I put the domain in the teslaCamArchiveCrentials file. I put it into the hidden one under /root as well, that is how it is working now: three lines with username, password and domain in the file and //<server>/<share> /mnt/archive cifs credentials=/root/.teslaCamArchiveCredentials,auto 0 0 in fstab

Twiederh, 2019-11-25:
I have to apologize, the problem was a misconfiguration on my side. I managed to configure my Synology correctly and now mounting works flawlessly. Thank you very much.

closed 2019-11-25 by Twiederh

madpad33, 2019-12-29:
I had do activate SMB3 on the synology, after that it worked.

Jughead135, 2020-11-04:
@Twiederh Could you expand on what you did with your NAS, please?  I'm in the same boat with my Synology.  The Pi mounts just fine to a Windows folder, and it can "see" the NAS by IP (not by name, though); but I keep getting either "Permission Denied" or "Operation Not Supported" errors.  I've tried inputting the domain as you suggested, but no dice.  Really scratching my head here....

Twiederh, 2020-11-12:
Hi,
I just had an issue with teslausb that forced me to reinstall from scratch. This time it worked without any issues. If I remember correctly, I tried using a domain user and ended up using a local user on the Synology. The only point might be that you need to make sure that the user in question has access to the share - read and write.
In my case currently I have a group that has access to the share and the user inherits the access. It works fine like this.

Many regards,
Thomas.
--------------------------------------------------------
Thomas Wiederhold
Tel. +491796907106
thomas@wiederhol.de<mailto:thomas@wiederhol.de>
https://wiederhol.webex.com/meet/thomas
--------------------------------------------------------

Jughead135, 2020-11-13:
> The only point might be that you need to make sure that the user in question has access to the share - read and write.

Ultimately, we're saying the same thing.  My issue turned out to be that `/etc/fstab` didn't have the updated path, so the share was wrong.  Thanks for the follow-up!