marcone/teslausb #373: apt-get update "Couldn't create temporary file... for passing config to apt-key"

aberen001, 2019-12-08:
Hello,
I am getting an error when trying to run the "apt-get update" command. I re-flash and set up the Pi from scratch and it works for a couple days, but then I run into this error. Not sure where to start. Any ideas?

(ssh)
sudo -i
/root/bin/remountfs_rw
sudo apt-get update
<error>

marcone, 2019-12-08:
Make sure the filesystem wasn't made read-only due to an error (check `dmesg`)

githubsean, 2019-12-10:
I got this. I have no idea why, but /tmp was had 755 directory permissions instead of 777.
I just did the following and apt was able to run
`sudo chmod 777 /tmp`

marcone, 2019-12-10:
Seems like that shouldn't really matter if `apt` is run as root though, should it?

githubsean, 2019-12-10:
Oh yeah... of course... duh. 
I did run apt as root, and I did get that error, it was my brute force solution to something saying "I can't access X" - chmod 777 X. 
I'll have a play later tonight (in 12 hours Sydney/Australia) it seemed easy enough to reproduce.

aberen001, 2019-12-14:
"sudo chmod 777 /tmp" got rid of the error. Would love to figure out the why to fix it permanently

marcone, 2019-12-30:
The likely problem was that the permissions for /tmp got changed because something changed the permissions of /var/spool, which was symlinked to /tmp.

I made /var/spool its own tmpfs filesystem instead of symlinking it, so this should be fixed now.

closed 2019-12-30 by marcone