marcone/teslausb #377: Command not found errors in teslausb-headless-setup.log

GoTeslaGo, 2019-12-13:
Pi is connected to the internet. this is a small section from teslausb-headless-setup.log

Note there are 216 such lines of the same in this file and nothing else

/boot/teslausb-headless-setup.log: line 208: Fri: command not found
/boot/teslausb-headless-setup.log: line 209: Fri: command not found
/boot/teslausb-headless-setup.log: line 210: Fri: command not found
/boot/teslausb-headless-setup.log: line 211: Fri: command not found
/boot/teslausb-headless-setup.log: line 212: Fri: command not found
/boot/teslausb-headless-setup.log: line 213: Fri: command not found
/boot/teslausb-headless-setup.log: line 214: Fri: command not found
/boot/teslausb-headless-setup.log: line 215: Fri: command not found
/boot/teslausb-headless-setup.log: line 216: Fri: command not found

i have run sudo /etc/rc.local to restart setup but this just adds 2 more of the same lines as above

also tried running /root/bin/setup-teslausb diagnose > /mutable/diagnostics.txt but get the following error:

-bash: /mutable/diagnostics.txt: No such file or directory

have also rebooted the Pi but same issue

have tried with the Pi plugged into my computer via USB and plugged into the wall so connected via wifi

marcone, 2019-12-14:
> /boot/teslausb-headless-setup.log: line 208: Fri: command not found
> /boot/teslausb-headless-setup.log: line 209: Fri: command not found

This looks like what you would get if you tried to run the log file, e.g. if you typed `/boot/teslausb-headless-setup.log` on the command line.

GoTeslaGo, 2019-12-14:
So I should run the tail -f command instead?

marcone, 2019-12-14:
yes

closed 2019-12-14 by marcone