marcone/teslausb #382: Teslausb - pi not being recognized after update 2019.40.50.1

miles267, 2019-12-24:
Has anyone else seen this?  I've tried rebooting the car, the pi, and reconnecting usb cables but the dashcam icon no longer appears with Teslausb after this update today.

marcone, 2019-12-24:
Still works for me. Did you plug it into a PC to see if the drive shows up OK and still has the TeslaCam folder?

yaroslavros, 2019-12-24:
Same issue here. Recognised correctly in PC and do not see any issues with g_mass_storage however Tesla model 3 with 2019.40.50.1 does not consider it to be a valid storage.

Using RPI4 with external USB disk.

senf42, 2019-12-24:
It still shows up for me as well.

yaroslavros, 2019-12-25:
Did complete wipe/reinstall and things seem to be back on track.

A bit strange coincidence with 2019.40.50.1 upgrade...

mattfrautnick, 2020-01-02:
I am having this issue even after a full wipe/reinstall. P3D- running 2019.40.50.5

ScottESanDiego, 2020-01-03:
Mine broke (fast-flashing LED, not visible as a umass device when connected to a PC), I re-imaged and it worked for about a day - then back to the same behavior.  Something is certainly odd since 2019.40.50.* .

marcone, 2020-01-03:
@ScottESanDiego fast-flashing LED means it's archiving, and during archiving it does not present as a mass storage device.

ScottESanDiego, 2020-01-03:
I don't think it's that fast-flash though.  It doesn't connect to WiFi (so I cannot SSH into it), and it stays in that state for over a day (not sure quite how long, but at least a day).

ozvic, 2020-03-09:
I've had mine drop out a couple of times. Rather than starting from scratch, I plugged into Linux box and ran xfs_repair. ie. Mount, unmount, xfs_repair. Or xfs_repair -L if mounting doesn't work. Can't seem to do it directly on the Pi as Device busy and locks etc. Possible to have script run xfs_repair (optionally) on reboot?

marcone, 2021-03-09:
Nothing actionable in this issue, so closing.
@ozvic to run xfs_repair on boot, create the file /forcefsck

closed 2021-03-09 by marcone