marcone/teslausb #384: Tesla API to keep your car awake

mboerma, 2019-12-26:
This might be more of a remark or question than an issue. I have my car (An EU 100D Model S model year 2020) in Sentry mode and also have the "keep car awake" settings set for teslausb. 

Maybe this is just new to the 2019.40.x firmware, but the USB ports are no longer powered. They only work when I turn on conditioning. 

Q1) Do other people also notice this?
Q2) Is this new behaviour or was that the case all along for Model S Tesla's?
Q3) What does the "Keep Tesla awake" API call do?
Q4) Is there any way I can make things work (again)? USB battery? Other settings?

Many thanks!

simonrb2000, 2019-12-30:
I’m on 2019.40.50.1 and enabling sentry mode keeps my USB ports powered. Try a reboot (both scroll wheels)

mboerma, 2019-12-30:
Thanks. I did try that. No change. Do you have a 2020 Model S?

simonrb2000, 2019-12-30:
I have a 2019 Model X. Should be the same
MCU so should have the same behaviour. I manually turn on sentry mode when I get home and once the pi has finished uploading it turns off Sentry Mode then the car goes to sleep.

mboerma, 2019-12-30:
I also have Sentry mode on manually. I want it to stay on since I park on the street in Amsterdam. But the sentry mode does not prevent it from powering down the USB ports since the latest 2019.40.50.1 update or at least lowering the power on them. I now have a USB battery as backup.

mboerma, 2019-12-30:
I have to check one thing which I changed. I installed [an organizer in the center console](https://www.amazon.de/gp/product/B07N3W7YJD/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1). This plugs into the cigarette lighter thing. Maybe that has caused some minor changes to the power of the USB ports when in Sentry mode.

mboerma, 2019-12-30:
Nope didn't help. So either something has happened with the teslausb software of with the Tesla firmware. :-(

simonrb2000, 2019-12-31:
I have a 4 port hub plugged into the cars data USB port and my TeslaUSB plugged into that. No issues here. Very strange.

eddieyee, 2019-12-31:
My USB ports powers down after about 2 minutes on sentry mode now too (2019.40.50.1).  Tried both the data ports and have the same issue.  Running on a Pi4 (same happens on a Pi0w) on a Model X 2019 (raven).   I've also been having a lot of trouble with Sentry while out and about.  Pretty much not usable.  I've only had a sentry clip after leaving the car and sentry activates within a minute of me leaving.

marcone, 2019-12-31:
> My USB ports powers down after about 2 minutes on sentry mode now too

To clarify: do you mean normal Sentry mode (i.e. when you're parked somewhere and turn on Sentry mode to keep an eye on things), or when teslausb turns on Sentry mode during archiving?

eddieyee, 2019-12-31:
When I'm at home and activate sentry mode manually (it's set not enable at home) I'll see the teslausb pop up on my network (pingable) for a little under two minutes.   I'll have to confirm  how long it takes to turn off when Sentry is automatically enabled when I'm not at the house, but it does seem around two minutes.

mboerma, 2019-12-31:
Same for me. I have Sentry mode on "manually" / by default. And also have the setting configured in the config file. I can see in the log (/mutable/archiveloop.log) that Sentry mode is turned on and switched back to the previous state when I run the Pi from a battery or when I have it on my desk.

Two minutes seems to be right. I now also notice it when I turn on the Climate manually from behind my desk. The Pi turns off.

marcone, 2019-12-31:
Very odd that USB turns off even when Sentry mode is on. You'd think that something that uses the USB ports would keep them powered. Perhaps with 2019.40.50 the car decides to turn off the USB ports because it doesn't see a USB drive (since the drive is not exposed to the car during archiving). I may have to switch to archiving a snapshot instead, so the car is happy, or go back to the old way of keeping the car awake by sending a "wake up" event every minute or so.

mboerma, 2020-01-01:
> Very odd that USB turns off even when Sentry mode is on. You'd think that something that uses the USB ports would keep them powered. Perhaps with 2019.40.50 the car decides to turn off the USB ports [...]

Maybe that could be it. I had a thumb drive installed previously with music on it. I removed it. I will see if that makes a difference. Following your thinking this might help to keep the Tesla powering the USB ports.

mboerma, 2020-01-01:
@marcone That seems to do the trick! I no have a small Sandisk thumbdrive installed plus the Pi Zero in the other USB port. It has now been running for 10 minutes archiving clips.

simonrb2000, 2020-01-01:
I’m on 2019.40.50.1 and now .7 and .1 hasn’t changed in behaviour on my 2019 MX. Maybe I am lucky. Haven’t tested .7 yet. Although in my garage sentry mode is constantly activating (headlights and HAL image) for some reason.. maybe that’s keeping my USB ports powered.. who knows.....

There is camp mode now.. if that has an API then maybe use this to USB whilst it achieves?

Cheers

mboerma, 2020-01-01:
Maybe there is a difference in hardware between 2019 and 2020 models. My car is parked on the streets of Amsterdam. With sentry mode always on. That does not prevent it from switching off the USB ports when is doesn't see a USB drive. See the post by @marcone on this. I can only support this hypothese because I now have a thumb drive installed next to the Pi Zero and it has been running now for over an hour.

simonrb2000, 2020-01-01:
I did read his post before my reply.

Maybe there is a change between 2019/2020. My 2019 is MCU2 and AP3 (FSD) which is what the 2020 models have.

Anyways, hopefully there is a fix coming.

Cheers

marcone, 2020-01-03:
I tried reproducing this on my 2018 S running 2019.40.50.7, but it stays awake when Sentry mode is on, even if the drive is not connected to the car. With Sentry off, the USB ports turn off after about 2 minutes.

mboerma, 2020-01-04:
I just upgraded to 2019.40.50.7. It has the same behaviour. I now even switches of WITH another USB thumb drive in the other port. Maybe it is even looking for a TeslaCam folder on that drive and can't find it and thus switches of the power. 

Manually switching Sentry Mode in the Tesla app does turn the Pi on for a minute or two, then switches off again even with Sentry mode on (I always have that on).

simonrb2000, 2020-01-04:
Very strange. I’m on same firmware and yesterday my car behaved as always (same as marcone). It uploaded all footage and turned off Sentry mode when finished.

🤷‍♂️

mboerma, 2020-01-04:
It now even falls asleep with Sentry mode turned on. I’ll post a message on the Tesla Motors club forum to see if others have similar issues.

simonrb2000, 2020-01-04:
Defo doesn’t sound right that it goes to sleep whilst on Sentry mode! It keeps the AP computer awake so car can’t sleep. Have you tried to trigger sentry mode once car sleeps?

mboerma, 2020-01-04:
You mean walk up to the car without a key fob and see if it triggers (flashes lights)? I will try. 

I agree, it should stay awake even if it can’t record, or thinks it can’t record, videos. I’ll keep monitoring it and doing a soft and hard reset if needed.

simonrb2000, 2020-01-04:
Yes, that’s what I meant. 👍

mboerma, 2020-01-04:
No. It just sleeps... Just did a soft reset. Earlier a hard reset. Turning on the HVAC switches the USB ports on, temporarily. The USB drive "trick" no longer works. :-(

simonrb2000, 2020-01-04:
So Sentry Mode is enabled (circle icon on top of screen is red and not black) but car sleeps and doesn’t activate when your near it without a key? (headlight flash and recording message on screen).

Cheers

mboerma, 2020-01-04:
Correct.

simonrb2000, 2020-01-04:
That sounds like an issue with your car. I’d open a service request. I know how you feel; these new voice commands don’t seem to be working for UK owners 😔

Cheers

mboerma, 2020-01-04:
Yes, that is probably best. I might get support by the end of 2020....

The voice commands don't work anymore here in the Netherlands either. The yankees are the only one cheering the latest updates, I think.

simonrb2000, 2020-01-04:
Lol to the support. It’s the same over here.

Watched a YouTube video from Australia and it works really well for them.

eddieyee, 2020-01-05:
I am still having sentry mode issues on 2019.40.50.7, and I'm thinking it's due to the time it takes to copy the files to the nas.   In my config I do not have snapshots enabled, and it was copying over about 4gb of sentry mode clips.   Just a thought--   if sentry mode is enabled and it does not see the TeslaCam directory it shuts off in two minutes.  When it is copying files to the NAS, is the TeslaCam folder available to the vehicle?

marcone, 2020-01-05:
> When it is copying files to the NAS, is the TeslaCam folder available to the vehicle?

No it's not, but this didn't used to matter (and still doesn't, for me)

Radishworks, 2020-01-09:
The Tesla App wakes the USB somehow...

I thought I'd jump in on this and point out something I've noticed to see if anyone has any ideas?  I have my own server that talks to my MS and I'm using this project (thanks marcone!) to run the RasPiZero to collect sentry videos.  I've been working on pulling the sentry videos from the RasPiZero (via FTP), rather than have them pushed by the Zero.  This keeps all the login credentials off the Zero, and the thought is I can better keep the car awake via my server rather than the Zero trying to do it.  Just like this thread describes its a trick to keep the USB port powered up via the API.  

Here's what I noticed that's interesting:  The Tesla App itself seems to power up the USB as soon as you open it and look at the car.  You would think this was just sending the /wake_up API call, but apparently not.  Has anyone poked at the Tesla App to see what its sending to wake up the USB just by opening the App?  Do M3's do the same?

mboerma, 2020-01-09:
@Radishworks With my Model S. The USB ports don't power up just opening the Tesla app. I have to turn on the climate control or switch the sentry mode on-off to do that. Apparently if check for a TeslaCam folder on connected USB drives. If those are not found it powers the USB ports down within a couple of minutes (two seems to be right).

Radishworks, 2020-01-09:
@mboerma, hmm interesting.  For me I can stand outside the car, open the Tesla App and see the USB power up.  If the App opens to the our Powerwall screen (because it was last open) nothing happens to the car, but as soon as I swipe to the car screen the USB powers up.  My MS is 2018, with 2.5 hardware, and v2019.40.50.7 (although I've noticed this behavior with a lot of the updates).    

I wonder...  do you have FSD/Summon on your car?  I seem to recall, a while back Tesla sped up the summon response by waking the car faster (maybe more?) with the App.  Could FSD/Summon be the difference?

mboerma, 2020-01-09:
@Radishworks I don't have FSD. Not of much use in the Netherlands for the money you pay for it. Mine is a Model S from model year 2020. 3.0 hardware I suppose I have no idea how to check that. Same firmware as you.

Radishworks, 2020-01-09:
It would be interesting to hear if anyone else can confirm what I'm seeing?  It would be great to find an API call that would switch on the USB for everyone.  I can say that its not just the  /wake_up API call, as I'm sending that via my server and it doesn't turn on the USB.

marcone, 2020-01-09:
@Radishworks my car behaves the same as yours when opening the Tesla app on my phone, and always has. It is also a 2018 S with 2.5 hardware.

Radishworks, 2020-01-09:
@marcone, FSD/Summon?

marcone, 2020-01-09:
EAP (which includes summon), no FSD

marckatznelson, 2020-01-17:
> I've been working on pulling the sentry videos from the RasPiZero (via FTP), rather than have them pushed by the Zero. 
Hi - I have been looking for this solution as well; I have been facing lots of issue with car sleeping off, upload issues etc.
How did you manage to FTP and pull the recordings? Can you please provide some instructions? that is awesome,  I myself have been asking it on different threads, and not one seems to know how to do it. Thanks!!

Radishworks, 2020-01-17:
@marckatznelson, sorry no magic bullet here.  Its pretty easy to set up a FTP on the zero, lots of examples around for that.  I have some php code on my server FTPs into the zero and loops through the videos and transfers them.  Had a proof of concept working but got hung up on the car going to sleep.  I think a part solution, at least for those who have FSD/EAP, is to figure out what the Tesla app is doing to wake up the USB, as discussed above.  I've poked at it a little but haven't focused a huge effort on it yet, if I find a solution I will post it.

Radishworks, 2020-01-24:
I can't take credit for this, but there may be a really simple solution to keeping the cars awake while coping files.  I stumbled on this article RE: camp mode, here: https://www.hacker9.com/tesla-camper-mode-hack-usb.html

I've run a couple of tests to keep the car awake, first with a mouse plugged into the USB port and kept it moving, this didn't seem to work.  But! when I plugged a keyboard into a USB port and tapped keys I was able to keep the car awake for over 5 minutes, instead of the usual 2.  A keyboard seems to work!

Since the RasPiZ can emulate a USB keyboard, the trick might be to switch the RasPiZ from a thumbdrive to a keyboard that's tapping on keys while files are being copied.  Seems to be plenty of articles on the Google machine about emulating keyboards with RasPiZs.    

I will play with this in my spare time, but I wanted to post it in case someone beats me to using this trick.

marcone, 2020-01-25:
Not a bad idea, but the article is about using it while camping/sleeping, when somebody is in the car. Does it work when nobody's in the car? (were you inside or outside the car when you tried this?)

If implemented this way, it would also simplify the logic to save/restore sentry state.

Radishworks, 2020-01-25:
When I tried this with a USB keyboard I was not in the car, I put the keyboard out the window and was tapping keys while watching the blinking light on the RasPiZ.  At least for 5+ minutes tapping keys seems to keep the USB power on.

eddieyee, 2020-01-25:
If you toggle sentry mode off then on every minute it will keep the USB enabled.  I think it resets the two minute timer it is waiting for the TeslaCam file system.

Perhaps this might be a good way to keep the system alive programmatically.

Radishworks, 2020-01-25:
@eddieyee - Problem with switching sentry on/off is you need to have all your login credentials on the Pi.   This is a security risk if the car was stolen, especially if you have more than one Tesla on your app.

mboerma, 2020-01-25:
Would be great if this can be overcome. At the moment with my car the teslausb archiving part doesn't work. I tried two Pi Zero's and multiple configurations.

marcone, 2020-01-26:
If anyone has successfully kept the car awake by making the Pi simulate keyboard presses, please let me know because so far I haven't managed to do so. It appears the car just stops reading keyboard input, and the Pi times out writing keypresses.

mboerma, 2020-01-26:
Not sure if this is what you mean with keyboard presses, but this is what I tried:
Toggled Sentry mode so the Pi Zero woke up and connected to the wifi. It started the archive loop. I connected with SSH to the Pi and started typing some random "commands" and pressed enter every few seconds. Still after two minutes or so the connection was gone and the Pi was powered off.

Radishworks, 2020-01-26:
I went back and re-tested several physical USB keyboard, mice and even PlayStation Controllers, I'm sorry to say, I can't get it to work again.  When I had it working with the keyboard, 5 min+, I must have been fooled by something else, I was using my phone's timer, maybe the T-App was keeping it alive?  Sorry, my bad and my apologies.  Anyhow, I did a little more research and there is a ton of posts and videos of mice being used in Teslas, but it appears, by reports that latest updates have disabled them.  This is understandable for security reasons.  Now I'm wondering if that Mouse Jigglier really works at all anymore, even though the article is from 2020.

Will keep looking for other ways to solve this issue.

marcone, 2020-01-26:
It looks like USB keyboards don't work at all with the car. USB mice used to work apparently, but it looks like Tesla disabled that at some point.

eddieyee, 2020-01-26:
So when the TeslaCam folder becomes unavailable, the car essentially goes to sleep after two minutes, right?  How about providing a throwaway TeslaCam folder for the vehicle while the real TeslaCam directory gets synchronized?   Alternatively instead of using the sentry mode to keep the car awake, use camp mode?

Radishworks, 2020-01-27:
@eddieyee - "... the car essentially goes to sleep after two minutes, right?"  Yes, more importantly the USB power shuts down after 2 minutes of getting out of the car, hence shutting down the pi.  Creating a throw away directory might help avoiding extra clips when the car is syncing, but it still won't solve the problem of the USB power turning off.  RE: Camp mode, problem here is there is no API call that will put the car into Camp Mode, at least not one I've been able to find.  General rule is if the Tesla App can't do it there isn't an REST API call for it.  Regardless, even if there was a camp mode API call, it would still require your Tesla login password to be on the pi, security risk as mentioned above.  

We are all looking for some simple elegant solution, if there is one.

Poncherelly, 2020-01-27:
Not an elegant one but having a battery connected to the pi would get to extra time. I don't have my first TESLA yet but was hoping to check to see if there were any fuses powered only while charging. If there was, then I would plug in my dashcam battery pack. While charging (in my garage) the pi would stay powered and transfering.  While driving, the battery pack wouldn't pull from the car and wouldn't affect range but the pi would still be plugged in to the data USB port for power if the pack goes dead. 

The two unknowns at this time are:

1) There's is a fuse only powered when the car is charging 
2) the pi can be powered by the battery pack and if the pack dies can gracefully move to USB power through the data Cable. 

Again not ideal, but a thought.

marcone, 2020-05-14:
@mboerma is this still a problem with current car firmware?
If so, do the USB ports in your car turn on when you open the Tesla app on your phone?

mboerma, 2020-05-15:
@marcone I haven't tried for some time. I barely drive the car these days and haven't update the software. I will do so this weekend and see what happens. Anything I need to look at? Settings?

marcone, 2020-05-15:
No, just open the app and see if that powers up the USB ports.

mboerma, 2020-05-15:
Open the app? There is an app? Or do you mean the Tesla app?

I just did a full update and will put it in the car and see if I can keep the connection alive.

marcone, 2020-05-15:
Yes I meant the Tesla app. So basically:
- plug something (e.g. the Pi, or a USB powered light) in to one of car's USB ports, so you can see when the port is powered
- get out of the car,lock it, and wait for the USB ports to power down
- open the Tesla app on your phone
- check whether or not the USB ports are now powered

[61 more replies]

closed 2021-06-26 by marcone