marcone/teslausb #385: Make it compatible with SentryView IOS app

plmilord, 2019-12-26:
Hello,

Option request… Is it possible to make it working with SentryView IOS app? They have an option to view clips through an AP from a SanDisk Connect Wireless Drive! As, within the TeslaUSB project, there is possibilities to enable the AP function and share /TeslaCam folder with guest/guest, we are very near to make it working as an SanDisk Connect Wireless Drive!

I also asked to SentryView app developers to make it work with TeslaUSB project!

…This will rock everything!

Thank-you,

Pierre-Luc

(...if it's already compatible, let me know how make it working properly! Have tried many things, but no luck for now!)

marcone, 2019-12-26:
I don't have an IOS device, and don't know what SentryView expect, so there isn't anything I can do here. TeslaUSB already makes the SentryClips folder available in a Samba share.

plmilord, 2019-12-27:
Have tried to modify samba smb.conf file, but I'm stuck with a read only file system!? My thought... the current samba share /TeslaCam require l/p: guest/guest, and the SentryView app seems to require Anonymous login without password.

It's why I tried to add to smb.conf:

>   public = yes
  only guest = yes
  browseable = yes
  read only = no
  writeable = yes
> 

Have you any idea how to bring those changes to smb.conf file? I will try different things around that!

Thx.

marcone, 2019-12-27:
I'm not sure where you got the idea that the default user/pass is guest/guest.
The default is pi/raspberry (just like for logging in), and you can enable guest mode by specifying `export SAMBA_GUEST=true` in the config file, which is explained right in the sample config file.

plmilord, 2019-12-27:
While `SAMBA_GUEST=true`, I need to identify myself as guest/guest to get access the shared folder /TeslaCam.

With those setting enumerated earlier, I want to make that share without login/password.

marcone, 2019-12-27:
hmm, when I enable SAMBA_GUEST, I can log in with either guest/guest (didn't know that), but I can also just check the "guest" box on MacOS, and Windows doesn't even ask me to authenticate at all, so I really think the "without login/password" mode you're looking for is the same as the guest mode you can enable with SAMBA_GUEST.

simonrb2000, 2019-12-30:
Firstly there is an app that works on iOS with TeslaUSB already. Secondly does disabling snapshots disable the Samba share option? 

Cheers

plmilord, 2019-12-30:
Don't know about the snapshot... what is the name of the ios app!?

Any idea how to disable the SSID broadcast?

plmilord, 2019-12-30:
Got it for the ios app...: `Cam Viewer for Tesla`!

I will give it a try!

aerohoff, 2019-12-30:
How do we point Cam Viewer at TeslaUSB?

simonrb2000, 2019-12-30:
Connect to your TeslaUSB pi’s WiFi in the car. Then you should be able to browse to the TeslaCam folder.

closed 2019-12-31 by marcone

deadlyknapsack, 2020-01-20:
@simonrb2000 I'm struggling a little bit with this. When I connect to the Pi's AP and I open Cam Viewer, I don't see the option to browse the folder on the Pi. Any directions on how to set it up?