marcone/teslausb #387: New Feature Multiple WIFI Configurations

repocho, 2019-12-28:
Adding the possibility to have more than one WIFI Connection in addition to the posiblity to connect through OpenVPN (issue #386) will allow to be able to copy the files in multiple locations.

marcone, 2019-12-29:
I'll probably add this at some point.

beckerstef, 2019-12-29:
Edit /etc/wpa_supplicant/wpa_supplicant.conf there you can insert more then one WLAN

Sample:

country=DE 
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev 
update_config=1 
network={
     ssid="wlan1"
     scan_ssid=1
     psk="passwort1"
     key_mgmt=WPA-PSK
}
network={
     ssid="wlan2"
     scan_ssid=1

pyjamasam, 2020-02-03:
On thing I also did did was setup a symlink from `/etc/wpa_supplicant/wpa_supplicant.conf` to `/boot/wpa_supplicant.conf`

That way I am able to just pull the SD, pop it in my computer and add or remove networks quite easily.

And yes as @beckerstef said in their comment, you can add multiple networks and the Pi will connect to what ever it sees.

YourMom11, 2020-02-14:
Could I trouble you to provide this NOOB the terminal instructions for all this?

YourMom11, 2020-02-26:
This works great. I also added id_str="AP1" field to each network to ensure DHCP. You can reference the interface in /etc/network/interfaces.

i.e.
iface AP1 inet dhcp
iface AP2 net dhcp

And for those brave enough, you can add the following to /etc/wpa_supplicant/wpa_supplicant.conf as a way to connect to open/non-secured networks:

network={
    key_mgmt=NONE
    priority=-999
    }

u4ik2020, 2020-03-31:
Adding on to this.  I would love to be able to copy the files on a public WiFi?  Is this secure enough.  The primary advantage I am trying to gain, is that if someone damages/steals the car I would have some change that the raspberry would update info at some point showing the perpetrator(s), assuming I haven't just punched a big hole in the security side.

proddick, 2020-04-26:
If you connected to a public WIFI where and how would you copy the files?  

My Pi is configured to copy to my home server and when I am away at friends/hotel/etc (not at present of course!) I use a tiny WIFI box that has a VPN back to my home.  The WIFI box has the same SSID so the Pi connects and uploads - slow, but it does seem to work.

YourMom11, 2020-04-26:
Public wifi connections are helpful to those using RCLONE for archiving rather than CIFS. I’m backing up directly to Dropbox via RCLONE

pyjamasam, 2021-08-30:
Hmm.  Interesting.  Its been a while since I have played with this stuff (or even run it).

Does `/etc/wpa_supplicant/wpa_supplicant.conf` exist on disk?  (I mean I am assuming it does but can't hurt to check).  Another thing might be that the /boot drive is mounted as read only.  That happens some times with some distros, but i can't remember if its the case for this one (though that might have changed since the version I was playing with to something more recent).

And now that I read my comment better from before and refresh my memory...  You need to have the real file sitting on the /boot partition.  So that you can edit it elsewhere, and then the link is what lives in the /etc/wpa_suppliant directory.  ln (at least on my mac currently) takes source_file as the first parameter and link_name as the 2nd.  So you have your arguments backwards.  Don't know if that would result in the error or not.

Hopefully some of these bits help you.

marcone, 2021-08-30:
> When I issue after sudo -i & while in /etc/wpa_supplicant folder
> `ln -s /etc/wpa_supplicant/wpa_supplicant.conf /boot/wpa_supplicant.conf`
> Error is - `ln: failed to create symbolic link '/boot/wpa_supplicant.conf': Operation not permitted`
> 
> When trying from within /boot folder
> `ln -s /etc/wpa_supplicant/wpa_supplicant.conf /boot/wpa_supplicant.conf`
> Error is exactly the same - `ln: failed to create symbolic link '/boot/wpa_supplicant.conf': Operation not permitted`
> 
> Can you assist please?

As @pyjamasam mentioned, you have it backwards. The commands you typed above will attempt to create a symbolic link on /boot that points to /etc. That won't work for two reasons:
1. /boot is a FAT32 partition and does not support symbolic links
2. the file needs to be on /boot (so that it can be edited on the PC), and /etc/wpa_supplicant/wpa_supplicant.conf needs to be a symbolic link to /boot/wpa_supplicant.conf

cleanev, 2021-08-30:
Thank you @pyjamasam and @marcone. Appreciate your input and specifics to get this working. Below command worked on creating symbolic link

`ln -s /boot/wpa_supplicant.conf /etc/wpa_supplicant/wpa_supplicant.conf`

Where wpa_supplicant.conf was moved over to /boot
Take care

marcone, 2022-09-26:
> I'll probably add this at some point.

Almost 3 years later and I still haven't gotten around to it, mainly because I have no use for the feature myself.
Since it's fairly easy for people to manually add additional SSIDs to connect to after initial install, it doesn't seem worthwhile to complicate setup for this. I'm going to close this issue, but happy to accept a PR to add this.

closed 2022-09-26 by marcone

iCSpotRun, 2023-03-08:
I know this is closed, but it may be worthwhile to add instructions for this under the "one step..." or clone instructions.

Thanks for the great work