marcone/teslausb #388: not able to finish the setup

L07tB0y, 2019-12-29:
I just flashes the latest 2.0 and when going through he setup it does not create the cam and music folders if I run the rc.local I get the following before it errors out. Pi Zero is able to connect to my wifi and the OS seems to be working fine as I can SSH in. 

create-backingfiles: cam: 200GB, music: 100%, mountpoint: /backingfiles
create-backingfiles: cam: 200GB, music: 100%, mountpoint: /backingfiles
Delete snapshots and recreate recording and music drives? (yes/cancel)yes
archiveloop: no process found
umount: /mnt/cam: no mount point specified.
umount: /mnt/music: no mount point specified.
umount: /backingfiles/snapshots/snap*/mnt: no mount point specified.
/tmp/create-backingfiles.sh: line 40: 200*1024*1024*1024B: value too great for base (error token is "1024B")
/tmp/create-backingfiles.sh: line 71: / 1024 : syntax error: operand expected (error token is "/ 1024 ")

marcone, 2019-12-29:
Use `200G`, not `200GB` to specify the cam size
(though I'd recommend using a much smaller value, unless you really expect to get 25+ hours worth of Sentry events between archives)

closed 2019-12-29 by marcone

L07tB0y, 2019-12-29:
Yeah got that figured out before I saw your email :) now the issue is a little different I got the P I working and the car 2020 Model S sees the drive (sort of) as the dash cam icon appears but it does not record anything if I press the dash cam button it takes a while and than dash cam icon comes up with an x icon for a sec. The funny thing is even if I unplug the usb from the car it keeps showing the dash cam icon. I have tried using two usb cables and even a cigarette lighter power Bec than I can move the cable to computer and back to Tesla and when I plug it into the computer I see the drives mount and I can put stuff in them etc but when I go back to Tesla under music USB it says not connected I get the icon for dash cam but nothing gets recorded. I can ssh to the pi so I know it is booted up. I am running 2019. 40.50.1 and have tried both the 2.0 and 1.3 both do the same thing.

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

marcone, 2019-12-29:
"even if I unplug the usb from the car it keeps showing the dash cam icon" seems to be pointing at problems unrelated to teslausb.

L07tB0y, 2019-12-29:
If I plug in a normal the TESLA behaves normal it sees the icon and records and if I unplug the icon goes away it is only when I use the TESLAusb I get the phantom dashcam icon.

Sent from ProtonMail Mobile