marcone/teslausb #389: Setup completed, music works, but not archiving video

hu-a, 2019-12-29:
We got the music to sync, but it won't upload video to the archive. Here's the various log files:

[Archive Loop](https://pastebin.com/EeQnjDjc)

[Setup Log](https://pastebin.com/FypXWLwn)

[Setup .conf file](https://pastebin.com/1fvjkMN8)

We tried to check if there was a problem writing to the server, so we ssh'd into the pi and manually mounted the archive and tried to write files to it. The result was that when tried using `cp` it was able to copy files over to the server, but when we used `mv` we got the error 

`mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-back.mp4': Operation not supported`

```
root@teslausb:~# mount -t cifs //GEORGE-ENVYX360/TeslaCam /mnt/cam -o user=tesla
Password for tesla@//GEORGE-ENVYX360/TeslaCam:  *****
root@teslausb:~# cd /backingfiles/TeslaCam/SavedClips/
root@teslausb:/backingfiles/TeslaCam/SavedClips# ls
2019-12-27_15-23-23
root@teslausb:/backingfiles/TeslaCam/SavedClips# mv 2019-12-27_15-23-23/ /mnt/cam
mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-back.mp4': Operation not supported
mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-front.mp4': Operation not supported
mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-left_repeater.mp4': Operation not supported
mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-right_repeater.mp4': Operation not supported
root@teslausb:/backingfiles/TeslaCam/SavedClips# mv 2019-12-27_15-23-23/ /mnt/cam/
mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-back.mp4': Operation not supported
mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-front.mp4': Operation not supported
mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-left_repeater.mp4': Operation not supported
mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-right_repeater.mp4': Operation not supported
root@teslausb:/backingfiles/TeslaCam/SavedClips# ls /mnt/cam
2019-12-27_15-23-23  This is on the envy
root@teslausb:/backingfiles/TeslaCam/SavedClips# mv 2019-12-27_15-23-23/
mv: missing destination file operand after '2019-12-27_15-23-23/'
Try 'mv --help' for more information.
root@teslausb:/backingfiles/TeslaCam/SavedClips# cd 2019-12-27_15-23-23/
root@teslausb:/backingfiles/TeslaCam/SavedClips/2019-12-27_15-23-23# ls
2019-12-27_15-22-18-back.mp4   2019-12-27_15-22-18-left_repeater.mp4
2019-12-27_15-22-18-front.mp4  2019-12-27_15-22-18-right_repeater.mp4
root@teslausb:/backingfiles/TeslaCam/SavedClips/2019-12-27_15-23-23# mv 2019-12-27_15-22-18-front.mp4 /mnt/cam/2019-12-27_15-23-23/
mv: cannot create symbolic link '/mnt/cam/2019-12-27_15-23-23/2019-12-27_15-22-18-front.mp4': Operation not supported
root@teslausb:/backingfiles/TeslaCam/SavedClips/2019-12-27_15-23-23# cp 2019-12-27_15-22-18-front.mp4 /m
nt/cam/2019-12-27_15-23-23/
root@teslausb:/backingfiles/TeslaCam/SavedClips/2019-12-27_15-23-23# rm 2019-12-27_15-22-18-front.mp4
root@teslausb:/backingfiles/TeslaCam/SavedClips/2019-12-27_15-23-23# ls
2019-12-27_15-22-18-back.mp4           2019-12-27_15-22-18-right_repeater.mp4
2019-12-27_15-22-18-left_repeater.mp4
root@teslausb:/backingfiles/TeslaCam/SavedClips/2019-12-27_15-23-23#
```

marcone, 2019-12-29:
Oops, that got broken with a recent change (https://github.com/marcone/teslausb/commit/7827a8c58b742dde7425adbcd03ff680abef8117).

I just pushed a fix, so either reinstall, or:
```
sudo -i
/root/bin/setup-teslausb selfupdate
/root/bin/setup-teslausb
```

miteecsman, 2019-12-29:
Yea - that worked!  The selfupdate didn't work but a reinstall did get the correct version.

closed 2019-12-31 by marcone