marcone/teslausb #391: /etc/fake-hwclock.data not found on boot

nickbruton, 2020-01-02:
Hi, having problems getting the date and time to set correctly on my pi zero w
On boot is says it can't find /etc/fake-hwclock.data which does exists as a lint to /mutable/etc/fake-hwclock.data which the means the date and time is incorrect, have you any idea how to fix this?
I do have a diagnostics.txt file if you need it and I can figure out how to attach it
Thanks

marcone, 2020-01-02:
> On boot is says it can't find /etc/fake-hwclock.data

Where are you seeing this message?

> which does exists as a lint to /mutable/etc/fake-hwclock.data which the means the date and time is incorrect, have you any idea how to fix this?

NTP should be setting the time at some point. Even if fake-hwclock worked, it would only set the time closer to actual, it still wouldn't be correct.

nickbruton, 2020-01-02:
Can't find it now but as you say ntp should be running but it still only ever gets the time as Feb 14 1999

nickbruton, 2020-01-02:
But as the pi zero reboots when you start the car ntp will never be able to set the correct time?

marcone, 2020-01-02:
It should set the time when it starts archiving. Are you actually running in to any problems because the time is not set at boot?

nickbruton, 2020-01-02:
I don't think it can upload any saved clips because the time is incorrect, I will test it again shortly to confirm

marcone, 2020-01-02:
Like I said it should set the time when it starts archiving. You should see something like this in archiveloop.log:
```
Thu 14 Feb 02:12:43 PST 2019: Waiting for time to be set by ntpd...
Thu 14 Feb 02:13:10 PST 2019: Time still not set, attempting to force it
Thu  2 Jan 09:42:55 PST 2020: Archiving...
```
Note that even if fake-hwclock worked, the time would still not be correct, since it would simply set the time to whatever time it was the last time it updated the fake-hwclock.data file, which could be hours or days ago.

marcone, 2020-01-02:
as to why it complains about /etc/fake-hwclock.data not existing: it's apparently because /sbin/fake-hwclock is run before the remaining filesystems are mounted:
```
Feb 14 02:12:03 teslausb2 fake-hwclock[67]: Unable to read saved clock information: /etc/fake-hwclock.data does not exist
Feb 14 02:12:03 teslausb2 systemd[1]: Started Apply Kernel Variables.
Feb 14 02:12:04 teslausb2 systemd[1]: Started udev Coldplug all Devices.
Feb 14 02:12:04 teslausb2 systemd[1]: Starting Helper to synchronize boot up for ifupdown...
Feb 14 02:12:04 teslausb2 systemd[1]: Started Set the console keyboard layout.
Feb 14 02:12:04 teslausb2 kernel: EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Feb 14 02:12:04 teslausb2 systemd[1]: Started Remount Root and Kernel File Systems.
...
Feb 14 02:12:13 teslausb2 systemd[1]: Mounting /backingfiles...
Feb 14 02:12:13 teslausb2 systemd[1]: Mounting /mutable...
Feb 14 02:12:13 teslausb2 systemd[1]: Mounted /boot.
```
I'm not familiar enough with the linux boot process to say why it tries to run that command so early, or how to fix that.

commit 7a7890e8 (2020-01-04): Fix fake-hwclock

fake-hwclock was run too early during boot, making it run later
so that it actually works.

Fixes #391
closed 2020-01-04 by marcone