marcone/teslausb #396: couldn't determine idle interval

steveclv, 2020-01-12:
I thought that if the pi was connected to home Wifi and a sentry event occurred it would automatically be moved to the home share - but it isn't and I need to force it with a touch /tmp/archive_is_unreachable command. When I tail the archive log I see lots of this

Sun 12 Jan 13:12:54 GMT 2020: taking snapshot of cam disk: /backingfiles/snapshots/snap-000054/snap.bin
Sun 12 Jan 13:12:57 GMT 2020: took snapshot
Sun 12 Jan 13:12:58 GMT 2020: comparing /backingfiles/snapshots/snap-000053/snap.bin.toc and /backingfiles/snapshots/newsnap/snap.bin.toc
Sun 12 Jan 13:12:58 GMT 2020: making links for /backingfiles/snapshots/newsnap/mnt, retargeted to /backingfiles/snapshots/snap-000054/mnt
Sun 12 Jan 13:13:01 GMT 2020: made all links for /backingfiles/snapshots/newsnap/mnt
**Sun 12 Jan 14:11:01 GMT 2020: waiting up to 90 seconds for idle interval
Sun 12 Jan 14:12:42 GMT 2020: couldn't determine idle interval**

Any ideas?

Thanks
Steve

marcone, 2020-01-12:
> I thought that if the pi was connected to home Wifi and a sentry event occurred it would automatically be moved to the home share

No, that's not how it works. See the last bullet point on https://github.com/marcone/teslausb/wiki/Troubleshooting

closed 2020-01-12 by marcone

Jughead135, 2020-12-15:
Not necessarily an issue, but a question--this topic is the only one I've found related:

What does the OP's highlighted text mean?  Is there something "wrong" (or, at least not fully working) when the system can't determine the idle interval...?

I just copied from the OP below, but I get the exact same message in my `/mutable/archiveloop.log` file every time it sits idle long enough (> 1 hr).

```Sun 12 Jan 14:11:01 GMT 2020: waiting up to 90 seconds for idle interval
Sun 12 Jan 14:12:42 GMT 2020: couldn't determine idle interval```