marcone/teslausb #397: Led blinking sequence not increasing, stuck?

deadlyknapsack, 2020-01-12:
I'm trying to install the latest stable release on my Pi Zero W with a Sandisk 128GB sd card. Image is etched fine and I changed the .conf accordingly. When I power the Pi with the sd card in it, it blinks twice and then the led shows it's processing. Led turns off, blinks twice and does it again. It's not visible in my network, I don't see the Access Point on the Pi show up and I'm not sure what to do...

Any tips?

marcone, 2020-01-12:
Double-check your wifi information. If you think it correct, put the sd card in a PC and look at the teslausb-headless-setup.log file on the boot drive.

deadlyknapsack, 2020-01-12:
I triple checked the wifi, it is good. 

Also, I don't see any logfiles, at all. When I let it sit for half an hour (seems like it is going off an on again), nothing on boot is changed. There is supposed to be some kind of marker file for the setup right?

marcone, 2020-01-12:
> I triple checked the wifi, it is good.
> 
> Also, I don't see any logfiles, at all. When I let it sit for half an hour (seems like it is going off an on again), nothing on boot is changed. There is supposed to be some kind of marker file for the setup right?

There should be several, like TESLAUSB_SETUP_STARTED, WIFI_ENABLED, TESLAUSB_SETUP_FINISHED

If you're not even seeing /boot/teslausb-headless-setup.log though, then something else is wrong because creating that file is pretty much the first setup does.

> Just to be sure, Balena says the drive is unusually big for a drive. Is that just a warning, to make sure you don't accidentally flash a SSD or something or might 128gb be to big for teslausb?

That's just an Etcher warning to prevent you from accidentally flashing the boot drive of your computer. Teslausb has no limit on drive size.

deadlyknapsack, 2020-01-12:
I don't know what to tell you... just unwrapped a new Sandisk and new Raspberry Pi Zero W... I could try a different SD, but I only have one Zero laying around. Not common for these things to be dead on arrival right?

marcone, 2020-01-12:
When you plug the sd card in to a PC, what files do you see in the "boot" drive?
What is the content of the /boot/cmdline.txt and /boot/config.txt files?

deadlyknapsack, 2020-01-12:
Exactly the same as right after etching.

cmdline.txt:

dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=22eddddb-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2,g_ether quiet

config.txt

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtoverlay=dwc2

list of directory and files:

overlays
bcm2708-rpi-0-w.dtb
bcm2708-rpi-b-plus.dtb
bcm2708-rpi-b.dtb
bcm2708-rpi-cm.dtb
bcm2708-rpi-zero-w.dtb
bcm2708-rpi-zero.dtb
bcm2709-rpi-2-b.dtb
bcm2710-rpi-3-b-plus.dtb
bcm2710-rpi-3-b.dtb
bcm2710-rpi-cm3.dtb
bootcode.bin
cmdline.txt
config.txt
COPYING.linux
fixup_cd.dat
fixup_db.dat
fixup_x.dat
fixup.dat
issue.txt
kernel.img
kernel7.img
LICENCE.broadcom
LICENSE.oracle
ssh
start_cd.elf
start_db.elf
start_x.elf
start.elf
teslausb_setup_variables.conf
teslausb_setup_variables.conf.sample
wpa_supplicant.conf.sample

mobiletru, 2020-01-12:
Did you format sd card correctly ?

marcone, 2020-01-12:
>```
> ######################################################################
> 
> # Default variables for CIFS (Windows/Mac file sharing) setup
> export ARCHIVE_SYSTEM=cifs
> export archiveserver=your_archive_name_or_ip
> export sharename=your_archive_share_name
> export shareuser=username
> export sharepassword=password
>```

Just in case this is the actual conf file you're using: the 4 variables above should all be specified. I assume your archive server isn't actually called `your_archive_name_or_ip`

> ```
> export SSID='teslausb'
> export WIFIPASS='teslausb'
>```

Similarly here, this should be the name and password of the wifi network that the Pi is going to connect, e.g. your home wifi. Or did you actually set up a separate wifi access point for the Pi to connect to, with name `teslausb` and password `teslausb` ?

deadlyknapsack, 2020-01-12:
> Just in case this is the actual conf file you're using: the 4 variables above should all be specified. I assume your archive server isn't actually called `your_archive_name_or_ip`

Right. I assumed I could skip this, as I intent to only use the AP of the Pi on the go. My Tesla is not in range of my wifi unfortunately. Will it prevent setup? I could temporarily setup a share.

> Similarly here, this should be the name and password of the wifi network that the Pi is going to connect, e.g. your home wifi. Or did you actually set up a separate wifi access point for the Pi to [...]

Yes, I did. Just to make sure that my wifi wasn't trying to force the client on the 5Ghz.

marcone, 2020-01-12:
> Right. I assumed I could skip this, as I intent to only use the AP of the Pi on the go. My Tesla is not in range of my wifi unfortunately. Will it prevent setup? I could temporarily setup a share.

The archive not being reachable will certainly cause setup to fail. It should fail much later than what you're apparently seeing though, and log the failure in /boot/teslausb-headless-setup.log.
If you don't want it to archive, you can set `ARCHIVE_SYSTEM` to `none`. However, your conf file is also missing the variables needed for specifying the Pi's own AP information (`AP_SSID` and `AP_PASS` at a minimum, and you might want `SAMBA_ENABLED` too)

> Yes, I did. Just to make sure that my wifi wasn't trying to force the client on the 5Ghz.

And you're sure this 'teslausb' wifi is working properly?

Which image did you flash?

deadlyknapsack, 2020-01-12:
> The archive not being reachable will certainly cause setup to fail. It should fail much later than what you're apparently seeing though, and log the failure in /boot/teslausb-headless-setup.log.

I'm trying again with the settings set to a share on my Macbook Pro.

> If you don't want it to archive, you can set `ARCHIVE_SYSTEM` to `none`. However, your conf file is also missing the variables needed for specifying the Pi's own AP information (`AP_SSID` and `AP_PASS` at a minimum, and you might want `SAMBA_ENABLED` too)

I figured after I few tries I disabled the AP option for now and just get it working as simply set up as possible.

> And you're sure this 'teslausb' wifi is working properly?
>
>
> Which image did you flash?

Yes, 100% working fine. Both Macbook Pro and iPhone are running on it. 
I'm using this one: teslausb-20190615

mobiletru, 2020-01-12:
https://www.youtube.com/watch?v=ETs6r1vKTO8   
Have you watched this video

deadlyknapsack, 2020-01-12:
> https://www.youtube.com/watch?v=ETs6r1vKTO8
> Have you watched this video

Absolutely. I'm doing exactly what he is doing. It just fails the moment I put the SD in the Pi. It looks like it is in some infinite loop. I just reformatted it again, using terminal to be sure. Etched it again and still only flashes twice, following by some pulsating blink (processing) and two blanks and again and again and again. 

When I power off, put the SD in my mac, nothing has changed on the SD.

mobiletru, 2020-01-12:
Find a pc with a sd card reader and watch that video 10 times

deadlyknapsack, 2020-01-12:
> Find a pc with a sd card reader and watch that video 10 times

I'm not an idiot. I have 3 other Pi's running in my household. I don't need to watch a 30-minute video 10 times for this to get to work. I really appreciate the help, but that is a really unhelpful remark.

mobiletru, 2020-01-12:
export ARCHIVE_SYSTEM=cifs
export archiveserver=your_archive_name_or_ip
export sharename=your_archive_share_name
export shareuser=username
export sharepassword=password

your not watching the video

deadlyknapsack, 2020-01-12:
> export ARCHIVE_SYSTEM=cifs
> export archiveserver=your_archive_name_or_ip
> export sharename=your_archive_share_name
> export shareuser=username
> export sharepassword=password
> 
> your not watching the video

Can you go "help" somebody else? Thanks.

deadlyknapsack, 2020-01-12:
@marcone I just did everything you suggested, but still the Pi keeps looping with a freshly etched SD, updated .conf. After 10-15 minutes, still nothing has changed on the SD card. Could it be a hardware issue? Actual broken Pi?

mobiletru, 2020-01-12:
export ARCHIVE_SYSTEM=none  
 you didn't do everything he suggested

marcone, 2020-01-12:
I have no idea. The flashes you're seeing are likely not the flashes that the teslausb setup process would show, since it would have created the log file if it got that far, and you're not seeing the log file, so it's stopping very early in the boot process.
Could be a defective Pi, or inadequate power to the Pi.
One more thing you can try is to not put a config file at all, i.e. just flash the image and then stick it in the Pi and turn it on. This will make setup end early, but it should leave a USB network interface configured, which should allow you to ssh in to the Pi.

deadlyknapsack, 2020-01-13:
Alright, couple of tries later, I switched the USB cord and I'm getting somewhat further. The log is now being created, but it fails:

```
Sat 15 Jun 22:27:12 BST 2019 : Detecting whether to update wpa_supplicant.conf
Mon 13 Jan 18:17:33 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Mon 13 Jan 18:17:33 GMT 2020 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Mon 13 Jan 18:17:33 GMT 2020 : Rebooting...
Mon 13 Jan 18:18:01 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Mon 13 Jan 18:18:01 GMT 2020 : Grabbing main setup file.
Mon 13 Jan 18:18:22 GMT 2020 : get_script failed, retrying
Mon 13 Jan 18:18:46 GMT 2020 : get_script failed, retrying
Mon 13 Jan 18:19:09 GMT 2020 : get_script failed, retrying
Mon 13 Jan 18:19:33 GMT 2020 : get_script failed, retrying
```

I read something about the local time being off, because NTP wouldn't sync. I changed the time manually through SSH, because none of the Googled ntp fix suggestions worked for me. Is my ISP blocking something that sets the time correctly? 

Even with manual time set to - somewhat accurate, talking milliseconds off) - it fails the get_script. 

Any suggestions?

marcone, 2020-01-14:
I'm not sure I understand your comments. You say that your Pi is not connected to wifi, but you're also asking if your ISP is blocking something. If your Pi is not on your wifi, then whatever your ISP does is irrelevant. First step should be to get your Pi onto wifi. That's not a teslausb problem though.

deadlyknapsack, 2020-01-14:
I agree. My edit/additions to my comment are just new findings. At first I thought it was an NTP issue, because the get_script was failing (as I was reading about other issues), then I saw it wasn't connected to wifi in the first place and I tried a different approach. I just thought I updated my comment. I'm still trying and I do realise it's not a teslausb problem.

mobiletru, 2020-01-14:
Your wifi is not 2.4 ghz keep it simple

marcone, 2020-01-14:
@mobiletru you're not helping. If you read above you'll see that @deadlyknapsack specifically created a 2.4GHz-only network.

@deadlyknapsack are you sure you actually have a Pi Zero W, and not a Pi Zero? You wouldn't be the first person to accidentally order the wrong one (been there myself). To rule out any issue with the teslausb image, you might consider installing a standard Raspbian desktop image and see how that goes. If that still doesn't work, I'd say you have a defective Pi.

closed 2020-01-14 by marcone

deadlyknapsack, 2020-01-17:
@marcone 100% Pi Zero W. I double checked the print on the PCB. Also, I specifically ordered it, because the stack of normal Zero's I have laying around aren't compatible. I was already planning on installing Raspbian and see how the Wifi connection goes. Thanks for your help.

henroFall, 2020-01-17:
Make sure to use a utility to force the card to a fat32 format? I've
recently had problems with 128 gb cards when I forget that step. I had a
habit of just blowing all partitions off the card, formatting exFAT , then
let Win32DiskImager make the card. But that quit working and my pi would
boot like you describe. The SD card would get super hot, too. Like can't
touch it hot.

Once I formatted the card as fat32 that all quit.

Did you specially format the card first ?

marcone, 2020-01-17:
@henroFall Honestly, that sounds kinda bogus to me. If you flash a Raspbian image onto the card (either one of the teslausb prebuilts or a standard Raspbian one), that effectively formats the card by overwriting the partition table and creating two new partitions. I don't see what formatting it first would do, unless you do a "slow" format where it writes every sector of the card, which might allow it to find and remap bad blocks.

henroFall, 2020-01-17:
I know, right? I've been working with Pis for years and never did anything
to preformat the SD cards first. But the other day I was stuck hard and the
only thing that fixed it was using guiformat to format the card. I agree,
it blows all of that out and makes new partitions, anyway, so I don't get
it. But it worked, and it's easy to try/eliminate so I thought I'd suggest
it.

My symptoms were pretty much the same as what was described here or I
wouldn't have brought it up.

deadlyknapsack, 2020-01-19:
@marcone I flashed Buster Lite, created a wpa_supplicant.conf, booted in in the same pi and watched it through HDMI boot nicely and was online. So, Pi is working, wifi is working. I then formatted the SD card, etched your image, replaced the teslausb_setup_variables.conf and copied the working wpa_supplicant.conf on to the sd card. I then booted it, with HDMI connected and I can see it go through the first boot up steps, I can see the ip it is getting from my router. Then, it reboots, does a bunch of stuff and guess what... it still fails to get the get_script. It looks like it connects to wifi just fine the first boot run, then it says something like: "checking to see if wpa_supplicant.conf needs to be updated" and after that it doesn't work... it just tries the get_script infinitely...

deadlyknapsack, 2020-01-19:
I'm gonna try the manual installation steps and see how far I get. I just don't get it.

deadlyknapsack, 2020-01-19:
@marcone still have one question, if you don't mind. Now that I did the manual installation and it is working in my Tesla, I would like it to act as an AP. In your description, you mention to add the AP details to the teslausb_setup_variables.conf, but where is that file stored when I do a manual installation?

deadlyknapsack, 2020-01-19:
@marcone Sorry for spamming, but I couldn't let it go... I actually got it to work. I'd like to share my findings, because there might be something in it, worth to you:

When I was looking through the logs on my tv, while it was attached to HDMI, I noticed something like:

```[Starting] Disable Wifi if country isn't defined```

When I was making a `wpa_supplicant.conf` for the Buster Lite tryout, I remember using an example `wpa_supplicant.conf` which started with `country=US`. When I etched your latest release again (for a gazillionth time), I created a file next to `wpa_supplicant.conf.sample` called `wpa_supplicant.conf`and put this into it:

```
country=NL
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
  ssid="my_ssid_name"
  psk="my_wifi_password"
  key_mgmt=WPA-PSK
}
```

When I booted the pi with the teslausb sd in it, it connected to wifi fine, where it didn't when I just filled out the wifi details in the `teslausb_setup_variables.conf`

Does that make sense? With the `country`set to `NL` the `get_script`wasn't failing like it used to for literally dozens of tries.

marcone, 2020-01-20:
Could be a race condition. The setup script will add a default "country=US" when no country is specified, but of course if wifi was already disabled because the country code was missing, then it can't download the setup script.

marcone, 2020-01-23:
The latest prebuilt image includes the `country=...` entry in wpa_supplicant from the start, so if that was the issue it should be fixed now.