marcone/teslausb #398: Unable to mount to CISF

marckatznelson, 2020-01-13:
Hi - I went through couple of posts on this but did not find a solution, therefore posting it again.
I have tried flashing the microsd many times, but every time I have been unsuccessful in mounting to CISF. The log file says "verify-and-configure-archive: STOP: no working combination of vers and sec mount options worked".
I have been able to SSH and record the sentry videos. However, these never uploaded to my windows 10 share. When I inserted the Pi to my PC, I saw all the videos in Teslacam folder. However, the music on my share did not copy to my MUSIC folder.

I hope you could kindly help me in troubleshooting. Attaching the logs.
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/4056185/teslausb-headless-setup.log)
[reran-setup.txt](https://github.com/marcone/teslausb/files/4056186/reran-setup.txt)
[diagnostics.txt](https://github.com/marcone/teslausb/files/4056187/diagnostics.txt)

marcone, 2020-01-14:
Your diagnostics.txt looks like you pasted the diagnostics output into a shell, and then copied the result of that.

marckatznelson, 2020-01-14:
I didn't know how to export it in .txt from putty. Thanks for getting back so soon. Here is another attempt, after reflashed the drive in the mean time. there are some errors and music file did not get copied over.
[diagnostics.txt](https://github.com/marcone/teslausb/files/4056891/diagnostics.txt)
please let me know the right way to generate. pretty new at it.

marcone, 2020-01-14:
It looks to me like you stopped and restarted setup a number of times, without ever letting it fully finish. I suggest starting over from scratch and waiting until the "CAM" and "MUSIC" drives show up on your computer.

marckatznelson, 2020-01-14:
Yes. I did. Both CAM and MUSIC displayed on the PC and then I inserted it into TESLA. I tried to then run diagnostic remotely. Since, nothing was happening, I tried to rerun the setup remotely based on your suggestion to someone else.
Should I rerun the setup after connecting the Pi to PC or fully redo it.

marcone, 2020-01-14:
starting over from scratch

marckatznelson, 2020-01-14:
ok. thanks. It worked now.

closed 2020-01-14 by marcone