marcone/teslausb #401: Roadie App and Device

jbienz, 2020-01-17:
So I just stumbled upon [Roadie](https://www.tryroadie.com/). It looks like it could be built on TeslaUSB. Of course since TeslaUSB is MIT licenses, that's prefectly legit. They don't have to contribute anything back. But since the app seems quite nice and is free. if Roadie is based on TeslaUSB it would be nice to have the Roadie app work with TeslaUSB.

I can't find any forum for the Roadie. Only a support e-mail address of support@tryroadie.com. I haven't tried contacting them. Before doing that, I was wondering if any of the contributors of this project are part of that product.

miles267, 2020-01-17:
Whoa. This “looks” cool.

marcone, 2020-01-17:
> I can't find any forum for the Roadie. Only a support e-mail address of [support@tryroadie.com](mailto:support@tryroadie.com). I haven't tried contacting them. Before doing that, I was wondering if any of the contributors of this project are part of that product.

I don't know whether Roadie uses any part of teslausb

closed 2020-01-17 by marcone