marcone/teslausb #404: get_script failed, retrying...

jasonhyland, 2020-01-20:
Hi, having an issue on a Pi W getting this setup. teslausb-headless-setup.log;

```
Sat 15 Jun 22:27:31 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat 15 Jun 22:27:31 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sat 15 Jun 22:27:31 BST 2019 : Rebooting...
Sat 15 Jun 22:28:18 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat 15 Jun 22:28:18 BST 2019 : Grabbing main setup file.
Sat 15 Jun 22:28:19 BST 2019 : get_script failed, retrying
Sat 15 Jun 22:28:22 BST 2019 : get_script failed, retrying
```
But I'm not convinced the Pi is connecting to my Wifi. I setup a dedciated 2.4ghz SSID, and I can't see any station attached or and DHCP address issued. Here's my wpa_supplicant.conf.bak file;

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
  ssid="testla"
  psk="TEMPPASS"
  key_mgmt=WPA-PSK
}
```

I saw this which relates to the clock being wrong, which is clearly is on mine 

[image]

But any ideas about if it is or isn't getting a wifi connection? Thanks!

marcone, 2020-01-20:
The ".bak" file is irrelevant, it isn't used by anything.
You can try adding "country=US" to the wpa_supplicant.conf.sample in the boot drive after flashing and before putting the sd card in the Pi.

marcone, 2020-01-23:
The latest prebuilt image includes the `country=...` entry in wpa_supplicant from the start.

closed 2020-01-23 by marcone