marcone/teslausb #405: How can I verify its connecting to share?

blimeybloke, 2020-01-20:
I am not sure how I can verify its connecting to share.  I am using unraid and created a share.
I put server as 192.168.0.4
Share name is: TeslaCam

The share allows for guest access and I have verified that it works from my mac.  For the username/password section I just blanked them out:
export archiveserver=192.168.0.4
export sharename=TeslaCam
export shareuser=
export sharepassword=

Should I # out the share u/p lines since I have guest access enabled down below:
# uncomment to enable a samba server that exports the recordings
export SAMBA_ENABLED=true
# uncomment to enable guest access to the samba server
export SAMBA_GUEST=true

Hmm or should both of those be commented out?

thanks

marcone, 2020-01-20:
I'm not sure what you're trying to do. If setup completed, then it could connect to your share.
If setup did not complete, there's probably a message in the log that tells you why.

blimeybloke, 2020-01-20:
I am just trying to see if it is connecting to the share.  Everything finished and I am getting 2 blinks, but nothing is going into the share.
thanks

DMBlakeley, 2020-01-20:
This is what I do to confirm that files are being transferred to the share:

* To test, plug RPi into your computer, create a “SavedClips” folder in the CAM drive. Save some jpeg files in the SaveClips folder. Then open pi shell:
    * ssh pi@teslausb.local
    * sudo -i
    * /root/bin/remountfs_rw
    * touch /tmp/archive_is_unreachable
* LED on the pi should start to flash rapidly, the CAM drive will unmount, test files should now be transferring, CAM drive will remount. Check the shared “TeslaCam” folder and verify that test files transferred.
* Issue “halt” in pi shell and disconnect pi from computer.
* DONE

marcone, 2020-01-20:
> I am just trying to see if it is connecting to the share. Everything finished and I am getting 2 blinks, but nothing is going into the share.

What were you expecting to be copied to the share?
Did you check the log?

blimeybloke, 2020-01-20:
DMBlakeley - Thank you, I will test that tonight when I get home
Marcone - I am not sure tbh.. hopefully the video files?  I will test what DMBlakeley suggested and report back.  Thank you

marcone, 2020-01-20:
> Marcone - I am not sure tbh.. hopefully the video files?

Which video files? Do you actually have any Sentry events or manually saved clips?

closed 2020-01-21 by blimeybloke