marcone/teslausb #406: Folder TeslaCam is not synchronize

denisjoshua, 2020-01-21:
Hi there all, 
frst of all thanks a lot Marcone for this wonderfoll updates of Tesla USB

I just setuup a TeslaUSB on my new PiZero and I think everithing it's ok.
When I put some music on my NAS shared folder and I power on the PI the musis is synchronize.

But when I put a sample.mp4 in the PI TeslaCam folder this sample.mp4 is not arrive in the NAS TeslaCam shared folder.
Then I create the Savedclips and SentryClips folder inside the TeslaCam folder. 

So my question is... I must rename the sample.txt in some strange world or somethink like 01.12.30-blablabla.mp4 ?
How I can simulate Tesla car using my PC please ? ... the car still must arrive.

I allready have the /home/pi/diagnostics.txt file but I don't know how to take it out of my PI
When I go to \\192.162.1.200 (my PI IP) it ask me user and pass but the pi user and pass that I use 
to connect to the PI via ssh is not work. 

Thanks a lot again 
Denis

marcone, 2020-01-21:
> But when I put a sample.mp4 in the PI TeslaCam folder this sample.mp4 is not arrive in the NAS TeslaCam shared folder.
> Then I create the Savedclips and SentryClips folder inside the TeslaCam folder.

Only files inside those SavedClips and SentryClips folders are archived, not files directly inside of TeslaCam.

> I allready have the /home/pi/diagnostics.txt file but I don't know how to take it out of my PI
> When I go to \192.162.1.200 (my PI IP) it ask me user and pass but the pi user and pass that I use
> to connect to the PI via ssh is not work.

The default username and password for the Pi's Samba server are pi/raspberry, but since you can ssh in to the Pi, you can also just use scp to copy the file over, or manually copy it to your archive share, e.g. `sudo mount /mnt/archive; sudo cp /home/pi/diagnostics /mnt/archive`

denisjoshua, 2020-01-21:
Dear Marcone, 
thanks a lot for the quick answe.

So I finally explore the PI using my PC and using pi/raspberry user/pass. 
Is not the same pass that I use when I ssh into PI (that one I changed with help of your security page) 
How do I change the pi samba password please in order tobe the same as pi ssh user ? 

I have this situation: 
- if I put the PI on the PC USB there are 2 usb disks... MUSIC and CAM
- if I go in the CAM disk I have TeslaCam folder and there I maked the Savedclips folder and SentryClips
- I put a sample file (teslasavedcam_test.mp4 and teslasentrycam_test.mp4) in this folder and reset the PI

The strange thinks is: 
- I then open via explorer the \\192.168.1.200 and there is only the TeslaCAM folder not TeslaMusic
- if I open the TeslaCam folder and then SentryClips the folder is empty (also Savedclips folder) 
 so I didn't fine there the teslasavedcam_test.mp4 and teslasentrycam_test.mp4 files
- if I ope the NAS shared folder I cand find the teslasavedcam_test.mp4 and teslasentrycam_test.mp4 files

So cause I'm newbe... the USB drive of PI is not the same as TeslaCAm folder when I navigate via explorer into the PI ip ? 
How many TeslaCam we have in the PI please ? 

I think the best idea is to make a newbe like user manual that explain the PI structure and the steps that it make when we put it on the PC USB or the Tesla USB. 

Last think... I make the "sudo mount /mnt/archive" command but I cant see folder there
Is "total 8" but nothing soww me: 

```
pi@teslausb:/mnt/archive $ sudo mount /mnt/archive
pi@teslausb:/mnt/archive $ sudo ls -la
total 8
drwxr-xr-x 2 root root 4096 Jan 18 14:40 .
drwxr-xr-x 7 root root 4096 Jan 18 14:40 ..

```
Thanks a lot again 
Denis

marcone, 2020-01-21:
> How do I change the pi samba password please in order tobe the same as pi ssh user ?

Using the `smbpasswd` command.
 
> * I then open via explorer the \192.168.1.200 and there is only the TeslaCAM folder not TeslaMusic

The music drive is not exposed as a network share.

> * if I open the TeslaCam folder and then SentryClips the folder is empty (also Savedclips folder)
>   so I didn't fine there the teslasavedcam_test.mp4 and teslasentrycam_test.mp4 files

This is probably because teslausb expects the file-layout that the car creates, and the car normally writes files in subfolders inside of SavedClips and SentryClips, e.g. TeslaCam/SentryClips/2020-01-19_12-54-45/2020-01-19_12-52-22-back.mp4 

> So cause I'm newbe... the USB drive of PI is not the same as TeslaCAm folder when I navigate via explorer into the PI ip ?

Right.

> How many TeslaCam we have in the PI please ?

I don't understand what you're asking.

> I think the best idea is to make a newbe like user manual that explain the PI structure and the steps that it make when we put it on the PC USB or the Tesla USB.

You're welcome to contribute such documentation.

> Last think... I make the "sudo mount /mnt/archive" command but I cant see folder there
> Is "total 8" but nothing soww me:
>
> ```
> pi@teslausb:/mnt/archive $ sudo mount /mnt/archive [...]

This is probably because you were already inside the /mnt/archive folder before mounting the server. Try:
```
cd
sudo -i
mount /mnt/archive
ls -la /mnt/archive
```

denisjoshua, 2020-01-21:
Thanks a lot for the answer. 
Now the mount is work :-) 

I can't make e documentation cause I use translate to write here. 
I'm italian... (really I'm romanian that live in Italy) 

Anyway what i mean is... where I must put a sample video files 
in order to see that if I restart the PI,
the *.mp4 files will arrive in my NAS shared folder please ? 

Thanks a lot again 
Denis

marcone, 2020-01-22:
Like I said above, the car writes the files into subfolders of SentryClips, so TeslaCam/SentryClips/dummyfolder/yourtestfile.mp4 should work.

closed 2020-01-22 by marcone

denisjoshua, 2020-01-22:
Well I put the PI into PC USB... and in the USB unit I make a sample file like: 
G:\TeslaCam\SentryClips\2020-01-22\2020-01-22_12-52-22-back.mp4

Then I reset the PI... still in my USB PC interface. 
When PI power on, the LED flash continuously for some seconds and then flash 2 times every second. 
At this point was mounted the 2 USB unit in my PC. (TeslaCam and TeslaMusic) 

Go in the NAS shared folder TeslaCam/SentryClips and there it is the folder 2020-01-22
...but the 2020-01-22_12-52-22-back.mp4 is not there. 
Is this wrong please ? can be cause was 0 kb ? 

Thanks again 
Denis

marcone, 2020-01-22:
Yes, 0-byte files are not archived. Use a file of at least 100KB.

denisjoshua, 2020-01-22:
As sun as I return home I test another 1 mb file and... 
Yes :-) ... it's work. 
The new file is in the nas shared folder. 

Thanks a lot again for all help 
Denis